pymodbus-dev/pymodbus #395: dev.read_discrete_inputs can return wrong response class, sync issue?

mberntsen, 2019-03-15:
I've got an application that reads and writes coils. I simulate the devices using pymodbus server.
When I quit the server and restart it, I sometimes get a wrong response object in my client code:

This is what my client does:
rq = self.dev.write_coils(0, co)
wait 100 ms
di = self.dev.read_discrete_inputs(0, 4)
ir = self.dev.read_holding_registers(0, 4)

sometimes di or ir return the pymodbus.bit_write_message.WriteMultipleCoilsResponse class, which suggests a sync error in the transaction manager.

I cannot share all my code right now, but i'm happy to supply more information if needed, by restarting the server simulator a few times it's easily reproducible.

log output of pymodbus after quitting (connection refused) and restarting (and at the end quitting again):
2019-03-15 12:06:00,179 ERROR    sync:207 Connection to (localhost, 1504) failed: [Errno 111] Connection refused
2019-03-15 12:06:00,180 ERROR    sync:207 Connection to (localhost, 1507) failed: [Errno 111] Connection refused
2019-03-15 12:06:00,180 ERROR    sync:207 Connection to (localhost, 1505) failed: [Errno 111] Connection refused
2019-03-15 12:06:00,180 ERROR    sync:207 Connection to (localhost, 1506) failed: [Errno 111] Connection refused
2019-03-15 12:06:00,280 ERROR    factory:225 Unable to decode response Modbus Error: Unknown response 0
2019-03-15 12:06:00,280 ERROR    transaction:188 Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/transaction.py", line 168, in execute
    request.unit_id)
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2019-03-15 12:06:00,281 ERROR    factory:225 Unable to decode response Modbus Error: Unknown response 0
2019-03-15 12:06:00,283 ERROR    transaction:188 Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/transaction.py", line 168, in execute
    request.unit_id)
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2019-03-15 12:06:00,282 ERROR    factory:225 Unable to decode response Modbus Error: Unknown response 0
2019-03-15 12:06:00,282 ERROR    factory:225 Unable to decode response Modbus Error: Unknown response 0
2019-03-15 12:06:00,283 ERROR    transaction:188 Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/transaction.py", line 168, in execute
    request.unit_id)
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2019-03-15 12:06:00,284 ERROR    transaction:188 Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/transaction.py", line 168, in execute
    request.unit_id)
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2019-03-15 12:06:00,285 ERROR    factory:225 Unable to decode response Modbus Error: Unknown response 0
2019-03-15 12:06:00,287 ERROR    transaction:188 Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/transaction.py", line 168, in execute
    request.unit_id)
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2019-03-15 12:06:00,288 ERROR    factory:228 index out of range
2019-03-15 12:06:00,288 ERROR    transaction:188 Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/transaction.py", line 168, in execute
    request.unit_id)
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/home/martijn/.local/lib/python3.5/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2019-03-15 12:06:01,778 ERROR    sync:207 Connection to (localhost, 1507) failed: [Errno 111] Connection refused
2019-03-15 12:06:01,778 ERROR    sync:207 Connection to (localhost, 1508) failed: [Errno 111] Connection refused
2019-03-15 12:06:01,779 ERROR    sync:207 Connection to (localhost, 1506) failed: [Errno 111] Connection refused
2019-03-15 12:06:01,779 ERROR    sync:207 Connection to (localhost, 1504) failed: [Errno 111] Connection refused
2019-03-15 12:06:01,782 ERROR    sync:207 Connection to (localhost, 1504) failed: [Errno 111] Connection refused

dhoomakethu, 2019-03-15:
What version of pymodbus are you running ?

mberntsen, 2019-03-20:
2.1.0

dhoomakethu, 2019-03-21:
Could you please give a try with the latest release candidate `pip install pymodbus==2.2.0rc4` and update ?

closed 2021-03-08 by github-actions[bot]