pymodbus-dev/pymodbus #396: tcp framer isframeready, checkframe incomplete -> resetframe??

mberntsen, 2019-03-20:
excerpt from the socket_framer.py:
`
while True:
            if self.isFrameReady():
                if self.checkFrame():
                    if self._validate_unit_id(unit, single):
                        self._process(callback)
                    else:
                        _logger.debug("Not a valid unit id - {}, "
                                      "ignoring!!".format(self._header['uid']))
                        self.resetFrame()
                else:
                    _logger.debug("Frame check failed, ignoring!!")
self.resetFrame()
`
if the header is received, isFrameReady() will return true, 
if the data is not complete yet, checkFrame() will return false,
this leads to the frame being dropped, while the missing data leads to an invalid frame in the next read (the data is received, but nog yet put into the buffer
If you would wait for the data, but it would not come, the framer would get stuck in the loop, but the current code leads to dropped frames while the data (checked with wireshark) is correct, 1 tcp frame for the request and 1 for the response, with all data complete.

excerpt from my log:
2019-03-20 21:37:15,287 DEBUG    socket_framer:147 Processing: 0x3 0xe5 0x0 0x0 0x0 0x4 0x0 0x2
2019-03-20 21:37:15,287 DEBUG    socket_framer:159 Frame check failed, ignoring!!
2019-03-20 21:37:15,287 DEBUG    transaction:374 Getting transaction 997
2019-03-20 21:37:15,287 DEBUG    transaction:180 Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2019-03-20 21:37:15,287 DEBUG    transaction:111 Current transaction state - TRANSACTION_COMPLETE
2019-03-20 21:37:15,287 DEBUG    transaction:115 Running transaction 998
2019-03-20 21:37:15,288 DEBUG    transaction:206 SEND: 0x3 0xe6 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x0 0x0 0x5
2019-03-20 21:37:15,288 DEBUG    sync:75 New Transaction state 'SENDING'
2019-03-20 21:37:15,288 DEBUG    transaction:209 Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2019-03-20 21:37:15,288 DEBUG    transaction:282 Incomplete message received, Expected 6 bytes Recieved 8 bytes !!!!
2019-03-20 21:37:15,289 DEBUG    transaction:284 Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2019-03-20 21:37:15,289 DEBUG    transaction:214 RECV: 0x1 0x0 0x3 0xe6 0x0 0x0 0x0 0xd
2019-03-20 21:37:15,289 DEBUG    socket_framer:147 Processing: 0x1 0x0 0x3 0xe6 0x0 0x0 0x0 0xd
2019-03-20 21:37:15,289 DEBUG    socket_framer:159 Frame check failed, ignoring!!

dhoomakethu, 2019-03-21:
There is a potential fix for this issue in v2.2.0rc4, please give a try on the latest release candidate  `pip install pymodbus==2.2.0rc4`

mberntsen, 2019-03-26:
The release candidate has been running since friday without any issue, looks good

closed 2019-03-27 by dhoomakethu