pymodbus-dev/pymodbus #397: How to cancel a client write request after it times out ?

ondrej1024, 2019-03-21:
### Versions

* Python: 2.7
* OS: Raspbian Stretch Lite
* Pymodbus: 1.2.0

### Pymodbus Specific
* Client: tcp - sync

### Description

When issuing a client write request `client.write_register() `via synchronous TCP method, in some cases the request times out without receiving a response from the server. So far so good. 
However, even after closing the connection the request seems to remain in the queue of the server socket and is eventually processed by the server.
Instead I expect the request to be removed from the server queue when closing the connection so I can be sure that it is not processed anymore after the timeout.
Is this a bug or a feature ? Any workaround to cancel the request definitely?

### Code and Logs

```python
   from pymodbus.client.sync import ModbusTcpClient

   # Call modbustcp client to write register value
   client = ModbusTcpClient(host=ip_addr, port=tcp_port)
   if client.connect() == False:
      syslog.syslog(syslog.LOG_ERR, "Connection to Modbus slave failed")
      return -2
      
   reply = client.write_register(address=reg, value=rval, unit=mod)
   client.close()

   if reply == None:
      # No reply received: timeout occured
      syslog.syslog(syslog.LOG_ERR, "Timeout")
      return -3
      # It was observed that the write_register() request is still performed after timeout.
      # How can this be avoided ?
 
```

dhoomakethu, 2019-03-21:
You are using very old  version of pymodbus, Please give a try with latest pymodbus==2.2.0.rc4 .

ondrej1024, 2019-03-21:
OK, now I have tried with the latest stable version 2.1.0, installed with pip. The issue persists. 

Here are the details:

I have a ModbusTCP server listening on port 5010:
```
Netid        State          Recv-Q         Send-Q                  Local Address:Port                 Peer Address:Port        
tcp          LISTEN         0              1                           127.0.0.1:5010                      0.0.0.0:*           
```
I start a PyModbus connection with `client.connect()` and get 2 new connections in state ESTAB:
```
Netid       State         Recv-Q         Send-Q                  Local Address:Port                  Peer Address:Port         
tcp         LISTEN        0              1                           127.0.0.1:5010                       0.0.0.0:*            
tcp         ESTAB         0              0                           127.0.0.1:54487                    127.0.0.1:5010         
tcp         ESTAB         0              0                           127.0.0.1:5010                     127.0.0.1:54487        
```
Now I request to write a register with `client.write_register()`, connection status remains the same, even after the request has timed out : 
```
Netid       State         Recv-Q         Send-Q                  Local Address:Port                  Peer Address:Port         
tcp         LISTEN        0              1                           127.0.0.1:5010                       0.0.0.0:*            
tcp         ESTAB         0              0                           127.0.0.1:54487                    127.0.0.1:5010         
tcp         ESTAB         0              0                           127.0.0.1:5010                     127.0.0.1:54487        
```
Now I close the connection with `client.close()`, and connection status changes:
```
Netid       State             Recv-Q        Send-Q                Local Address:Port                 Peer Address:Port         
tcp         LISTEN            0             1                         127.0.0.1:5010                      0.0.0.0:*            
tcp         CLOSE-WAIT        1             0                         127.0.0.1:5010                    127.0.0.1:33877        
tcp         FIN-WAIT-2        0             0                         127.0.0.1:33877                   127.0.0.1:5010         
```

It remains like this until the request is eventually processed by the server. After this the 2 connections are closed:
```
Netid        State          Recv-Q         Send-Q                  Local Address:Port                 Peer Address:Port        
tcp          LISTEN         0              1                           127.0.0.1:5010                      0.0.0.0:*    
```

This behaviour is a problem because the client assumes the request has failed because it wasn't performed within the timeout but it succeeds later on.
Is it possible to cancel the request after `client.write_register()` has timed out to be sure it will not be processed anymore?

dhoomakethu, 2019-03-21:
what happens if you explicitly call `client.close()` when the timeout happens ? Will that help ?

ondrej1024, 2019-03-21:
I _always_ call `client.close()` directly after `client.write_register()`.

dhoomakethu, 2019-03-22:
Is the server here is pymodbus server or any third party modbus tcp slave ?

ondrej1024, 2019-03-25:
The server is a custom server developed by me based on the C library [libmodbus](https://libmodbus.org).

Thinking some more about this I've come to the conclusion that PyModbus is behaving correctly. It looks like it has no way of knowing if a connection to the server is still pending in the queue or if it is already being processed by the server.

I observed that all connections from the PyModbus client become immediately "ESTAB" even though the server accepts one connection at a time. This is probably e feature of the TCP handling of the Linux kernel.

This is an example of 3 client requests, 1 is being processed and 2 still in the queue of the TCP socket:
```
Netid           State             Recv-Q            Send-Q                        Local Address:Port                        Peer Address:Port            
tcp             LISTEN            2                 1                                 127.0.0.1:5010                             0.0.0.0:*               
tcp             ESTAB             0                 0                                 127.0.0.1:39517                          127.0.0.1:5010            
tcp             ESTAB             0                 0                                 127.0.0.1:5010                           127.0.0.1:39517           
tcp             ESTAB             0                 0                                 127.0.0.1:37113                          127.0.0.1:5010            
tcp             ESTAB             0                 0                                 127.0.0.1:56285                          127.0.0.1:5010            
tcp             ESTAB             12                0                                 127.0.0.1:5010                           127.0.0.1:56285           
tcp             ESTAB             12                0                                 127.0.0.1:5010                           127.0.0.1:37113           
```
So I think I need to handle the connection timeout in the server by time stamping the connection when the request is performed and checking the timeout when the connection is processed.

dhoomakethu, 2019-03-25:
Thanks for the clarification, I have checked with the pymodbus server and it closes the connection whenever a client is disconnected. Since this is an issue with external server, I will close this issue.

closed 2019-03-27 by dhoomakethu