pymodbus-dev/pymodbus #398: StartSerialServer async and port parameters

divi255, 2019-03-22:
async StartSerialServer should pass bytesize, parity and stopbits to twisted SerialPort.

the only way to do this now is to write own StartSerialServer helper.

dhoomakethu, 2019-03-22:
What version of pymodbus are you using ? Could you please share any code samples on how you are initialising the client ?  If you follow the latest examples on [async serial client based on twisted](https://github.com/riptideio/pymodbus/blob/master/examples/common/async_twisted_client_serial.py), you can pass these serial values as kwargs.
```
    proto, client = AsyncModbusSerialClient(schedulers.REACTOR,
                                            method="rtu", 
                                            port=SERIAL_PORT, 
                                            timeout=2,
                                            parity="N",
                                            stopbits=8,
                                            proto_cls=ExampleProtocol)
```

divi255, 2019-04-10:
actually I see it's already fixed in 2.2.0

https://github.com/riptideio/pymodbus/commit/e4f202cb0973b08c8c38c0f667c4b29b80a1a64e

closed 2019-04-10 by divi255