pymodbus-dev/pymodbus #399: RedisSlaveContext bug

charliboy, 2019-03-23:
### Versions

* Python:3.6.4
* OS:win 10
* Pymodbus:2.10
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

When using RedisSlaveContext, communication is not possible, and communication address validation is wrong 

### Code and Logs

```python
 # code and logs here.
#!/usr/bin/env python
"""
Pymodbus Synchronous Server Example
--------------------------------------------------------------------------

The synchronous server is implemented in pure python without any third
party libraries (unless you need to use the serial protocols which require
pyserial). This is helpful in constrained or old environments where using
twisted is just not feasible. What follows is an example of its use:
"""
# --------------------------------------------------------------------------- #
# configure the service logging
# --------------------------------------------------------------------------- #
#server code
import logging

from pymodbus.datastore import ModbusServerContext
from pymodbus.datastore.database import RedisSlaveContext
from pymodbus.device import ModbusDeviceIdentification
# --------------------------------------------------------------------------- #
# import the various server implementations
# --------------------------------------------------------------------------- #
from pymodbus.server.sync import StartTcpServer

FORMAT = ('%(asctime)-15s %(threadName)-15s'
          ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def run_server():

    # ----------------------------------------------------------------------- #
    store = RedisSlaveContext( )
    context = ModbusServerContext(slaves=store, single=True)

    # ----------------------------------------------------------------------- #
    # initialize the server information
    # ----------------------------------------------------------------------- #
    # If you don't set this or any fields, they are defaulted to empty strings.
    # ----------------------------------------------------------------------- #
    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = '1.5'

    # ----------------------------------------------------------------------- #
    # run the server you want
[... 168 lines cut ...]
2019-03-23 22:37:26,094 MainThread      DEBUG    socket_framer  :147      Processing: 0x0 0x2 0x0 0x0 0x0 0x3 0x1 0x85 0x2
2019-03-23 22:37:26,094 MainThread      DEBUG    factory        :246      Factory Response[133]
2019-03-23 22:37:26,094 MainThread      DEBUG    transaction    :363      Adding transaction 2
2019-03-23 22:37:26,094 MainThread      DEBUG    transaction    :374      Getting transaction 2
2019-03-23 22:37:26,094 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2019-03-23 22:37:26,094 MainThread      DEBUG    transaction    :111      Current transaction state - TRANSACTION_COMPLETE
2019-03-23 22:37:26,094 MainThread      DEBUG    transaction    :115      Running transaction 3
2019-03-23 22:37:26,094 MainThread      DEBUG    transaction    :206      SEND: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x1 0x0 0x0 0x0 0x1
2019-03-23 22:37:26,094 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2019-03-23 22:37:26,094 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2019-03-23 22:37:26,094 MainThread      DEBUG    transaction    :284      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2019-03-23 22:37:26,094 MainThread      DEBUG    transaction    :214      RECV: 0x0 0x3 0x0 0x0 0x0 0x3 0x1 0x81 0x2
2019-03-23 22:37:26,094 MainThread      DEBUG    socket_framer  :147      Processing: 0x0 0x3 0x0 0x0 0x0 0x3 0x1 0x81 0x2
2019-03-23 22:37:26,094 MainThread      DEBUG    factory        :246      Factory Response[129]
2019-03-23 22:37:26,094 MainThread      DEBUG    transaction    :363      Adding transaction 3
2019-03-23 22:37:26,094 MainThread      DEBUG    transaction    :374      Getting transaction 3
2019-03-23 22:37:26,094 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Traceback (most recent call last):
  File "C:/Users/LiPing/PycharmProjects/untitled/testclient.py", line 113, in <module>
    run_sync_client()
  File "C:/Users/LiPing/PycharmProjects/untitled/testclient.py", line 50, in run_sync_client
    assert(not rq.isError())     # test that we are not an error
AssertionError

```

dhoomakethu, 2019-03-29:
Can you give me your redis schema ? And are there any entries in your redis db when you start polling ?

charliboy, 2019-03-29:
> Can you give me your redis schema ? And are there any entries in your redis db when you start polling ?

Redis is the default configuration, not new schema. I understand that is the automatic generation when data is writing,But when the data is written from the client, the server will check the address first, and the server can not pass the address check.

dhoomakethu, 2019-03-29:
Yes that is because the `RedisSlaveContext` does not initialise the db with default values and the db is essentially empty. If you can give a try by creating dummy entries with redis-cli `set pymodbus:h:<offset> <value>` and see if the writes works as expected. If you are using python3 , the read might not work after this and might end up with struct error. Having said that redis context purely experimental and is not complete. Feel free to play-around and raise a PR if you wish.

charliboy, 2019-04-04:
@dhoomakethu ,thanks
I've corrected some bugs in the redis_datastore file and can now pass the above client code test in Python 3. But I haven't tested it in Python 2. 

#Redis database initialization code 
   host='localhost'
    port=6379
    r=redis.Redis(host=host, port=port)
    pre='pymodbus'
    num=100
    vals=[b'\x00\x00']*num
    c=('%s:c:%s' % (pre, v) for v in  range(num))
    c=dict(zip(c, vals))
    r.mset(c)
    d = ('%s:d:%s' % (pre, v) for v in range(num))
    d = dict(zip(d, vals))
    r.mset(d)
    vals=[0]*num
    h = ('%s:h:%s' % (pre, v) for v in range(num))
    h = dict(zip(h, vals))
    r.mset(h)
    i = ('%s:i:%s' % (pre, v) for v in range(num))
    i = dict(zip(i, vals))
    r.mset(i)
#redis_datastore.py
import redis
from pymodbus.interfaces import IModbusSlaveContext
from pymodbus.utilities import pack_bitstring, unpack_bitstring

# ---------------------------------------------------------------------------#
# Logging
# ---------------------------------------------------------------------------#
import logging;

_logger = logging.getLogger(__name__)

# ---------------------------------------------------------------------------#
# Context
# ---------------------------------------------------------------------------#
class RedisSlaveContext(IModbusSlaveContext):
    '''
    This is a modbus slave context using redis as a backing
    store.
    '''

    def __init__(self, **kwargs):
        ''' Initializes the datastores

        :param host: The host to connect to
        :param port: The port to connect to
        :param prefix: A prefix for the keys
        '''
        host = kwargs.get('host', 'localhost')
        port = kwargs.get('port', 6379)
        self.prefix = kwargs.get('prefix', 'pymodbus')
        self.client = kwargs.get('client', redis.Redis(host=host, port=port))
        self.__build_mapping()

    def __str__(self):
        ''' Returns a string representation of the context

        :returns: A string representation of the context
        '''
        return "Redis Slave Context %s" % self.client

    def reset(self):
        ''' Resets all the datastores to their default values '''
        self.client.flushall()

    def validate(self, fx, address, count=1):
        ''' Validates the request to make sure it is in range
[... 179 lines cut ...]
        :param offset: The address offset to start at
        :param count: The number of bits to read
        '''
        response = self.__get_reg_values(key, offset, count)
        response = [int((r or self.__reg_default).decode()) for r in response]
        print('reg response',response)
        return response

    def __set_reg(self, key, offset, values):
        '''

        :param key: The key prefix to use
        :param offset: The address offset to start at
        :param values: The values to set
        '''
        count = len(values)
        # s = divmod(offset, self.__reg_size)
        # e = divmod(offset+count, self.__reg_size)

        # current = self.__get_reg_values(key, offset, count)

        key = self.__get_prefix(key)
        request = ('%s:%s' % (key, v) for v in range(offset, offset+count + 1))
        request = dict(zip(request, values))
        self.client.mset(request)

dhoomakethu, 2019-04-04:
Thanks @charliboy I am working on some enhancements as well for redis store, I will update it here once thats ready.  Can you confirm if the reads are also working fine on Python3 ?

charliboy, 2019-04-04:
@dhoomakethu ,Yes, at present, the above code works well on pyhon 3 and can pass the test  code on the client side.

dhoomakethu, 2019-11-05:
Should be fixed with v2.2.0 and above

closed 2019-11-05 by dhoomakethu