pymodbus-dev/pymodbus #401: Distribute License with Source

cdgarren, 2019-04-02:
Should the license be distributed with the source tarball? I'm trying to create a yocto recipe to use this and it generally expects there to be a license file, but the download from pypi does not have it included.

dhoomakethu, 2019-04-03:
[LICENSE](https://github.com/riptideio/pymodbus/blob/master/doc/LICENSE) is part of the doc folder. You can modify the recipe to fetch it from there.

closed 2019-04-03 by dhoomakethu

cdgarren, 2019-04-03:
I understand that it's part of the doc folder. However, yocto recipes can easily pull from pypi, and your distribution there does NOT include the License file.

dhoomakethu, 2019-04-03:
Got it, I will make sure it is available as part of pypi in future releases. Thanks

ffontaine, 2019-10-22:
It seems that latest release (2.3.0rc1) on pypi doesn't contain the license file. It would be really nice to add it on next release.

dhoomakethu, 2019-11-05:
@ffontaine @cdgarren Please check v2.3.0. The license should be part of the source distribution now.

commit 3fb83cc1 (2019-10-29): Fix #461 - Udp client/server , Fix #401 - package license with source, #457 Fix typo's in docstrings, #455-Support float16