pymodbus-dev/pymodbus #402: Use pycryptodome instead of pycrypto.

lalit-lintel, 2019-04-05:
### Versions

* Python: 2.7.12
* OS: Ubuntu 18.04
* Pymodbus: 2.1.0 [twisted]
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp - async

### Description

I am trying to use Mod bus server on TCP protocol, but when I installed pymodbus and I saw it's installed pycrypto, which is deprecated and dead software. 

I already have installed pycryptodome in my application, which is a conflict with pycrypto, 
we can't have both pycrypto and pycryptodome at the same time,

Can we have pymodbus[twisted] release which can use pycryptodome instead of pycrypto?

dhoomakethu, 2019-04-08:
I am not sure of the amount of effort this might need, I would be glad to accept a PR if you can give a try with pycryptodome and can assure nothing breaks.

tracernz, 2019-04-08:
AFAICT twisted doesn't use PyCrypto? https://github.com/twisted/twisted/pull/953

lalit-lintel, 2019-04-20:
Added pull request with changes in the setup file, which will allow installing Cryptodome instead Crypto.

Pull request #409

PR #411 (2019-04-22): Remove unused pycrypto dep
It has not been needed by Twisted for a long time, and has been unmaintained for a long time.

Fixes #402.
closed 2021-03-08 by github-actions[bot]