pymodbus-dev/pymodbus #404: better retries for incorrect data received

wildstray, 2019-04-10:
### Versions

* Python: 2.7.9, 3.4.2
* OS: Debian 8.10
* Pymodbus: 2.2.0
* Modbus Hardware (if used):  Qeed TCP bridge and modbus RTU meters (QE-POWER-T-STD, QE-BR-ETH485 datasheets on http://www.qeed.it/)

### Pymodbus Specific
* Client: tcp - sync

### Description

When I try to read some meters across a TCP bridge (so I use ModbusTcpClient) sometimes (randomly, probably due to buggy firmware) I received inconsistent reply eg. the zero response you can see in the attached Wireshark capture.
In this cases I expected a retry, instead I receive an exception or response.isError() = True but no retries. In some cases I obtain an exception, in some other (with the same responses!) isError().
I noticed the retry loop (```while retries > 0 ```) in transaction.py works with raw retries and expect an empty reply to retry. In my case the response is always non-empty, but need to be validated for lenght and unit id to verify if it's a valid response (I think self.client.framer.processIncomingPacket, unluckily done only after the reply loop).

### Code and Logs

isError() = True case:

```python
2019-04-10 17:34:51,855 pymodbus.transaction modbus_tcp.32 DEBUG Current transaction state - IDLE
2019-04-10 17:34:51,856 pymodbus.transaction modbus_tcp.32 DEBUG Running transaction 1
2019-04-10 17:34:51,856 pymodbus.transaction modbus_tcp.32 DEBUG SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x2 0x3 0x0 0xf4 0x0 0x4
2019-04-10 17:34:51,857 pymodbus.client.sync modbus_tcp.32 DEBUG New Transaction state 'SENDING'
2019-04-10 17:34:51,857 pymodbus.transaction modbus_tcp.32 DEBUG Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2019-04-10 17:34:51,897 pymodbus.transaction modbus_tcp.32 DEBUG Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2019-04-10 17:34:51,897 pymodbus.transaction modbus_tcp.32 DEBUG RECV: 0x0 0x1 0x0 0x0 0x0 0x5 0x0 0x0 0x0 0x0 0x0
2019-04-10 17:34:51,898 pymodbus.transaction modbus_tcp.32 DEBUG Getting transaction 1
2019-04-10 17:34:51,898 pymodbus.transaction modbus_tcp.32 DEBUG Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'

```

another isError() = True:

```python
2019-04-10 17:42:54,501 pymodbus.transaction modbus_tcp.32 DEBUG Current transaction state - TRANSACTION_COMPLETE
2019-04-10 17:42:54,501 pymodbus.transaction modbus_tcp.32 DEBUG Running transaction 37
2019-04-10 17:42:54,502 pymodbus.transaction modbus_tcp.32 DEBUG SEND: 0x0 0x25 0x0 0x0 0x0 0x6 0x2 0x3 0x1 0x74 0x0 0x2
2019-04-10 17:42:54,502 pymodbus.client.sync modbus_tcp.32 DEBUG New Transaction state 'SENDING'
2019-04-10 17:42:54,502 pymodbus.transaction modbus_tcp.32 DEBUG Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2019-04-10 17:42:54,527 pymodbus.transaction modbus_tcp.32 DEBUG Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2019-04-10 17:42:54,527 pymodbus.transaction modbus_tcp.32 DEBUG RECV: 0x0 0x25 0x0 0x0 0x0 0x6 0x81 0x20 0xd0 0x6a 0xf9 0x15
2019-04-10 17:42:54,527 pymodbus.transaction modbus_tcp.32 DEBUG Getting transaction 37
2019-04-10 17:42:54,527 pymodbus.transaction modbus_tcp.32 DEBUG Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'

```

Exception:

```python
2019-04-10 17:32:38,376 pymodbus.transaction modbus_tcp.32 DEBUG Current transaction state - IDLE
2019-04-10 17:32:38,377 pymodbus.transaction modbus_tcp.32 DEBUG Running transaction 1
2019-04-10 17:32:38,377 pymodbus.transaction modbus_tcp.32 DEBUG SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x2 0x3 0x0 0xf4 0x0 0x4
2019-04-10 17:32:38,377 pymodbus.client.sync modbus_tcp.32 DEBUG New Transaction state 'SENDING'
2019-04-10 17:32:38,378 pymodbus.transaction modbus_tcp.32 DEBUG Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2019-04-10 17:32:38,399 pymodbus.transaction modbus_tcp.32 DEBUG Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2019-04-10 17:32:38,399 pymodbus.transaction modbus_tcp.32 DEBUG RECV: 0x0 0x1 0x0 0x0 0x0 0x4 0x2 0x0 0x0 0x0
2019-04-10 17:32:38,400 pymodbus.factory modbus_tcp.32 ERROR Unable to decode response Modbus Error: Unknown response 0
2019-04-10 17:32:38,400 pymodbus.transaction modbus_tcp.32 ERROR Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/andrea/LWE/MPO/Myna/wolf/pymodbus/transaction.py", line 177, in execute
    request.unit_id)
  File "/home/andrea/LWE/MPO/Myna/wolf/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/home/andrea/LWE/MPO/Myna/wolf/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request

```

dhoomakethu, 2019-04-10:
Please share your client initialisation code .

wildstray, 2019-04-10:
```python
    def poll(self):
        requests = []
        typlen = {'b': 1, 'B': 1, 'h': 1, 'H': 1, 'i': 2, 'I': 2, 'q': 4, 'Q': 4, 'f': 2, 'd': 4}
        measures = []
        client = ModbusClient(host=self.address, retries=5, retry_on_empty=True, framer=ModbusSocketFramer)
        for row in self.regmap:
            (regname, descr, regtype, scale, addr) = row
            addr = int(addr)
            scale = float(scale)
            if addr > 40000:
                addr -= 40001
            result = client.read_holding_registers(addr, typlen[regtype], unit=self.slaveid)
            if result.isError():
                os._exit(255)
            if regtype == 'b':
                decoded = decoder.decode_8bit_int()
            if regtype == 'B':
                decoded = decoder.decode_8bit_uint()
            if regtype == 'h':
                decoded = decoder.decode_16bit_int()
            if regtype == 'H':
                decoded = decoder.decode_16bit_uint()
            if regtype == 'i':
                decoded = decoder.decode_32bit_int()
            if regtype == 'I':
                decoded = decoder.decode_32bit_uint()
            if regtype == 'q':
                decoded = decoder.decode_64bit_int()
            if regtype == 'Q':
                decoded = decoder.decode_64bit_uint()
            if regtype == 'f':
                decoded = decoder.decode_32bit_float()
            if regtype == 'd':
                decoded = decoder.decode_64bit_float()
            print (regname, descr, regtype, decoded)
            measures.append({'measure_id': regname, 'value': decoded * scale})
        client.close()
```

wildstray, 2019-04-11:
It seems to me the transaction is completed also with invalid data received, at least for socket (at the moment I'm not fitted for an RS485 serial test). NB: Qeed equipments, probably due to poor firmware, cause erratic readings also with another sw we use (DOG: Ontology-Powered OSGi Domotic Gateway). We cannot change equipment, so the only option is to deal with invalid respose and retry the transaction.

wildstray, 2019-04-12:
```
Not a valid unit id - 0, ignoring!!
```
Ok, I just see that there is a little validation (_validate_unit_id()) in the socket framer and this would be enoght to deal with out issue in most cases. But we need to retry the transaction when the unit id is wrong, not resetFrame() and just return an error. Any hints?

dhoomakethu, 2019-04-12:
We are trying to do something inline with the specification (emphasis on the `may` be part below) (refer section 2.4.1 from http://www.modbus.org/docs/Modbus_over_serial_line_V1_02.pdf)

> When a reply is received, the Master checks the reply before starting the data processing. The checking may result in an error,
> for example a reply from an unexpected slave, or an error in the received frame. In case of a reply received from an unexpected
> slave, the Response time-out is kept running. In case of an error detected on the frame, a retry **may** be performed. 

The simplest work around would be re-initiate the transaction.

closed 2022-05-25 by janiversen