pymodbus-dev/pymodbus #414: write_register pymodbus(not updating the value)

NagarjunaReddy1242, 2019-05-07:
hi,
I have connected one damper to modbus controller.when i was updated the value damper should move, for that i have used modbus write_register function.
address is 40065
i have mentioned below the following code

import pymodbus
import serial
from pymodbus.pdu import ModbusRequest
from pymodbus.client.sync import ModbusSerialClient as ModbusClient #initialize a serial RTU client instance
from pymodbus.transaction import ModbusRtuFramer
import time,os

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client= ModbusClient(method = "rtu", port="COM13",stopbits = 1, bytesize = 8, parity = 'N', baudrate=9600)
#connect to the serial modbus server
connection = client.connect()
message='hello'
print(message)
print connection
#Starting add, num of reg to read, slave unit.
#def hello():
while True:
    address=40065
    value=5
    result=client.write_register(0x0041,0x030,unit=0x01)
    print (result)
    time.sleep(2)
        
client.close()
o/p:
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:send: 0x1 0x6 0x0 0x41 0x0 0x30 0xd9 0xca
DEBUG:pymodbus.transaction:recv: 0x1 0x6 0x0 0x41 0x0 0x30 0xd9 0xca
DEBUG:pymodbus.factory:Factory Response[6]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 2
WriteRegisterResponse 65 => 48

problem:
value is not updating.

dhoomakethu, 2019-05-08:
You are receiving a valid response for your write register request, either you are writing to the wrong register or your slave is not able to update the register.

NagarjunaReddy1242, 2019-05-08:
Slave address is correct . It is working when i was used modscan tool.

dhoomakethu, 2019-05-09:
You will have to use the offset 0x40 and not 0x41, Offset 0x00 corresponds to register 40001, similarly 0x40(64) corresponds to register 40065.

NagarjunaReddy1242, 2019-05-09:
Thanks you for giving your response. It's working

closed 2019-05-09 by dhoomakethu