pymodbus-dev/pymodbus #420: Transaction failed. Connection reset by peer.

Sr4l, 2019-07-02:
#96 # Versions

* Python: 3.5
* OS: Ubuntu 16.04 and Windows 10 tested
* Pymodbus: 1.5.2 and 2.2.0
* Modbus Hardware (if used): ABB Freelance Controller

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

### Description
**What i've done:**
Write one program requesting registers every 100ms (data recorder).
Writer another program writing every 100ms a value (remote controller).

**What happend:**
If started on different PCs all works fine.

If started on one PC, different Python Interpreters.
The program started first works as expected.
The second started program crash's.
The error message differs in different pymodbus versions.

### Code and Logs
pymodbus 2.2.0
```python
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x0 0x0 0x0 0xa 0xff 0xf 0x1 0x90 0x0 0x16 0x3 0x10 0x10 0x3f
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Transaction failed. ([Errno 104] Connection reset by peer) 
DEBUG:pymodbus.framer.socket_framer:Processing: 
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Modbus Error: [Input/Output] [Errno 104] Connection reset by peer
```

pymodbus 1.5.2
```python
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x0 0x0 0x0 0xa 0xff 0xf 0x1 0x90 0x0 0x16 0x3 0x10 0x10 0x3f
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)) 
DEBUG:pymodbus.framer.socket_framer:Processing: 
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)
```

### Question and notes
Is this a known limitation?

I keep the connections in both programs open/alive. If I use one program for both task and share the connection it works. If I use two PCs, one for each program, it works.

dhoomakethu, 2019-07-22:
Apologies for the delayed response, where is the server running ? Also on the same PC ?

Sr4l, 2019-07-27:
Clients run on PC with versions and OSs mentioned above.

Server is running on a ABB Freelance AC700F (https://new.abb.com/control-systems/essential-automation/freelance/controller/ac700f)

dhoomakethu, 2019-07-28:
I would suggest to have both the operations running in the same script with locks to do only one operation at a time. The problem could be that the programs are receiving wrong responses which are not intended for them . It could also be the limitation on the server ad it could not handle the requests fast. Connection reset is generally initiated by the other party. Please try with a longer intervals for read and write and see how the setup behaves.

Sr4l, 2019-08-08:
I simply can't connect with a second process over the same IP only solution is to share the same connection.

I  tried other ModbusTCP implementation as well. As long as there is one process in connected state, not even requesting or sending coils or registers. The second process can connect, but gets a `broken pipe` or `connection reset by peer` if he tries to use the connection.

Seems to be a problem/limitation on the server side, hardly fixable in an embedded PLC.

Thanks you for your time.

closed 2019-08-08 by Sr4l