pymodbus-dev/pymodbus #421: "Incomplete message received" - Cannot read values of registers

HAMH96, 2019-07-02:
### Versions

* Python: 3.4.2
* OS: Debian (Raspberry 3B)
* Pymodbus: 2.2.0
* Modbus Hardware (if used): Multifunction Energy Meter YG889E-9SY

### Pymodbus Specific
* Server: Serial - Sync
* Client: RS485  - I don't know if sync or async
I use a RS485 to USB converter connected to raspberry 

### Description

What were you trying, what has happened, what went wrong, and what did you expect?
Hi, I am trying to read the measured variables from the energy meter, using modbus and a python 3 code, I am able to do the connection, but when I ask for a value of a register it gives me an error: 
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)

### Code and Logs

from pymodbus.client.sync import ModbusSerialClient as ModbusClient
#from pymodbus.register_read_message import ReadInputRegisterResponse
#The line above is commented because it gives me an error saying that it is not able to import that

client = ModbusClient(method='rtu',port='/dev/ttyUSB0',stopbits=1,bytesize=8,parity='N',baudrate=9600,timeout=0.3)

connection = client.connect()

print(connection)

value = client.read_input_registers(1,1)

#value = client.read_discrete_inputs(1,1)

#value = client.read_holding_registers(1,1)

#value = client.read_coils(1,1)

print(value)

The connection is done succesfully because it prints TRUE, but whatever read request I try gives me the same error. Also I tried with print(value.bits[0]) but it says that 'ModbusIOException' object has no attribute 'bits'. Please help, I am new in this.

dhoomakethu, 2019-07-03:
You are missing `unit` kwarg in your requests. The requests needs to be in the format  `value = client.read_input_registers(<offset>,<count>, unit=<unit-id>)`.

HAMH96, 2019-07-03:
> You are missing `unit` kwarg in your requests. The requests needs to be in the format `value = client.read_input_registers(<offset>,<count>, unit=<unit-id>)`.

What does "unit"means in that instruction? I don't understand so well

dhoomakethu, 2019-07-04:
The slave address you are trying to communicate.

ಗುರು, ಜುಲೈ 4, 2019 01:22 ದಿನಾಂಕದಂದು HAMH96 <notifications@github.com> ಅವರು
ಬರೆದಿದ್ದಾರೆ:

cheryjose, 2020-03-26:
@HAMH96 did you find a solution for this problem, I am having this problem now

ietamaher, 2021-01-19:
@HAMH96 did you find a solution for this problem? I am also having this problem.

closed 2021-03-08 by github-actions[bot]