pymodbus-dev/pymodbus #429: Updating Server Problem

pjrimmer, 2019-08-02:
Hello all, hopefully someone can help me please!

I have taken the following updating server example and adapted it for my needs:
https://pymodbus.readthedocs.io/en/latest/source/example/updating_server.html

My setup is a PLC running as a Modbus RTU master reading data from a Modbus RTU slave device. In between these nodes I have two Raspberry Pi's both with 2 x USB to RS485 adapters, which I am trying to get working as Store & Forward or Gateway devices. So each Pi acts as a Slave on one port and a Master on the other port.

I can get them working as repeaters using remote contexts, however, I want them to be able to store the data so I can manipulate the values later - this is for research into Modbus security.

I have identical code (shown below) running on both of the Pi's. The 2nd or remote Pi (closest to the slave device) is working fine. It reads the 8 holding registers and updates the slave context as expected.

However, the 1st or local Pi (closest to the master device) works for a certain amount of time but then fails with the error shown in the log section below. Once this error has occurred, it no longer even attempts to read data from the 2nd Pi. Hence it just keeps responding with static data to the PLC Master device.

Any help greatly appreciated.

Many thanks!

### Versions

* Python: 2.7.16
* OS: Raspbian GNU/Linux 10 (buster)
* Pymodbus: 2.2.0
* Modbus Hardware (if used): Raspberry Pi's & FTDI FT232RL USB to RS485 Adapters

### Log

Unhandled error in Deferred:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/base.py", line 1272, in run
    self.mainLoop()
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/base.py", line 1281, in mainLoop
    self.runUntilCurrent()
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/base.py", line 902, in runUntilCurrent
    call.func(*call.args, **call.kw)
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/task.py", line 239, in __call__
    d = defer.maybeDeferred(self.f, *self.a, **self.kw)
--- <exception caught here> ---
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/defer.py", line 151, in maybeDeferred
    result = f(*args, **kw)
  File "./updating_server.py", line 27, in updating_writer
    values = rr.registers
exceptions.AttributeError: 'ModbusIOException' object has no attribute 'registers'

### Code

#!/usr/bin/env python

# import libraries
from pymodbus.server.asynchronous import StartSerialServer
from pymodbus.client.sync import ModbusSerialClient
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.transaction import ModbusRtuFramer
from twisted.internet.task import LoopingCall

# configure the service logging
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

# define update process
def updating_writer(a):
    log.debug("updating the context")
    client = ModbusSerialClient(method='rtu', port='/dev/ttyUSB1', timeout=5)
    client.connect()
    rr = client.read_holding_registers(1, 8, unit=0x16)
    context = a[0]
    fc = 3
    slave_id = 0x16
    address = 0x1
    values = rr.registers
    log.debug("new values: " + str(values))
    context[slave_id].setValues(fc, address, values)
    client.close()

def run_updating_server():
    # initialize the data store
    store = ModbusSlaveContext(hr=ModbusSequentialDataBlock(1, [0]*8), zero_mode=False)
    context = ModbusServerContext(slaves=store, single=True)

    # run the server you want
    time = 5
    loop = LoopingCall(f=updating_writer, a=(context,))
    loop.start(time, now=False)
    StartSerialServer(context, port='/dev/ttyUSB0', framer=ModbusRtuFramer)

if __name__ == "__main__":
    run_updating_server()

pjrimmer, 2019-08-02:
Additional information:

I can reproduce the problem by using just a simple modbus master script on the local pi to poll the remote pi. Sometimes it works and sometimes it fails. It is almost as if the remote pi is too busy to respond so the request times out.

I have tried increasing the timeout period for the _ModbusSerialClient_ but this then seems to cause problems with updating the context.

Is inside _def updating_writer(a):_ the write place for the _ModbusSerialClient_ request?

pjrimmer, 2019-08-02:
I have done some more tests and it is definitely an issue with the remote pi not being able to respond to the read requests while it is updating its own slave context values.

I know this because as a test I changed the remote pi to just incrementing the values of the slave context as per the original example and the problem still occurs.

Any help on this would be greatly appreciated.

tracernz, 2019-08-02:
The synchronous client calls will block the event loop right?

pjrimmer, 2019-08-03:
Thanks for the reply. So should I try an asynchronous client instead then? I've tried but not had much luck with them so far!

closed 2021-03-08 by github-actions[bot]