pymodbus-dev/pymodbus #430: Modbus request Queue feature

srikanthsrnvs, 2019-08-08:
### Description

I'm working with a Modbus motor, and using registers to set it's position synchronously. Writing data to the register results in a Response before the motor has completed it's movements, and so multiple calls to move the motor, result in erratic behaviour. Here's some pseudo code below

### Code and Logs

```python
movement1 = motor.move(10) # using client.write_single_register()
movement2 = motor.move(1)   # using client.write_single_register()

```

This results in the motor only moving 1 unit, since it received a new command before the movement actually occurred. 
Since these commands are going to be hooked up to a tensor flow model, there will likely be many hundreds of commands being input, and so I'm stumped on how to solve this.

Any advice/idea would help a lot!

dhoomakethu, 2019-08-08:
You can wait for the response to arrive before going with the next
transaction. Also try with different read timeouts so that you are not
sending the next request before the motor can process the request and send
back the response.

srikanthsrnvs, 2019-08-08:
@dhoomakethu The motor does not send a response after it completes the movement. It sends only an acknowledgement that it has received an update to a register.

srikanthsrnvs, 2019-08-15:
Anyone know how I can solve this?

tracernz, 2019-08-29:
I can see two ways:
- poll the motor controller for it's status and/or position
- wait at least long enough for the requested move to complete

The first is much less error prone.

srikanthsrnvs, 2019-09-02:
> * poll the motor controller for it's status and/or position

This was the solution. I implemented a priority queue where polling has a smaller priority than a command, and ran the queue in a loop.

closed 2019-09-02 by srikanthsrnvs