pymodbus-dev/pymodbus #432: Check Modbus port before reading

muzzammilhussain, 2019-08-24:
I am using Python 2.7 with pymodbus on rasberry 3 , I am trying to read the logs from one of the solar inverters by Growatt Company I am having a very strange issue , The inverter has two ports RS485 and rs232 , on rs485 a smart energy meter is working with modbus , on the other port Growatt own monitoring device was connected using Modbus I removed it and attachted a rasberry with pymodbus , now having issue that sometimes they both read together then inverter goes into fault , I want to do something like , growatt device is doing like check if the modbus is being read or something , I cant figure out the solution

dhoomakethu, 2019-09-17:
If you can control the readings of both device from the same script, use `locks` to restrict access to the bus when other is reading.

dexterhussain, 2019-09-19:
yes its done thanks :)

closed 2020-07-04 by dhoomakethu