pymodbus-dev/pymodbus #433: ModuleNotFoundError: No module named 'modbus_mapper'

krishnacy, 2019-08-26:
Hi there, 

I am new to python and I am trying to write a small code to read modbus registers from a csv file with the help of https://github.com/riptideio/pymodbus/blob/master/examples/contrib/modbus_mapper.py
However, I'm getting the following error, please help me sort this issue.

I'm trying to import the modbus_mapper as follows:
from pymodbus.client.sync import ModbusTcpClient
from modbus_mapper import csv_mapping_parser

Error I'm receiving:
 RESTART: C:/Users/AppData/Local/Programs/Python/Python37-32/mod1.py 
Traceback (most recent call last):
  File "C:/Users/AppData/Local/Programs/Python/Python37-32/mod1.py", line 7, in <module>
    from modbus_mapper import csv_mapping_parser
ModuleNotFoundError: No module named 'modbus_mapper'

### Versions

Name: pymodbus
Version: 2.2.0
Summary: A fully featured modbus protocol stack in python
Home-page: https://github.com/riptideio/pymodbus/
Author: Galen Collins
Author-email: bashwork@gmail.com
License: BSD
Location: c:\users\appdata\local\programs\python\python37-32\lib\site-packages
Requires: six, pyserial
Required-by:
OS : Windows 10

### Pymodbus Specific
* Client: tcp

Thanks!!

dhoomakethu, 2019-08-26:
The examples in `contrib` folders are quiet outdated and are prone to be incompatible with latest pymodbus but in your case it looks like you are having a [PYTHONPATH](https://stackoverflow.com/questions/3701646/how-to-add-to-the-pythonpath-in-windows-so-it-finds-my-modules-packages) issue. Please make sure you set the `PYTHONPATH` correctly before running your program.

closed 2019-08-26 by dhoomakethu

krishnacy, 2019-09-25:
Hi @dhoomakethu 

Sorry for the late reply, Thank you for your support with the path issue, i am importing the def from the modbus_mapper and facing the following issue in the following function 

template = ['addr', 'fun', 'type', 'stat']
def mapping_decoder(mapping_blocks, decoder=None):

    for block in mapping_blocks.values():
        for mapping in block.values():
            print(mapping)
            print(block.values())
            mapping['addr'] = int(mapping['addr'])

Traceback (most recent call last):
  File "C:\Users\AppData\Local\Programs\Python\Python37-32\lap.py", line 311, in mapping_decoder mapping['addr'] = int(mapping['addr'])
TypeError: string indices must be integers

when printing the print(mapping) & print(block.values()) i am receiving the following output :
40001
dict_values(['40001', 'int', 'Pac'])

Please suggest how to solve this issue. Also, can you please update the modbus_mapper file which is compatible with the latest pymodbus version.