pymodbus-dev/pymodbus #434: Parity issues in the Async Asyncio Client

hackerboygn, 2019-08-29:
### Versions

* Python:3.7.4
* OS:Windows 10
* Pymodbus:2.2.0  （pip install）
* Modbus Hardware (if used): None

### Pymodbus Specific
* Client: rtu- async

### Description

I follow the manual test 'Async Asyncio Client Example', and I want change parity to even but don't work,  I use the monitoring software found out the parity always None. I think we forgot to pass this parameters to pyserial.

### Code and Logs

```python
 # code and logs here.
loop, client = ModbusClient(schedulers.ASYNC_IO, port='COM3',
                                baudrate=38400, parity='E', method="rtu")
```

hackerboygn, 2019-08-29:
#389

hackerboygn, 2019-08-30:
#436

dhoomakethu, 2019-09-17:
Closed with #436

closed 2019-09-17 by dhoomakethu