pymodbus-dev/pymodbus #437: 'ModbusIOException' object has no attribute 'registers'

Howeng98, 2019-09-02:
### Versions

* Python: 3.7
* OS: Windows10
* Pymodbus: 1.3.1
* Pymodbus3: 1.0.0
* Modbus Hardware (if used): 

### Description

I want to use Modbus to execute JY-CZ1000-GN.  And I use the code as below, and It can't worked. The point is it can worked last time (a few weeks ago),but now cannot.
I wonder that maybe it is not the software(code) problem, should be handware problem, I guess it is the wire didn't connect to ground ,so when we try to read register will received some circuit noise(?)

### Code and Logs
Code: 
 from pymodbus.client.sync import ModbusSerialClient as ModbusClient

scale = ModbusClient(method='rtu', port='/dev/ttyUSB0', parity='N', baudrate=9600, bytesize=8, stopbits=2, timeout=1)
connection = scale.connect()

try:
    result = scale.read_holding_registers(0,10, unit=1)
    print(result)
    print(result.registers)
except Exception as e:
    print(e)

scale.close()

Error Message:
Modbus Error: [Input/Output] No Response received from the remote unit
'ModbusIOException' object has no attribute 'registers'

dhoomakethu, 2019-09-02:
Try using the kwarg `strict=False` when initializing the client on windows environment.
`scale = ModbusClient(method='rtu', port='/dev/ttyUSB0', parity='N', baudrate=9600, bytesize=8, stopbits=2, timeout=1, strict=False)`. Please share the logs if you are still having the issues.

Howeng98, 2019-09-02:
@dhoomakethu 
Hi sir, sorry for typo , I am using Linux for running this code
And I had tried to add 'strict' into that line, but the error message still same as before

Error Msg:
Modbus Error: [Input/Output] No Response received from the remote unit
'ModbusIOException' object has no attribute 'registers'

I also had tried to use time.sleep(1) to make it delay for received complete data message,or raise Timeout be higher, but still also will get the error

dhoomakethu, 2019-09-02:
Please use the latest version of pymodbus (2.2.0) and see if it solves the issue. The issue could also be due to wrong serial settings. Please share the logs for further analysis.

Howeng98, 2019-09-03:
@dhoomakethu 
Thanks for you suggestion sir, I have try to use pymodbus (2.2.0) version but it still can't work. 

Log:
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x10 0x4 0x0 0x0 0x0 0xa 0x73 0x4c
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)) 
DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
DEBUG:pymodbus.transaction:Getting transaction 16
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
odbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)
'ModbusIOException' object has no attribute 'registers'

I will try another way to solve it

tompropst, 2019-10-12:
I am seeing this same error. In my case, I'm connecting over TCP and most the register reads succeed. Only about once a day (or less) do I see this error while reading every 10 seconds. 

From the log output above, it looks like the incomplete message should be [raising an exception](https://github.com/riptideio/pymodbus/blob/master/pymodbus/transaction.py#L260) but perhaps it is being ignored someplace so the actual exception occurs when trying to access the registers of the return object.

tracernz, 2019-10-12:
Those exceptions are caught and returned by functions higher up. As you can see in @Howeng98 reply above:

> 'ModbusIOException' object has no attribute 'registers'

The way error handling currently works in the public pymodbus API is exception objects are returned, not raised as exceptions. See the comment by @dhoomakethu https://github.com/riptideio/pymodbus/issues/298#issuecomment-386175043, and that issue for wider context.

AnkitaPadma, 2020-06-29:
I m facing the same issue. Any solution was found?

dhoomakethu, 2020-07-04:
use `result.isError()` to validate if the result is a valid response or an error response.

closed 2020-07-04 by dhoomakethu

AnkitaPadma, 2020-07-06:
@dhoomakethu  So to resolve the 'ModbusIOException' object has no attribute 'registers' i try resending the request multiple times till i receive a proper response.
Currently using a while loop.
Is there a service in pymodbus doing the same?

dhoomakethu, 2020-07-06:
@AnkitaPadma I am afraid that's the way to go. In general, unless the communication is  not stable the slaves should/would return the positive response. You can also check what kind of error is being returned to debug further.

jsr1611, 2020-07-14:
Hi folks,

I do face a similar issue.

I receive data but sometimes, it gives me the same error. Here is my print: 

Count	 ID	 Temp	 Humidity	 Part03	 Part05	 DateTime
...
5 		 4	 23.28 	 49.51 		 6631 	 759 	 2020-07-14 11:22:19.464
6 		 1	 24.4 	 46.74 		 9259 	 848 	 2020-07-14 11:22:20.231
6 		 3	 24.32 	 47.02 		 7334 	 1492 	 2020-07-14 11:22:20.631
6 		 4	 23.26 	 49.52 		 6641 	 760 	 2020-07-14 11:22:21.031
'ModbusIOException' object has no attribute 'registers'
8 		 1	 24.38 	 46.7 		 9279 	 844 	 2020-07-14 11:22:26.627
8 		 2	 24.73 	 45.81 		 8243 	 867 	 2020-07-14 11:22:26.627
8 		 3	 24.3 	 46.97 		 7293 	 1474 	 2020-07-14 11:22:27.251
8 		 4	 23.26 	 49.55 		 6699 	 763 	 2020-07-14 11:22:27.731
9 		 1	 24.42 	 46.7 		 9287 	 842 	 2020-07-14 11:22:28.402
9 		 2	 24.73 	 45.77 		 8252 	 868 	 2020-07-14 11:22:28.403
9 		 3	 24.34 	 46.95 		 7293 	 1465 	 2020-07-14 11:22:29.026
9 		 4	 23.25 	 49.52 		 6704 	 764 	 2020-07-14 11:22:29.282
10 		 1	 24.43 	 46.73 		 9294 	 841 	 2020-07-14 11:22:30.161
'ModbusIOException' object has no attribute 'registers'
11 		 1	 24.4 	 46.75 		 9329 	 838 	 2020-07-14 11:22:35.453
11 		 2	 24.75 	 45.78 		 8276 	 870 	 2020-07-14 11:22:35.453
11 		 3	 24.32 	 47.0 		 7362 	 1452 	 2020-07-14 11:22:35.853
11 		 4	 23.25 	 49.55 		 6720 	 763 	 2020-07-14 11:22:36.413
...

I am using it like this:
sensor1 = [modbusClient.read_input_registers(address=10, count=6, unit=1)]
register address=10,
count = 6 (reading all data I need at once stored as an array)
slave address = 1,

And, as it gives me the above error message it is missing in between sensor data.

copyCat304, 2020-09-11:
I found via trial and error that there should be atleast 20sec delay between register read or write call. 

from pymodbus.client.sync import ModbusSerialClient as ModbusClient
from pymodbus.constants import Endian
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.payload import BinaryPayloadBuilder
import time
print("import ok")
client = ModbusClient(method='rtu', bytesize = 8, baudrate = 9600,port='COM8', timeout=3, strict=False)
while(True):
    if client.connect():
        print("***********connected************")
        # connection is OK
        # read floats
        result  = client.read_holding_registers(0, 4, unit=1)
        print(result.registers)
        decoder = BinaryPayloadDecoder.fromRegisters(result.registers, Endian.Big, wordorder=Endian.Little)
        print("reslut01: " + str(decoder.decode_32bit_float()))
        time.sleep(20)
        result1  = client.read_holding_registers(20, 4, unit=1)
        print(result1.registers)
        decoder = BinaryPayloadDecoder.fromRegisters(result1.registers, Endian.Big, wordorder=Endian.Little)
        print("reslut02: " + str(decoder.decode_32bit_float()))
        time.sleep(20)
        result2  = client.read_holding_registers(8, 2, unit=1)
        print(result2.registers)
        decoder = BinaryPayloadDecoder.fromRegisters(result2.registers, Endian.Big, wordorder=Endian.Little)
        print("result03: " + str(decoder.decode_32bit_float()))
        print("********************************")
        time.sleep(20)
        client.close()

I hope this helps and please let me know if there is any efficient way of doing this task I am kinda a newbie.

obaidullahaamer, 2020-10-20:
Getting same issue. I tried Adding 20 sec delay and it worked. But is there any efficient way to solve this issue. That would make my life a lot easier.

dhoomakethu, 2020-10-20:
@obaidullahaamer it all boils down to how fast the modbus slave is responding and how many registeres are being queried. Ideally the client should not have to wait for 20sec to get the values. What is the model and make of your slave?

obaidullahaamer, 2020-10-20:
@dhoomakethu Iam using EM737CT (MetersUK) and trying to get 37 values from registers. When using 20s delay, it slows down the program considerably.

gabriel00mainka, 2021-04-08:
Have enyone the final code?

gabriel00mainka, 2021-04-08:
@jsr1611 or @Howeng98

sahilahluwalia, 2021-04-17:
Here is my code

from pymodbus.client.sync import ModbusTcpClient
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.datastore import ModbusSequentialDataBlock,ModbusSlaveContext,ModbusSparseDataBlock
client = ModbusClient('192.168.0.121',port=502,timeout=3)
for count in range(1):
    while True:
        try:
            results=client.read_input_registers(122)
            print(results.registers)    
        except results.isError():
            print("ERROR")
            continue
        break
    
i am getting this same error on my first try, if i am doing loop of more than 1 for example 100 than for the first time i will get 'ModbusIOException' object has no attribute 'registers' error than 99 times i will get value of register.

:( any way to overpass first try?

dhoomakethu, 2021-04-18:
@sahilahluwalia what version of pymodbus are you using and what is you modbus slave (real hardware or simulated device?)

mEgha061295, 2022-02-14:
Hi, here is my code :
import minimalmodbus
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
from pymodbus.register_read_message import ReadInputRegistersResponse

client = ModbusClient(method= 'rtu', port= '/dev/serial0', stopbits=1, bytesize=8, parity= 'N', baudrate= 57600, timeout = 0.9, strict = False)

connection = client.connect()
print(connection)

#value = client.read_input_registers(300, 4, unit= 0x01)
#if not value.isError():
#print(value.registers)

result = client.read_holding_registers(0, 4, unit=1)
print(result.registers)
decoder = BinaryPayloadDecoder.fromRegisters(result.registers, Endian.Big, wordorder=Endian.Little)
print("reslut01: " + str(decoder.decode_32bit_float()))
time.sleep(20)
result1 = client.read_holding_registers(20, 4, unit=1)
print(result1.registers)
decoder = BinaryPayloadDecoder.fromRegisters(result1.registers, Endian.Big, wordorder=Endian.Little)
print("reslut02: " + str(decoder.decode_32bit_float()))
time.sleep(20)
result2 = client.read_holding_registers(8, 2, unit=1)
print(result2.registers)
decoder = BinaryPayloadDecoder.fromRegisters(result2.registers, Endian.Big, wordorder=Endian.Little)
print("result03: " + str(decoder.decode_32bit_float()))
print("********************************")
time.sleep(20)
client.close()

I am getting the same error and dont know how to resolve it. Please help!!!!

Ansari28, 2022-07-30:
> Hi, here is my code : import minimalmodbus from pymodbus.client.sync import ModbusSerialClient as ModbusClient from pymodbus.register_read_message import ReadInputRegistersResponse
>
> client = [...]

Hey i am also facing same error. did you solve? how? please tell me.

janiversen, 2022-07-30:
There are a couple of problems here.

When the app RECEIVES modbus<something>exception it is a message coming from the device, which the app needs to handle. This is clearly not the expected response but an error message so it will not contain .registers, .bit etc.

When Pymodbus encounters a problem it raises modbus<something>exception, which the app needs to deal with in a try/except statement.

The simplest approach in both cases is to restart the connection (and look for the root cause).

20second delay have nothing to do in a real world. In the Homeassistant modbus integration we do a read a few milliseconds after a write to verify. Apart from that integration have been tested with around 1.000 registers which were continuously written and read, without problems. HA modbus uses Pymodbus v2.5.3 as well as dev.

If the app writes a register to set a temperature, it normally will take quite long time, before reading (actual) temperature will change....maybe that is the confusion.

When you get ModbusIOEception there are data in the exception that may hint on why there was a problem.

TQK0209, 2022-11-28:
here is my code:

from pymodbus.client.sync import ModbusSerialClient as ModbusClient 
import time
client = ModbusClient(method = 'rtu', port='COM3', stopbits = 1, bytesize = 8, parity = 'O' , baudrate= 115200,timeout=0.01)
connection = client.connect()

print(connection)
print(client)

print("---------------------------------------------")

result = client.read_holding_registers(3,1, unit=1)
    # print(result)
    # result = read = client.read_holding_registers(200,1)
print(result)

Error Message:
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 2 bytes (0 received)

help me

janiversen, 2022-11-28:
Seems you are using an old version, please use version 3.0.2

Without a debug log as pr documentation we cannot see what is happening and thus not help you.

I see some posibilítes:
- your client never successfully connect.
- your unit is not correct and thus get no response
- the address is not correct and should receive exception from device

janiversen, 2022-11-28:
And please do not start a new issue on another issue.

TQK0209, 2022-11-28:
I am updating version 3.11 for python to install the latest version of pymodbus

I'll post the results as soon as I'm done

janiversen, 2022-11-28:
Please do not continue on this issue, we normally do not watch closed issues, I saw this by accident.

TQK0209, 2022-11-28:
So where can I exchange it?

plss

TQK0209, 2022-11-28:
So where can I tell?

janiversen, 2022-11-28:
Open a new issue if there are a problem with pymodbus !! or if you have questions open a discussion.

nusgnojkrap, 2023-03-22:
client = ModbusClient(method='rtu', port='/dev/ttyUSB0', baudrate=9600, bytesize=8, stopbits=1, timeout=3)
client.connect()
rr = client.read_holding_registers(200,2,unit=6)
print(rr) # Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
print(rr.registers) # AttributeError: 'ModbusIOException' object has no attribute 'registers'

I don't know why the result of the read_holding_registers function appears as an error.

Help me.

janiversen, 2023-03-22:
WHy do you not know ? The text tells you directly you do not receive a response from your device.

A no response, can be simple as you use a wrong slave, wrong parameters for your USB, to the more unlikely that you use a wrong framer.

I do not see a problem in pymodbus.

janiversen, 2023-03-22:
Please be aware that you are writing on a closed issue, I only saw it pr accident, we do not watch closed issues.

janiversen, 2023-03-22:
you do not write which version you use, but in the newer versions, method= do not exist (please see the documentation).