pymodbus-dev/pymodbus #438: Modbus TCP/IP ReadHoldingRegisters - decoding error

jaroslaw-wieczorek, 2019-09-05:
### Versions

* Python: 3.7
* OS: Linux Manjaro
* Pymodbus: 2.20

### Description
I met problem with decoding requests and response messages with function_code == 3 Holding Registers: 
List of example packages - first package is ReadHoldingRegistersResponse and is decoded correctly. 
But another packets always return error message:  'unpack requires, a buffer of 4 bytes'
(when i try decode them as request) and 'unpack requires a buffer of 2 bytes' when i try decode them as response.

All messages contains 13 bytes.
      pkts = [
            b'\xa2\xae\x00\x00\x00\x07\x00\x03\x04\x00\x06\x00x', # good response message
            b'\xa2\xaf\x00\x00\x00\x07\x00\x03\x0c\x1c\x00\x10\xc5',
            b'\xa2\xb0\x00\x00\x00\x07\x00\x03\x0c,\x00\x10\xb5',
            b'\xa2\xb1\x00\x00\x00\x07\x00\x03\x0c<\x00\x10\xa5',
            b'\xa2\xb2\x00\x00\x00\x07\x00\x03\x0cL\x00\x10\x95',
            b'\xa2\xb3\x00\x00\x00\x07\x00\x03\x0c\\\x00\x10\x85',
            b'\xa2\xb4\x00\x00\x00\x07\x00\x03\x0cl\x00\x10u',
            b'\xa2\xb5\x00\x00\x00\x07\x00\x03\x0c|\x00\x06o',
            b'\xa2\xb6\x00\x00\x00\x07\x00\x03\x0b\xb8\x00\x028',
            b'\xa2\xb7\x00\x00\x00\x07\x00\x03\x0c\x1c\x00\x10\xc5',
            b'\xa2\xb8\x00\x00\x00\x07\x00\x03\x0c,\x00\x10\xb5',
            b'\xa2\xb9\x00\x00\x00\x07\x00\x03\x0c<\x00\x10\xa5',
            b'\xa2\xba\x00\x00\x00\x07\x00\x03\x0cL\x00\x10\x95',
            b'\xa2\xbb\x00\x00\x00\x07\x00\x03\x0c\\\x00\x10\x85',
            b'\xa2\xbc\x00\x00\x00\x07\x00\x03\x0cl\x00\x10u',
            b'\xa2\xbd\x00\x00\x00\x07\x00\x03\x0c|\x00\x06o',
            b'\xa2\xbe\x00\x00\x00\x07\x00\x03\x0b\xb8\x00\x028',
            b'\xa2\xbf\x00\x00\x00\x07\x00\x03\x0b\xb8\x00\x028',
            b'\xa2\xc0\x00\x00\x00\x07\x00\x03\x0c\x1c\x00\x10\xc5',
            b'\xa2\xc1\x00\x00\x00\x07\x00\x03\x0c,\x00\x10\xb5',
            b'\xa2\xc4\x00\x00\x00\x07\x00\x03\x0c\\\x00\x10\x85',
            b'\xa2\xc5\x00\x00\x00\x07\x00\x03\x0cl\x00\x10u',
            b'\xa2\xc6\x00\x00\x00\x07\x00\x03\x0c|\x00\x06o',
            b'\xa2\xc8\x00\x00\x00\x07\x00\x03\x0c\x1c\x00\x10\xc5',
            b'\xa2\xc9\x00\x00\x00\x07\x00\x03\x0c,\x00\x10\xb5']

```python

class ReadRegistersRequestBase(ModbusRequest):
	'''
	Base class for reading a modbus register
	'''
	_rtu_frame_size = 8
	
	def __init__(self, address, count, **kwargs):
		''' Initializes a new instance

		:param address: The address to start the read from
		:param count: The number of registers to read
		'''
		ModbusRequest.__init__(self, **kwargs)
		self.address = address
		self.count = count
	
	def encode(self):
		''' Encodes the request packet

		:return: The encoded packet
		'''
		return struct.pack('>HH', self.address, self.count)
	
	def decode(self, data):
		''' Decode a register request packet

		:param data: The request to decode
		'''
		self.address, self.count = struct.unpack('>HH', data)
		# in this place struct return error. 

```

dhoomakethu, 2019-09-17:
@jaroslaw-wieczorek  Sorry for the delayed response. Could you provide a complete debug log for this issue ?

jaroslaw-wieczorek, 2019-09-26:
I use kaitai struct library to extract Modbus TCP/IP from payload TCP/IP.
My decoder based on this example: https://pymodbus.readthedocs.io/en/v1.3.2/examples/message-parser.html
Problem is with decoding message because some devices send one byte more - something like 1 byte flag on the end of message.

During debug I noticed that ReadRegistersRequestBase in "decode" method get 5 bytes more than expected. So message cannot be decoded like a request and response. 

The same message in Wireshark or http://packetor.com/ is decoded correctly.

dhoomakethu, 2019-09-26:
That is interesting to know that you can still decode the message in wireshark . Could you please provide a screenshot of wireshark or the data you are feeding to pocketor for me to verify ?

jaroslaw-wieczorek, 2019-09-26:
[image]

[image]

janiversen, 2022-06-30:
Cannot be reproduced with 3.0dev4.

closed 2022-06-30 by janiversen