pymodbus-dev/pymodbus #443: Connection with a converter in the middle?

saanchezfer, 2019-09-11:
* Python: 3.7
* OS: Windows
* Pymodbus: Last
* Modbus Hardware (if used):  Cirwatt 410-vt7a-90b-10

### Description
Hi! I'm an internship in the Electric Company of my city. They told us to read some three-fase meters that have RS232 ports. We can read them very easy, but the idea is to have them far away from here, so they put [this](https://www.usriot.com/products/rs232-to-ethernet-converter.html) converter to have an ethernet output and put and antenna there. We configured it, but it's not possible to connect to the device using the ip that the converter have, because the meter only answers via RTU. The only way is emulating a serial port using the ip that the converter have. 
There is any way to connect to it in the code without emulating it with the software that USR gave to us?

dhoomakethu, 2019-09-17:
If the device can support Modbus TCP and if you can get a public IP assigned to it or if the device is accessible from with in your network, you can use pymodbus to read data over ModbusTCP. Otherwise it is not possible. You can take a look at https://github.com/riptideio/pymodbus/blob/master/examples/contrib/serial_forwarder.py. May be that is something you are looking for.

closed 2019-09-17 by dhoomakethu

saanchezfer, 2019-09-17:
No, the device only throws info through RTU, but we have to arrive to it through a converter because it will be far away from the computer.