pymodbus-dev/pymodbus #453: intermittent exceptions.ValueError: invalid literal for int() with base 16: ''

sherpya, 2019-10-10:
I'm trying to implement a simple client/server using twisted, I've made a connection between two serial ports adapters (ttyUSB0 and ttyUSB1).

server code:
```python
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.framer.ascii_framer import ModbusAsciiFramer
from pymodbus.server.asynchronous import StartSerialServer
from twisted.internet import reactor

def main():
    store = ModbusSlaveContext()
    context = ModbusServerContext(slaves=store, single=True)

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/bashwork/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = '2.2.0'

    StartSerialServer(context, identity=identity,
                      port='/dev/ttyUSB0', baudrate=9600,
                      framer=ModbusAsciiFramer, defer_reactor_run=True)

    reactor.run()

if __name__ == '__main__':
    main()
```

client code:

```python
from pymodbus.client.asynchronous.serial import AsyncModbusSerialClient
from pymodbus.client.asynchronous import schedulers
from pymodbus.client.asynchronous.twisted import ModbusSerClientProtocol
from pymodbus.constants import Endian
from pymodbus.payload import BinaryPayloadDecoder, BinaryPayloadBuilder
from twisted.internet import task
from twisted.internet.defer import inlineCallbacks

@inlineCallbacks
def main(reactor):
    _, client = AsyncModbusSerialClient(schedulers.REACTOR,
                                        method='ascii',
                                        port='/dev/ttyUSB1',
                                        baudrate=9600,
                                        proto_cls=ModbusSerClientProtocol)

    client = client.protocol

    string = 'abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ'

    builder = BinaryPayloadBuilder(byteorder=Endian.Big, wordorder=Endian.Little)
    builder.add_string(string)

    address = 0
    registers = builder.to_registers()
    print(registers)

    yield client.write_registers(address, registers, unit=1)

    result = yield client.read_holding_registers(address, len(registers))
    decoder = BinaryPayloadDecoder.fromRegisters(result.registers, byteorder=Endian.Big, wordorder=Endian.Little)
    decoded = decoder.decode_string(len(string))
    print(decoded)

    reactor.stop()

if __name__ == '__main__':
    task.react(main)
```

The code works intermittently, the server is launched and never stopped, the client is launched multiple times, unfortunately the error is into non user-catchable code.

There is something wrong in the design? I suppose an empty read can happen anyway in `dataReceived` so I wouldn't do a blind decode_data

dhoomakethu, 2019-10-21:
@sherpya do you have a traceback and logs at the time of error?

sherpya, 2019-10-22:
@dhoomakethu it is a defective usb/serial converter, the problem is here:
```
  File "/home/sherpya/venv/local/lib/python2.7/site-packages/twisted/internet/posixbase.py", line 597, in _doReadOrWrite
    why = selectable.doRead()
  File "/home/sherpya/venv/local/lib/python2.7/site-packages/twisted/internet/_posixserialport.py", line 61, in doRead
    return fdesc.readFromFD(self.fileno(), self.protocol.dataReceived)
  File "/home/sherpya/venv/local/lib/python2.7/site-packages/twisted/internet/fdesc.py", line 94, in readFromFD
    callback(output)
  File "/home/sherpya/venv/local/lib/python2.7/site-packages/pymodbus/client/asynchronous/twisted/__init__.py", line 101, in dataReceived
```

(I've manually added the exception in the same place), somehow data is an empty string here,
I suppose it is not so common, but it definitively can happen

closed 2021-03-08 by github-actions[bot]