pymodbus-dev/pymodbus #458: How dose asyncio client recover from server restart

simplezhao, 2019-10-21:
### Versions

* Python: 3.7
* OS: macOS
* Pymodbus: pymodbus-2.3.0rc1
* Modbus Hardware (if used): 

### Pymodbus Specific
* Client: tcpasync

### Description

What were you trying, what has happened, what went wrong, and what did you expect?
I want to use asyncio to run pymodbus client, Normally it is working. But when I stop the Modbus server and restart the Modbus server. My problem is :
the client tries to connect the server, but can't get anything from read_holding_registers any more.
### Code and Logs

```python
UNIT = 0x1
_, client = ModbusClient(schedulers.ASYNC_IO, port=5020, loop=server.loop)
# modbus task
async def modbus_task(app):
    logging.info('task created')
    while True:
        try:
            rr = client.protocol.read_holding_registers(1, 8, unit=UNIT)
            rr = await asyncio.wait_for(rr, timeout=3)
            print('get tables')
            print(app['tables'])
            # rr = client.read_holding_registers(1, 8, unit=UNIT)
            # logging.info(rr)
        except asyncio.TimeoutError:
            logging.debug('timeout!!')
        await asyncio.sleep(1)
        # print('test loop 2')
async def create_modbus_task(app):
    loop = asyncio.get_event_loop()
    task = loop.create_task(modbus_task(app))
    app['modbus'] = task

```
the log below:
```
DEBUG:pymodbus.transaction:Adding transaction 7
DEBUG:pymodbus.client.asynchronous.asyncio:recv: 0x0 0x7 0x0 0x0 0x0 0x13 0x1 0x3 0x10 0x0 0x2 0x0 0x3 0x0 0x4 0x0 0x5 0x0 0x6 0x0 0x7 0x0 0x8 0x0 0x9
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x7 0x0 0x0 0x0 0x13 0x1 0x3 0x10 0x0 0x2 0x0 0x3 0x0 0x4 0x0 0x5 0x0 0x6 0x0 0x7 0x0 0x8 0x0 0x9
DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
DEBUG:pymodbus.transaction:Getting transaction 7
DEBUG:pymodbus.client.asynchronous.asyncio:Client disconnected from modbus server: None
INFO:pymodbus.client.asynchronous.asyncio:Protocol lost connection.
DEBUG:pymodbus.client.asynchronous.asyncio:Waiting 100 ms before next connection attempt.
DEBUG:pymodbus.client.asynchronous.asyncio:Connecting.
WARNING:pymodbus.client.asynchronous.asyncio:Failed to connect: [Errno 61] Connect call failed ('127.0.0.1', 5020)
DEBUG:pymodbus.client.asynchronous.asyncio:Waiting 200 ms before next connection attempt.
DEBUG:pymodbus.client.asynchronous.asyncio:Connecting.
WARNING:pymodbus.client.asynchronous.asyncio:Failed to connect: [Errno 61] Connect call failed ('127.0.0.1', 5020)
DEBUG:pymodbus.client.asynchronous.asyncio:Waiting 400 ms before next connection attempt.
DEBUG:pymodbus.client.asynchronous.asyncio:Connecting.
WARNING:pymodbus.client.asynchronous.asyncio:Failed to connect: [Errno 61] Connect call failed ('127.0.0.1', 5020)
DEBUG:pymodbus.client.asynchronous.asyncio:Waiting 800 ms before next connection attempt.
DEBUG:pymodbus.client.asynchronous.asyncio:Connecting.
WARNING:pymodbus.client.asynchronous.asyncio:Failed to connect: [Errno 61] Connect call failed ('127.0.0.1', 5020)
DEBUG:pymodbus.client.asynchronous.asyncio:Waiting 1600 ms before next connection attempt.
DEBUG:pymodbus.client.asynchronous.asyncio:Connecting.
WARNING:pymodbus.client.asynchronous.asyncio:Failed to connect: [Errno 61] Connect call failed ('127.0.0.1', 5020)
DEBUG:pymodbus.client.asynchronous.asyncio:Waiting 3200 ms before next connection attempt.
DEBUG:pymodbus.client.asynchronous.asyncio:Connecting.
DEBUG:pymodbus.client.asynchronous.asyncio:Client connected to modbus server
INFO:pymodbus.client.asynchronous.asyncio:Protocol made connection.
INFO:pymodbus.client.asynchronous.asyncio:Connected to 127.0.0.1:5020.
```
the nothing output

dhoomakethu, 2020-07-04:
A potential fix is available in this branch https://github.com/riptideio/pymodbus/tree/async-updates. Please give it a try.

dhoomakethu, 2020-07-04:
This should be kind of fixed with branch https://github.com/riptideio/pymodbus/tree/async-updates. Please give it a try

janiversen, 2022-05-29:
It is fixed..

closed 2022-05-29 by janiversen