pymodbus-dev/pymodbus #460: AttributeError: 'ExceptionResponse' object has no attribute 'registers'

andreasoie, 2019-10-24:
# Version

* Python: 3.6.8
* OS: Windows 10
* Pymodbus: 2.2.0
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp = WAGO PLC 750
* Client: tcp=  MY PC

### Description

AttributeError: 'ExceptionResponse' object has no attribute 'registers'

### Code and Logs

```python
PS C:\Users\andre\OneDrive> & C:/Users/andre/AppData/Local/Programs/Python/Python36/python.exe c:/Users/andre/OneDrive/testy.py
Connected: True
2019-10-24 16:40:26,182 MainThread      DEBUG    transaction    :111      Current transaction state - IDLE
2019-10-24 16:40:26,182 MainThread      DEBUG    transaction    :116      Running transaction 1
2019-10-24 16:40:26,182 MainThread      DEBUG    transaction    :215      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x1 0x0 0x1
2019-10-24 16:40:26,182 MainThread      DEBUG    sync           :73       New Transaction state 'SENDING'
2019-10-24 16:40:26,183 MainThread      DEBUG    transaction    :224      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2019-10-24 16:40:26,193 MainThread      DEBUG    transaction    :300      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2019-10-24 16:40:26,193 MainThread      DEBUG    transaction    :229      RECV: 0x0 0x1 0x0 0x0 0x0 0x3 0x0 0x83 0x2
2019-10-24 16:40:26,194 MainThread      DEBUG    socket_framer  :147      Processing: 0x0 0x1 0x0 0x0 0x0 0x3 0x0 0x83 0x2
2019-10-24 16:40:26,194 MainThread      DEBUG    factory        :266      Factory Response[131]
2019-10-24 16:40:26,194 MainThread      DEBUG    transaction    :379      Adding transaction 1
2019-10-24 16:40:26,194 MainThread      DEBUG    transaction    :390      Getting transaction 1
2019-10-24 16:40:26,194 MainThread      DEBUG    transaction    :189      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Traceback (most recent call last):
  File "c:/Users/andre/OneDrive/testy.py", line 31, in <module>
    result = request.registers
AttributeError: 'ExceptionResponse' object has no attribute 'registers'

```

tracernz, 2019-10-24:
Replace:
```python
request = client.read_holding_registers(1,1)
result = request.registers
print(result)
```
With:
```python
request = client.read_holding_registers(1, 1)
if request.isError():
    # handle error, log?
    print('Modbus Error:', request)
else:
    result = request.registers
    print(result)
```
You are getting an exception response, most likely from the device at the other end.

closed 2019-11-05 by dhoomakethu

armenmolnar, 2022-07-27:
Somebody can help me ? What is the probelem?

from pymodbus.client.sync import ModbusTcpClient 
client = ModbusTcpClient('192.168.0.200')
connection = client.connect()
print("Connected: " + str(connection))
while(True):
    result=client.read_coils(0, True , unit=1)
    if result.bits[0] == True :
        client.write_coil(16, True , unit=1)
    else:
        client.write_coil(16, False , unit=1)

Error message:

AttributeError: 'ReadInputRegistersResponse' object has no attribute 'bits'

janiversen, 2022-07-27:
An input register is not a bit, but 16 bits.
```
    rr = client.read_input_registers(1, 8, unit=UNIT)
    assert not rr.isError()  # nosec test that we are not an error
    txt = f"### address 1 is: {str(rr.registers[0])}"
```
But your code example and the AttributeError do not match. ReadInputRegistersResponse comes from a read_input_registers call.