pymodbus-dev/pymodbus #462: ModbusSerialServer blocking causes response dislocation

iguqur, 2019-10-31:
### Versions

* Python:3.7.4
* OS:ubuntu 16.04
* Pymodbus:2.2.0
* Modbus Hardware (if used): 

### Pymodbus Specific

* RemoteServer: tcp - sync (libmodbus C)
* Server: rtu - sync - timeout=3 - RetryOnEmpty=True - Retries=1
* Client: rtu - sync - timeout=3 - RetryOnEmpty=False

### Description

ModbusSerialServer blocking causes response dislocation.

The steps like this:
1. Client send frame 1 to server.
2. Server send frame 1 to RemoteServer.
3. The RemoteServer not responding for some reason.
4. Client timeout 3s.
5. Client send frame 2 to server.
6. Server timeout 3s.
7. Server resend frame 1 to RemoteServer.
8. The RemoteServer not responding for some reason.
9. Client timeout 3s.
10. Client send frame 3 to server.
11. Server timeout 3s.
12. **Server get frame 2 and 3.**
13. **server send frame 2 to RemoteServer, and save frame 3 in memory.**
14. RemoteServer response frame 2.
15. Server response frame 2.
16. **Client Error!!! send frame 3, but get response of frame 2.**
17. Client send frame 4.
18. **Server get frame 4, and append frame it after frame 3 .**
19. **Server send frame 3 to RemoteServer.**
20. ...
21. **Client Error!!! send frame 4, but get response of frame 3.**

dhoomakethu, 2019-11-05:
Please provide code samples and logs to understand the issue better.

iguqur, 2019-11-18:
hi 

server.py
```python3
#!/usr/bin/env python
# -*- coding:utf-8 -*-

from pymodbus.server.sync import StartSerialServer
from pymodbus.client.sync import ModbusTcpClient
from pymodbus.datastore.remote import RemoteSlaveContext
from pymodbus.transaction import ModbusRtuFramer
from pymodbus.datastore import ModbusServerContext

import sys
import logging

FORMAT = ('%(asctime)-15s %(threadName)-15s'
          ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(stream=sys.stdout, level=logging.DEBUG, format=FORMAT)

client = ModbusTcpClient(port=502, timeout=3)
slaves = RemoteSlaveContext(client)
context = ModbusServerContext(slaves=slaves, single=True)
StartSerialServer(context, framer=ModbusRtuFramer, port='/dev/ttyTHS1', baudrate=115200)
```

client.py
```python3
#!/usr/bin/env python
# -*- coding:utf-8 -*-

import logging
import sys
import time

from pymodbus.client.sync import ModbusSerialClient

FORMAT = ('%(asctime)-15s %(threadName)-15s'
          ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(stream=sys.stdout, level=logging.DEBUG, format=FORMAT)

client = ModbusSerialClient(method='rtu', port='/dev/ttyUSB0', timeout=3.1, baudrate=115200)

client.write_register(40007, 12)

client.read_input_registers(30001, 1)

client.read_discrete_inputs(10001, 10)

client.read_coils(1, 10)

client.read_discrete_inputs(10010, 10)

client.read_coils(4, 10)
```

log for server.py
```log
2019-11-13 22:00:08,635 MainThread      DEBUG    sync           :45       Client Connected [/dev/ttyTHS1:/dev/ttyTHS1]
2019-11-13 22:00:08,636 MainThread      DEBUG    sync           :522      Started thread to serve client

# request 1 timeout!

2019-11-13 22:00:14,637 MainThread      DEBUG    rtu_framer     :180      Getting Frame - 0x6 0x9c 0x47 0x0 0xc
2019-11-13 22:00:14,637 MainThread      DEBUG    factory        :137      Factory Request[WriteSingleRegisterRequest: 6]
2019-11-13 22:00:14,638 MainThread      DEBUG    rtu_framer     :115      Frame advanced, resetting header!!
2019-11-13 22:00:14,638 MainThread      DEBUG    remote         :42       validate[6] 40007:1
2019-11-13 22:00:14,643 MainThread      DEBUG    transaction    :111      Current transaction state - IDLE
2019-11-13 22:00:14,644 MainThread      DEBUG    transaction    :116      Running transaction 1
2019-11-13 22:00:14,644 MainThread      DEBUG    transaction    :215      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x9c 0x47 0x0 0x1
[... 213 lines cut ...]
2019-11-13 22:00:26,714 MainThread      DEBUG    transaction    :390      Getting transaction 0
2019-11-13 22:00:26,714 MainThread      DEBUG    transaction    :189      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2019-11-13 22:00:26,714 MainThread      DEBUG    transaction    :111      Current transaction state - TRANSACTION_COMPLETE

# request 6, but get response 4!!!

2019-11-13 22:00:26,715 MainThread      DEBUG    transaction    :116      Running transaction 6
2019-11-13 22:00:26,715 MainThread      DEBUG    transaction    :215      SEND: 0x0 0x1 0x0 0x4 0x0 0xa 0xfc 0x1d
2019-11-13 22:00:26,715 MainThread      DEBUG    rtu_framer     :264      Changing state to IDLE - Last Frame End - 1573653626.7129531, Current Time stamp - 1573653626.71556
2019-11-13 22:00:26,715 MainThread      DEBUG    rtu_framer     :272      Waiting for 3.5 char before next send - 1.75 ms
2019-11-13 22:00:26,717 MainThread      DEBUG    sync           :73       New Transaction state 'SENDING'
2019-11-13 22:00:26,718 MainThread      DEBUG    transaction    :224      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2019-11-13 22:00:29,736 MainThread      DEBUG    transaction    :300      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2019-11-13 22:00:29,736 MainThread      DEBUG    transaction    :229      RECV: 0x0 0x1 0x2 0x0 0x0 0x84 0x3c
2019-11-13 22:00:29,736 MainThread      DEBUG    rtu_framer     :180      Getting Frame - 0x1 0x2 0x0 0x0
2019-11-13 22:00:29,737 MainThread      DEBUG    factory        :266      Factory Response[ReadCoilsResponse: 1]
2019-11-13 22:00:29,737 MainThread      DEBUG    rtu_framer     :115      Frame advanced, resetting header!!
2019-11-13 22:00:29,737 MainThread      DEBUG    transaction    :379      Adding transaction 0
2019-11-13 22:00:29,737 MainThread      DEBUG    transaction    :390      Getting transaction 0
2019-11-13 22:00:29,737 MainThread      DEBUG    transaction    :189      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
```

### Description

As long as ModbusSerialServer blocks onece, the next transactions will be dislocation and cannot automatically resume normal.

closed 2021-03-08 by github-actions[bot]