pymodbus-dev/pymodbus #465: PyModbus unreliable compared to minimalmodbus

TheChatty, 2019-11-03:
### Versions

* Python: 3.8.0
* OS: Windows 10 x64
* Pymodbus: 2.3.0
* Modbus Hardware (if used): Trovis 5573

### Pymodbus Specific
* Server: rtu - sync
* Client: rtu - sync

### Description

I am comparing two simple scripts basically doing the same. The minimalmodbus script is very reliable with a timeout >=0.15s. The PyModbus script is failing very often no matter which timeout I set (even 10s). When failure happens it claims to have received only 3 of 7 bytes.

minimalmodbus script:
```
import minimalmodbus
instrument = minimalmodbus.Instrument('COM6', 247)
instrument.serial.timeout=0.15
print('Trovis type %s'     %  instrument.read_register(  0,0))
print('Temp. outside %s°C' % (instrument.read_register(  9,0)/10))
print('Vorlauftemp.  %s°C' % (instrument.read_register( 12,0)/10))
print('Rücklauftemp. %s°C' % (instrument.read_register( 16,0)/10))
print('Speichertemp. %s°C' % (instrument.read_register( 22,0)/10))
print('Stellsig. Rk1 %s%%' % (instrument.read_register(106,0)))
```

PyModbus script:
```
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
client = ModbusClient(method='rtu', port='COM6', baudrate=19200, timeout=10)
client.connect()
print('Trovis type %s'     %  client.read_holding_registers(  0,1,unit=247).registers[0])
print('Temp. outside %s°C' % (client.read_holding_registers(  9,1,unit=247).registers[0]/10))
print('Vorlauftemp.  %s°C' % (client.read_holding_registers( 12,1,unit=247).registers[0]/10))
print('Rücklauftemp. %s°C' % (client.read_holding_registers( 16,1,unit=247).registers[0]/10))
print('Speichertemp. %s°C' % (client.read_holding_registers( 22,1,unit=247).registers[0]/10))
print('Stellsig. Rk1 %s%%' % (client.read_holding_registers(106,1,unit=247).registers[0]))
client.close()
```

### Code and Logs

```python
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0xf7 0x3 0x0 0x0 0x0 0x1 0x90 0x9c
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0xf7 0x3 0x2 0x15 0xc5 0xbe 0x92
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x2 0x15 0xc5
DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 247
DEBUG:pymodbus.transaction:Getting transaction 247
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Trovis type 5573
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:SEND: 0xf7 0x3 0x0 0x9 0x0 0x1 0x40 0x9e
DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1572804058.424627, Current Time stamp - 1572804058.431607
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Incomplete message received, Expected 7 bytes Recieved 3 bytes !!!!
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0xf7 0x3 0x2
DEBUG:pymodbus.framer.rtu_framer:Frame check failed, ignoring!!
DEBUG:pymodbus.framer.rtu_framer:Resetting frame - Current Frame in buffer - 0xf7 0x3 0x2
DEBUG:pymodbus.transaction:Getting transaction 247
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Traceback (most recent call last):
  File "C:\Users\JensKöhler\Documents\Scripts\Trovis553\minimal_PyModbus_ser.py", line 10, in <module>
    print('Temp. outside %s°C' % (client.read_holding_registers(  9,1,unit=247).registers[0]/10))
AttributeError: 'ModbusIOException' object has no attribute 'registers'
```

dhoomakethu, 2019-11-04:
@TheChatty if you are using this on a windows machine create the client with `strict=False` . That should fix the issue you are facing.

TheChatty, 2019-11-04:
@dhoomakethu: That solved it. Then I was experimenting with different inter_char_timeout value and found that 3.5 * self._t0 did the trick as well. Values lower than that had the error appear once in a while.

closed 2019-11-04 by TheChatty