pymodbus-dev/pymodbus #467: Deprecated coroutine

albireox, 2019-11-11:
### Versions

* Python: 3.8
* OS: macOS 10.15
* Pymodbus: 2.3.0
* Modbus Hardware (if used): WAGO

### Pymodbus Specific
* Client: tcp - async

### Description

Trying to import `from pymodbus.client.asynchronous.asyncio import AsyncioModbusTcpClient` produces a series of messages `[WARNING]: "@coroutine" decorator is deprecated since Python 3.8, use "async def" instead (DeprecationWarning)`. Python 3.8 has deprecated the `coroutine` decorator. Also, there is a filter in pymodbus that always shows these warnings so it's not possible to suppress them.

JamesJeffryes, 2020-09-14:
@dhoomakethu I'm happy to open a PR to address this. One thing to note is that converting to use of the async keyword would limit the library to 3.5+ but that seems like a pretty reasonable limitation to me.

dhoomakethu, 2020-09-14:
Yes, pymodbus 3 would be only available for Python 3.5 and above and it should be good to have that change. Feel free to raise a PR

swamper123, 2022-03-31:
This was cleaned up with #564 .
At least I couldn't find any decorator.

closed 2022-03-31 by swamper123