pymodbus-dev/pymodbus #472: Socket framer tries to decode partial ADU

marcin1j, 2019-12-04:
### Versions

* Python: 3.6.5
* OS: Linux 4.19
* Pymodbus: 2.1.0
* Modbus Hardware (if used): Modbus TCP/RTU gateway (microcontroller, custom hardware)

### Pymodbus Specific
* Server: not used
* Client: tcp async

### Description

Modbus TCP asyncio client tries to decode partially received TCP/IP ADU frame and complains about being unable to decode request.
This happens sparsely but strongly suggests a design flaw in TCP decoder. It expects a whole ADU frame to arrive in one instant, without any fragmentation etc. In fact TCP is a stream protocol and client should wait as long as necessary to receive a complete ADU and only then start decoding.

### Code and Logs
Pymodbus logs:
```
pymodbus.client.async.asyncio: send: 0x0 0x27 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0xe 0x0 0x8
pymodbus.client.async.asyncio: recv: 0x0 0x27 0x0 0x0 0x0 0x13 0x1
pymodbus.framer.socket_framer: Processing: 0x0 0x27 0x0 0x0 0x0 0x13 0x1
pymodbus.factory: Unable to decode response Modbus Error: Unknown response 0
asyncio: Exception in callback _SelectorSocketTransport._read_ready()
handle: <Handle _SelectorSocketTransport._read_ready()>
Traceback (most recent call last):
  File "/usr/lib64/python3.6/asyncio/events.py", line 145, in _run
    self._callback(*self._args)
  File "/usr/lib64/python3.6/asyncio/selector_events.py", line 730, in _read_ready
    self._protocol.data_received(data)
  File "/usr/lib64/python3.6/site-packages/pymodbus/client/async/asyncio/__init__.py", line 189, in data_received
    self._dataReceived(data)
  File "/usr/lib64/python3.6/site-packages/pymodbus/client/async/asyncio/__init__.py", line 136, in _dataReceived
    self.framer.processIncomingPacket(data, self._handleResponse, unit=unit)
  File "/usr/lib64/python3.6/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/usr/lib64/python3.6/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
```

ADU sent by the gateway looks like this:
 * MBAP: `00270000001301`
 * PDU: `031000000000000000004364cccd42480a3d`

dhoomakethu, 2019-12-04:
Are you observing the same behaviour with the latest pymodbus 2.3.0 ?

charliboy, 2019-12-11:
I have the same problem. It's caused by this bug :
https://github.com/riptideio/pymodbus/issues/469

marcin1j, 2019-12-11:
@dhoomakethu Sorry for my late reply. It seems I forgot to send my comment a couple of days ago...

First of all, switching to pymodbus 2.3.0 doesn't make any difference. Sadly all the extra logs (from both 2.2.0 and 2.3.0) are now gone but I assure there was no difference in log messages.

Now to the point. After a careful review my finding is that it looks like the library complains about incomplete frame but later, when next portion of data arrives, successfully decodes payload. I was wondering why client-server synchronization is not lost and here's the answer: incomplete frame that cause error message is not discarded and later its decoding is retried.

Now everything adds up: error message is erroneous and the payload is eventually decoded despite the error message.

@charliboy I think you reported a different issue. My problem could be triggered by sending MBAP and PDU by consecutive but separate socket writes. Data length was rather small (<256 bytes).
If I'm correct, in your case frame length exceeds read buffer size.

Also note that while I've seen this behavior in an early prototype gateway code (two separate writes), "production" version sends whole frame in a single call so in practice there should be no data fragmentation in my use case (which checks - there was no decoding error).
I've decided to report this problem because:

1. any correct protocol implementation should take into account stream nature of TCP protocol and conversely be prepared to receive Modbus frame in chunks,
2. the problem probably would be more apparent with longer frames.

charliboy, 2019-12-12:
@marcin1j  When the communication data is large, many frames data are received at the same time. due to the bug of https://github.com/riptideio/pymodbus/issues/469 ,the last frame data with MBAP header but incomplete will be discarded . The new data received next time contains the rest of the data of the previous frame, but when decoding, because there is no MBAP, the system cannot decode the data, so the error you mentioned will be reported.

janiversen, 2022-05-25:
We have added a number of corrections to deal with incomplete packets.

So your issue should be solved on dev.

If you still have an issue, feel free to submit a new PR.

closed 2022-05-25 by janiversen