pymodbus-dev/pymodbus #477: Asyncio server fails on every connection when the logging level is set to "debug"

lovasoa, 2019-12-18:
### Pymodbus Specific
* Server: tcp - async with asyncio

### Description

The asyncio server raises an error on every connection when the logging level is set to "debug".
I think this is due to this line:
https://github.com/riptideio/pymodbus/blob/master/pymodbus/server/asyncio.py#L247

```python
        _logger.debug("TCP client connection established [%s:%s]" % self.client_address)
```

there are two `%s`, but a single argument.

### Code and Logs

```
DEBUG:pymodbus.server.asyncio:Datastore unable to fulfill request: not all arguments converted during string formatting; Traceback (most recent call last):
  File "/Users/olojkine/Developpement/experimental/droop_controller/venv/lib/python3.7/site-packages/pymodbus/server/asyncio.py", line 59, in connection_made
    _logger.debug("Socket [%s:%s] opened" % transport.get_extra_info('sockname'))
TypeError: not all arguments converted during string formatting
```

lovasoa, 2019-12-18:
The same bug seems to exist in `connection_lost`, in the same class.

This may come from a quick copy&paste from the synchronous server. In the synchronous server, `self.client_adress` is a tuple `(host, port)`. In the async server, it is obtained from `transport.get_extra_info('peername')`, which is a string.

lovasoa, 2019-12-18:
Here is the documentation for the format returned by `transport.get_extra_info('peername')` and  `transport.get_extra_info('peername')` : 

https://docs.python.org/3/library/socket.html#socket-families

The value is not always a two-tuple.

I worked around the bug by binding the server on `127.0.0.1` instead of `localhost`.

janiversen, 2022-05-31:
Solved on dev.

closed 2022-05-31 by janiversen