pymodbus-dev/pymodbus #479: AttributeError: 'NoneType' object has no attribute 'cancel' in pymodbus.server.asyncio

lovasoa, 2019-12-19:
### Versions

* Python: 3.7
* OS: MacOS 10.15.1
* Pymodbus: 2.3.0

### Pymodbus Specific
* Server: tcp - asyncio

### Description

Launching a local asyncio server with verbose logging, connecting to it, and then disconnecting.

### Code and Logs

```python
import asyncio
from pymodbus.server.asyncio import StartTcpServer
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

import logging

if __name__ == '__main__':
    logging.basicConfig()
    logging.getLogger().setLevel(logging.DEBUG)

    context = ModbusServerContext(slaves=ModbusSlaveContext(), single=True)
    asyncio.run(StartTcpServer(context, address=("localhost", 5020), defer_start=False))
```

Then connecting to the tcp socket and immediately disconnecting : 

```
DEBUG:pymodbus.server.asyncio:Datastore unable to fulfill request: 'NoneType' object has no attribute 'cancel'; Traceback (most recent call last):
  File ".../venv/lib/python3.7/site-packages/pymodbus/server/asyncio.py", line 82, in connection_lost
    self.handler_task.cancel()
AttributeError: 'NoneType' object has no attribute 'cancel'
```

dhoomakethu, 2020-01-11:
Use `address=("0.0.0.0", 5020)` instead of `('localhost', 5020)`. That should solve this issue. 
Sorry for the delayed response.

lovasoa, 2020-01-11:
This is a useful workaround, but don't you think the underlying issue deserves a fix ? `"localhost"` is a perfectly valid host. If you don't have the bandwidth to fix the issue now, maybe you can just leave it opened ?

dhoomakethu, 2020-01-12:
I agree , that's is the reason why I have put the label as `enhancement`

lovasoa, 2021-03-03:
This is still an issue.

PR #947 (2022-06-30): Only cancel task if it is created

fixes #479
closed 2022-06-30 by janiversen