pymodbus-dev/pymodbus #485: pymodmus sync RTU client produces CRC failure

otdftr, 2019-12-30:
Upon receiving serial data, the slave unit id is returned as a two byte value MSB first. This behaviour  after the udate from pymodbus 1.3 > 2.3 an a debian (raspian) update).

Up to now, I could not find out the actual reason for this (and I am not a python programmer). I assume it could be a type cast somewhere in the receiving code.

For the time being, I improvised a quick fix which works for me. See my findings below.

## debug findings

Snippet from debug output shows the leading extra 0x00 in the RECV buffer. This leads to
* the second CRC byte being chopped of and eventually to a
* CRC failure

````
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x2 0x3 0x0 0x18 0x0 0x9 0x5 0xf8
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x0 0x2 0x3 0x12 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x14 0x53 0xf0 0x0 0x14 0x53 0xf0 0x0 0x0 0x0 0x14 0x6f
DEBUG:pymodbus.framer.rtu_framer:CRC invalid, discarding header!!
````

## ugly and improvised quick fix

Before starting further investigation, for a quick fix ("works for me") I hacked into the response length calculation (increase by one byte) and chopped of the first byte of the RECV buffer - that made the RTU communications work again:

````
DEBUG:pymodbus.transaction:SEND: 0x2 0x3 0x0 0x18 0x0 0x9 0x5 0xf8
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x2 0x3 0x12 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x14 0x54 0x30 0x0 0x14 0x54 0x30 0x0 0x0 0x0 0x14 0x8f 0xa
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x12 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x14 0x54 0x30 0x0 0x14 0x54 0x30 0x0 0x0 0x0 0x14
DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 2
DEBUG:pymodbus.transaction:Getting transaction 2
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
INFO:root:Response - Registers: [0, 0, 0, 20, 21552, 20, 21552, 0, 20]
DEBUG:pymodbus.payload:[0, 0, 0, 20, 21552, 20, 21552, 0, 20]
````

`# patch agaist commit fbdc470ae3e138c50e3659ec4ec8ebf39df58936`

````
~/pymodbus $ git diff
diff --git a/pymodbus/transaction.py b/pymodbus/transaction.py
index 06a80ac..083c6ea 100644
--- a/pymodbus/transaction.py
+++ b/pymodbus/transaction.py
@@ -75,6 +75,9 @@ class ModbusTransactionManager(object):
             self.base_adu_size = 7  # tid(2), pid(2), length(2), uid(1)
         elif isinstance(self.client.framer, ModbusRtuFramer):
             self.base_adu_size = 3  # address(1), CRC(2)
+            ## well for some reason address is two byte, leadin' zero
+            ## increase recv size + chop of first byte
+            self.base_adu_size = 4
         elif isinstance(self.client.framer, ModbusAsciiFramer):
             self.base_adu_size = 7  # start(1)+ Address(2), LRC(2) + end(2)
         elif isinstance(self.client.framer, ModbusBinaryFramer):
@@ -229,6 +232,7 @@ class ModbusTransactionManager(object):
                               "to 'WAITING FOR REPLY'")
                 self.client.state = ModbusTransactionState.WAITING_FOR_REPLY
             result = self._recv(response_length, full)
+            result = result[1:]
             if _logger.isEnabledFor(logging.DEBUG):
                     _logger.debug("RECV: " + hexlify_packets(result))
````

dhoomakethu, 2020-01-01:
@otdftr what is the slave under discussion here ? is it an external modbus device or pymodbus serial server running locally ?

otdftr, 2020-01-01:
hi @dhoomakethu, the devices are modbus electricity meters, both show the same behaviour. ID #2 is a Finder 7E.8.400, #21 (not shown above) an Eastron SDM630-Modbus V2.

bsuiram, 2020-05-20:
we are also seeing something like this on aarch64.
Same code works on x86_64. same external device, 

Failing aarch64 (arm64):
`2020-05-20 07:25:30,228 - DEBUG - pymodbus.transaction - Current transaction state - IDLE
2020-05-20 07:25:30,229 - DEBUG - pymodbus.transaction - Running transaction 1
2020-05-20 07:25:30,229 - DEBUG - pymodbus.transaction - SEND: 0xa 0x4 0x75 0x3f 0x0 0x1 0x1a 0xb1
2020-05-20 07:25:30,229 - DEBUG - pymodbus.client.sync - New Transaction state 'SENDING'
2020-05-20 07:25:30,230 - DEBUG - pymodbus.transaction - Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-05-20 07:25:30,258 - DEBUG - pymodbus.transaction - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2020-05-20 07:25:30,259 - DEBUG - pymodbus.transaction - RECV: 0xd 0xa 0x4 0x2 0x36 0x86 0x8a
2020-05-20 07:25:30,259 - DEBUG - pymodbus.framer.rtu_framer - CRC invalid, discarding header!!
2020-05-20 07:25:30,260 - DEBUG - pymodbus.framer.rtu_framer - Resetting frame - Current Frame in buffer - 0xd 0xa 0x4 0x2 0x36 0x86 0x8a
2020-05-20 07:25:30,260 - DEBUG - pymodbus.framer.rtu_framer - Frame check failed, ignoring!!
`

Working x86_64 (intel atom):
`2020-05-20 07:25:30,055 - DEBUG - pymodbus.transaction - Current transaction state - IDLE
2020-05-20 07:25:30,056 - DEBUG - pymodbus.transaction - Running transaction 1
2020-05-20 07:25:30,058 - DEBUG - pymodbus.transaction - SEND: 0xa 0x4 0x75 0x3f 0x0 0x1 0x1a 0xb1
2020-05-20 07:25:30,060 - DEBUG - pymodbus.client.sync - New Transaction state 'SENDING'
2020-05-20 07:25:30,062 - DEBUG - pymodbus.transaction - Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-05-20 07:25:30,108 - DEBUG - pymodbus.transaction - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2020-05-20 07:25:30,108 - DEBUG - pymodbus.transaction - RECV: 0xa 0x4 0x2 0x65 0x8b 0x77 0xc6
2020-05-20 07:25:30,110 - DEBUG - pymodbus.framer.rtu_framer - Getting Frame - 0x4 0x2 0x65 0x8b
2020-05-20 07:25:30,110 - DEBUG - pymodbus.factory - Factory Response[ReadInputRegistersResponse: 4]
2020-05-20 07:25:30,111 - DEBUG - pymodbus.framer.rtu_framer - Frame advanced, resetting header!!
2020-05-20 07:25:30,112 - DEBUG - pymodbus.transaction - Adding transaction 10
2020-05-20 07:25:30,112 - DEBUG - pymodbus.transaction - Getting transaction 10
2020-05-20 07:25:30,113 - DEBUG - pymodbus.transaction - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2020-05-20 07:25:30,113 - DEBUG - pymodbus.payload - [25995]`

dhoomakethu, 2020-05-24:
@bsuiram the receive data on the aarch64 looks messy, does this happen for all reads or once in a while ? What OS are you running ?

bsuiram, 2020-05-25:
It happens for all reads.
The platform is a docker container (arm64v8/python:3-alpine) running on a cisco ir1101 router.
We are currently trying to verify the serial connection from the container to the device.

After implementing the quickfix described above (chopping of the first byte of the RECV buffer) we are getting strange values from the registers we are trying to read from the device, but no crc errors.

bsuiram, 2020-06-12:
Do you need more infromation on this issue @dhoomakethu?

dhoomakethu, 2020-06-13:
@bsuiram Apologies for the delay. I haven't gotten any time to work on pymodbus offlate, so expect delays with the backlog clearing. Will check and update

branati, 2020-10-02:
For me run with:
self.base_adu_size = 8
and
result = result[3:]
Thank you friends!!!!

closed 2021-03-08 by github-actions[bot]