pymodbus-dev/pymodbus #488: Modbus server not retry to send packet when client not responding, Retries not working

modular-tester-wa, 2020-01-21:
### Versions

* Python: 3.7
* OS: Windows
* Pymodbus: 2.2.0
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: rtu
* Client: rtu

### Description
Modbus server not retry to send packet when client not responding. Packet is sent only first time, after that no any activity from server.

### Code

```python
 # code and logs here.
self.modbus_client = ModbusClient(method='rtu', port=5, baudrate=38400)
self.modbus_client.timeout = 0.5
self.modbus_client.Retries = 3
self.modbus_client.RetryOnEmpty = True
```
### Logs from pymodbus 
```
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 8
DEBUG:pymodbus.transaction:SEND: 0x6 0x1 0x0 0x0 0x0 0x10 0x3c 0x71
DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1579650757.720018, Current Time stamp - 1579650777.660689
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Incomplete message received, Expected 7 bytes Recieved 0 bytes !!!!
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 
DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
DEBUG:pymodbus.transaction:Getting transaction 6
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
```

jawnsy, 2020-10-05:
Dear maintainer:

Thanks for creating this awesome library!

I'm seeing a similar issue with the synchronous client, here are relevant parts of my code:

```python
from pymodbus.client.sync import ModbusTcpClient

with ModbusTcpClient("localhost", port=502, timeout=10, retries=10) as client:
    rq = client.read_discrete_inputs(0, count=1, unit=1)
    if rq.isError():
       # log a message
```

I worked around this by adding my own retry logic (if `rq.isError` is true), but it would be nice to understand why the internal retry mechanism doesn't seem to be working or if I can turn on additional debugging. Maybe I need to look into the async client version instead?

```
2020-10-04 17:45:40,148 pymodbus.transaction DEBUG Running transaction 20
2020-10-04 17:45:40,148 pymodbus.transaction DEBUG SEND: 0x0 0x14 0x0 0x0 0x0 0x6 0x1 0x2 0x0 0x0 0x0 0x57
2020-10-04 17:45:40,148 pymodbus.client.sync DEBUG New Transaction state 'SENDING'
2020-10-04 17:45:40,148 pymodbus.transaction DEBUG Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-10-04 17:45:40,149 pymodbus.transaction DEBUG Transaction failed. ([WinError 10053] An established connection was aborted by the software in your host machine) 
2020-10-04 17:45:40,149 pymodbus.framer.socket_framer DEBUG Processing: 
2020-10-04 17:45:40,149 pymodbus.transaction DEBUG Getting transaction 20
2020-10-04 17:45:40,149 pymodbus.transaction DEBUG Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
```

Platform info:

* Windows 10 [Version 10.0.18363.1082]
* Python 3.8.3 (Miniconda3)
* pymodbus 2.4.0 installed via pip

tracernz, 2020-10-05:
This line looks relevant: https://github.com/riptideio/pymodbus/blob/de3955e46087f5cdcd77184da9a681d7ad20c032/pymodbus/transaction.py#L169

closed 2021-03-08 by github-actions[bot]