pymodbus-dev/pymodbus #490: Maximum value for count at read_holding register

rodrigorpgsc, 2020-01-28:
### Versions

* Python: 3.6.9-64bits
* OS:Ubuntu bionic64
* Pymodbus: 2.2.0
* Modbus Hardware (if used):  None

### Pymodbus Specific
* Client: tcp - sync

### Description

The ammount of register from read_holding_register don`t sound rigth if the size is above 128. If you run the code below you`ll able to see the behaviour

### Code and Logs

```
con = ModbusTcpClient('10.0.2.2', '502')
result = con.read_holding_registers(1, 127)
len(result.registers)
result2 = con.read_holding_registers(1, 129)
len(result2.registers)

```
The output is 127 and 1, but should be 127 and 129. What is wrong at my code?

dhoomakethu, 2020-01-29:
The max number of registers you can poll using read_holding_registers is 125 as per modbus specs, And it is not guaranteed if any number greater than that would yield a proper value. you will have to split up your reads .

closed 2020-01-29 by dhoomakethu

rodrigorpgsc, 2020-01-29:
ah ok, I didn`t know that! Thanks for your reply

commit 83e57341 (2020-08-01): Asyncio bug fixes (#517)

* Closes  #491

* 1. update requirements
2. Fix examples
3. Fix #494 - handle_local_echo
4. Fix #500 -- asyncio serial client with already running loop
5. Fix #486 - Pass serial args for asyncio serial client
6. Fix #490 - Typo in decode_data for socker_framer
7. Fix #385 - Support timeouts to break out of responspe await when server goes offline
8. Misc updates

* #516 custom data block fix
commit 29f694c8 (2020-09-11): #515 fix repl broadcast (#531)

* 1. update requirements
2. Fix examples
3. Fix #494 - handle_local_echo
4. Fix #500 -- asyncio serial client with already running loop
5. Fix #486 - Pass serial args for asyncio serial client
6. Fix #490 - Typo in decode_data for socker_framer
7. Fix #385 - Support timeouts to break out of responspe await when server goes offline
8. Misc updates

* #516 custom data block fix

* Fix broadcast error  with REPL client #515

* Fix #509 Wrong unit ID referenced in framers

* Update documentation for serial forwarder example. Fixes #525

* Fix unit tests, support pyt [... cut]