pymodbus-dev/pymodbus #494: 'ModbusTcpClient' object has no attribute 'handle_local_echo' if client != ModbusSerialClient

wildstray, 2020-02-14:
Branch: dev

  File "/home/andrea/LWE/MPO/Myna/wolf/pymodbus/transaction.py", line 242, in _transact
    if self.client.handle_local_echo is True:
AttributeError: 'ModbusTcpClient' object has no attribute 'handle_local_echo'

Suggested fix:

```if getattr(self.client, 'handle_local_echo', False) is True:```

Instead of:

```if self.client.handle_local_echo is True:```

gcbirzan, 2020-02-19:
```python
if getattr(self.client, 'handle_local_echo'):
```
Is the pythonic way of doing that.

dhoomakethu, 2020-07-03:
Fixed with https://github.com/riptideio/pymodbus/commit/e402d9fdd2f1f8a4e178b8da0c28b3b17aeefac3#diff-40a48d0b5cf576d37ccece3543183f65R242
Please give a try with this branch https://github.com/riptideio/pymodbus/tree/asyncio-bug-fixes

commit 83e57341 (2020-08-01): Asyncio bug fixes (#517)

* Closes  #491

* 1. update requirements
2. Fix examples
3. Fix #494 - handle_local_echo
4. Fix #500 -- asyncio serial client with already running loop
5. Fix #486 - Pass serial args for asyncio serial client
6. Fix #490 - Typo in decode_data for socker_framer
7. Fix #385 - Support timeouts to break out of responspe await when server goes offline
8. Misc updates

* #516 custom data block fix
commit 29f694c8 (2020-09-11): #515 fix repl broadcast (#531)

* 1. update requirements
2. Fix examples
3. Fix #494 - handle_local_echo
4. Fix #500 -- asyncio serial client with already running loop
5. Fix #486 - Pass serial args for asyncio serial client
6. Fix #490 - Typo in decode_data for socker_framer
7. Fix #385 - Support timeouts to break out of responspe await when server goes offline
8. Misc updates

* #516 custom data block fix

* Fix broadcast error  with REPL client #515

* Fix #509 Wrong unit ID referenced in framers

* Update documentation for serial forwarder example. Fixes #525

* Fix unit tests, support pyt [... cut]
closed 2020-09-11 by dhoomakethu