pymodbus-dev/pymodbus #497: Sync ModbusSerialClient

JulesEmangeard, 2020-03-17:
### Versions

* Python: 3.7
* OS: Windows 10 x64
* Pymodbus: 2.3.0
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

I was trying to get Modbus RTU equipments informations (like brand or reference) using function code 43 (sub function 14) but when using the ModbusSerialClient the answer hasn't got the requiered length so returns a bad response. To fix this I commented 3 lines in the ModbusSerialClient to let the request end until timeout.

### Code and Logs
In pymodbus/client/sync.py:

class ModbusSerialClient(BaseModbusClient):
    def _wait_for_data(self):
        size = 0
        more_data = False
        if self.timeout is not None and self.timeout != 0:
            condition = partial(lambda start, timeout:
                                (time.time() - start) <= timeout,
                                timeout=self.timeout)
        else:
            condition = partial(lambda dummy1, dummy2: True, dummy2=None)
        start = time.time()
        while condition(start):

            avaialble = self._in_waiting()
            # if (more_data and not avaialble) or (more_data and avaialble == size):
            #     print("Break avaialble = %s size = %s" % (avaialble, size))
            #     break
            if avaialble and avaialble != size:
                more_data = True
                size = avaialble
            time.sleep(0.01)
        return size

dhoomakethu, 2020-03-18:
On windows machine, use `strict=False` when the client is initialized . That should solve the issues most of the time.
`client = ModbusSerialClient(method='rtu', ...., strict=False)`

dhoomakethu, 2020-04-21:
@JulesEmangeard Will be closing this .Please re-open if the solution provided does not help

closed 2020-04-21 by dhoomakethu