pymodbus-dev/pymodbus #499: Why does BinaryPayloadBuilder.add_bits flip bit order

teddyrendahl, 2020-04-14:
### Description
I have some `pymodbus` code that is writing different bits to a register and noticed that the `BinaryPayloadBuilder` is doing some interesting manipulation of my bits that I pass in before writing them. 

Now, I want to re-iterate, I am **not** talking about the word order or the byte order. I am talking about the order of the bits that I am passing in.
### Code and Logs
Now to give an example I will attempt to write  an INT16 with value 15 to a register in Big Endian byte order using both `add_bits` and the `add_16bit_int` methods. Obviously, if this was my actual code I would just use the latter, but this simple example is intended to illustrate my confusion. 
```python
builder = BinaryPayloadBuilder(byteorder=Endian.Big, wordorder=Endian.Little)
builder.add_16bit_int(15)
builder.to_coils()
[False,
 False,
 False,
 False,
 False,
 False,
 False,
 False,
 False,
 False,
 False,
 False,
 True,
 True,
 True,
 True]
```
Now this checks out, treating the 0th element in the list as the most significant (Big Endian), and the order of the bits within each byte as Big Endian, removing all leading zeros i'm left with `b'1111'` which is equivalent to 15.

Now, let's say I want to recreate this same register value, but instead use the `add_bits` method
```python
builder = BinaryPayloadBuilder(byteorder=Endian.Big, wordorder=Endian.Little)
builder.add_bits([False]  * 8
builder.add_bits([False] * 4 + [True] * 4)
[False,
 False,
 False,
 False,
 False,
 False,
 False,
 False,
 True,
 True,
 True,
 True,
 False,
 False,
 False,
 False]
 ```
This is not what I expected. I have the byte order I expected, but it appears the bits within the bytes I added got reversed? Once again, removing leading zeros this will give `b'11110000'` 240. 

What is the reason the bits are being being reversed like this? Is there something I'm missing about `add_bits`? Neither word or byte order should affect this.

dhoomakethu, 2020-04-14:
@teddyrendahl Interesting observation. Let me check and update (Apologies if it takes more time to get back)

transistorgrab, 2024-01-15:
I had some time today to look into this. Please find attached my observations. I had to go into the method ```pack_bitstring``` in **utilities.py** for the final culprit in release version pymodbus-3.6.3. I hope this helps. :)

```bits = [False]*4 + [True]*4```
```[False, False, False, False, True, True, True, True]```

```pack_bitstring``` starts with ```bits[0]``` which has the highest significance and goes up to ```bits[7]``` with the lowest significance.

```bits[0:4]``` all False results in ```packed == 0```.

```bits[4] == True``` results in ```packed == 128``` and with ```packed >>= 1``` in ```packed == 64```

```bits[5] == True``` results in ```packed == 192``` and with ```packed >>= 1``` in ```packed == 96```

```bits[6] == True``` results in ```packed == 224``` and with ```packed >>= 1``` in ```packed == 112```

```bits[7] == True``` results in ```packed == 240```

now with all 8 bits read in ```struct.pack``` is called with arguments ```">B", 240```
">B" means "big-endian, 'unsigned char' (integer)"

now ```ret == b'\xf0'```. 'f0' is the hex interpretation of 240 or 0b11110000

When **changing the order in which bits are read in** by using
```for bit in reversed(bits):```
we will get ```ret == b'\x0f'``` with 0f the hex interpretation of 15 or 0b00001111

The bit order will be preserved.

Testing this with a few other values seems to prove the correctness of this aproach.

janiversen, 2024-01-17:
Pull requests are welcome.

janiversen, 2024-07-21:
Closing as no longer an issue.

closed 2024-07-21 by janiversen

MrWaloo, 2024-09-08:
I am encountering the same issue than teddyrendahl: the bit order in the bytes is reversed. Like him, I absolutely am aware of the byte and word endianess, that's not the point.

I have a Wago PLC in which I can read the memory with Modbus requests. I am only encountering an issue when I read bits/coils.

In this PLC, I test pymodbus coil reading function function un a word addressed at %MW203. Every 16 bit of this word can be seen here for the test case:

[image]

So the 3 first bits are set, every others aren't. The value of the word is 7:
2 ** 0 + 2 ** 1 + 2 ** 2 = 1 + 2 + 4 = 7
I want to read the first byte of this word. This byte should have the value `b'\x07'`

```python
>>> from pymodbus import FramerType
>>> from pymodbus.payload import BinaryPayloadDecoder
>>> from pymodbus.client import ModbusTcpClient
>>> from pymodbus.pdu.bit_read_message import ReadCoilsRequest
>>> from pymodbus.pdu.register_read_message import ReadHoldingRegistersRequest
>>>
>>> rq1 = ReadCoilsRequest(15536, 8, 1)
>>>
>>> client = ModbusTcpClient(host='192.168.1.20', port=502, framer=FramerType.SOCKET)
>>> client.connect()
True
>>> resp = client.execute(rq1)
>>> resp.bits
[True, True, True, False, False, False, False, False]
```

Here we can see that the bits are read correctly: only the 3 first bits are set. But the good practice for endianess converts is to work with BinaryPayloadDecoder, let's do this :

```
>>> dec = BinaryPayloadDecoder.fromCoils(resp.bits)
>>> dec.decode_bits()
[False, False, False, False, False, True, True, True]
>>> dec._payload
b'\xe0'
```

The result of `decode_bits()` is reversed from what I expected.

This isn't due to the functions `pack_bitstrings` and `unpack_bitstrings`, because, as transistorgrab wrote, they perfectly do their job. It is due to the `fromCoils()` class method:
```python
    @classmethod
    def fromCoils(
        cls,
        coils,
        byteorder=Endian.LITTLE,
        _wordorder=Endian.BIG,
    ):
        """Initialize a payload decoder with the result of reading of coils."""
        if isinstance(coils, list):
            payload = b""
            if padding := len(coils) % 8:  # Pad zeros
                extra = [False] * padding
                coils = extra + coils
            chunks = cls.bit_chunks(coils)
            for chunk in chunks:
                payload += pack_bitstring(chunk[::-1]) # Here, chunk is reversed and should not be
            return cls(payload, byteorder)
        raise ParameterException("Invalid collection of coils supplied")
```

In the class `BinaryPayloadBuilder`, the function `to_coils()` has the same issue:
```python
    def to_coils(self) -> list[bool]:
        """Convert the payload buffer into a coil layout that can be used as a context block.

        :returns: The coil layout to use as a block
        """
        payload = self.to_registers()
        coils = [bool(int(bit)) for reg in payload for bit in format(reg, "016b")] # Here ...
        return coils
```
Since `format(reg, "016b")` returns a representation for human of reg the result cannot be used to loop over it.
```python
>>> format(7, "016b")
'0000000000000111'
>>> format(7, "008b") # just for the example
'00000111'
```
The representation returned by `format` is correct and shows the order from the highest significance to the lowest significance.
But a code returning the bit order from the lowest significance to the highest should be used.

Now: I have searched `fromCoils` and `to_coils` in the entire repository and have not found any detailed documentation of the expected results, but the way bits payload are managed does not fit my expectations. Am I the only one? Should it be corrected? Can I make a pull request?