pymodbus-dev/pymodbus #502: ModbusTcpServer cannot get value from client device.

qwyinxiang, 2020-05-07:
My device connection diagram is like this：
ModbusTCPServer<——>ModbusTCPClient1<——>Device1 With Modbus Protocol.
ModbusTCPServer<——>ModbusTCPClient2<——>Device2 With Modbus Protocol.
ModbusTCPServer<——>ModbusTCPClient3<——>Device3 With Modbus Protocol.                         
Can any one give me a example to obtain devices datas through ModbusTCPServer?
I  don’t particularly understand what ModbusSequentialDataBlock means？
My devices has a function code=3, and slave ID=1,2,3,and the count=2.
I don't know how can I obtain devices' data?

dhoomakethu, 2020-05-08:
Client is the one which can fetch data from a modus device a.k.a server. Is your device a real device or are you trying to simulate it with pymodbus?

qwyinxiang, 2020-05-11:
@dhoomakethu ，my device is real modbus sensors, such as noise sensor, temperature and humidity sensor and other real sensors, They all use the Modbus protocol. I have changed the noise sensor and temp&humi sensor's unitId to 1,2. and the sensors function code is 3, sensor's address is 0 and register count is 2.
When I use the ModbusTcpClient, I call the read_holding_registers() function to get sensors' data.

But now, I want to use the ModbusTcpServer to get sensor data. I don't know how to get data?
I tried to use the example below, but can't get the wanted data.
`    store = ModbusSlaveContext(

        hr=ModbusSequentialDataBlock(0, [17]*100),
        ir=ModbusSequentialDataBlock(0, [17]*100))

    context = ModbusServerContext(slaves=store, single=True)

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = '2.3.0'

    StartTcpServer(context, identity=identity, address=("localhost", 5020))`

closed 2021-03-08 by github-actions[bot]