pymodbus-dev/pymodbus #506: Client doesn't connect to Server correctly

nat-kh, 2020-05-28:
### Versions
* Python: 3.6.4
* Pymodbus:  2.3.0
* Modbus Hardware (if used): Micro820 (Client) 

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

### Description
I'm trying to implement a Modbus server using pymodbus to serve different modbus clients. 
I don't have errors in my code, but my client doesn't correspond to the right server. When I run the server code and the client code, I get the Register values from the Client's modbus map and not from the server's map(the one that I implemented of Course). 

### Code and Logs

**Server** 
```python
# ------------------------------------------------------------------ #
# Import various server implementations
# ------------------------------------------------------------------ #

from pymodbus.server.sync import StartTcpServer, ModbusTcpServer
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from threading import  Thread

# ------------------------------------------------------------------ #
# Import payload builder&decoder
# ------------------------------------------------------------------ #

from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder
from pymodbus.constants import Endian

# ------------------------------------------------------------------ #
# Configure logging
# ------------------------------------------------------------------ #

import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s'
          ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def run_server():
    # ----------------------------------------------------------------------- #
    # Set the starting address, initial value and the number of addresses.
    # ----------------------------------------------------------------------- #
    # Here we use the same reference block for each underlying store.
    # ----------------------------------------------------------------------- #

    start_addr = 0x00
    init_val = 0
    num_addr = 100 

    block = ModbusSequentialDataBlock(start_addr, [init_val] * num_addr)
    block1 = ModbusSequentialDataBlock(0, [9] * 100)
    slaves = {
                  0x01: ModbusSlaveContext(di=block, co=block, hr=block1, ir=block)
              }
    context = ModbusServerContext(slaves=slaves, single=False)

    #context[0].setValues(0x10, 40, [78000]*2)
    #regs_to_read = context[0].getValues(3, 40, 8)
    #print(regs_to_read)

    # ----------------------------------------------------------------------- #
    # initialize the server information
    # ----------------------------------------------------------------------- #
    # If you don't set this or any fields, they are defaulted to empty strings.
    # ----------------------------------------------------------------------- #
    identity = ModbusDeviceIdentification()
    identity.VendorName = ''
    identity.ProductCode = ''
    identity.ProductName = ''
    identity.ModelName = ''
    identity.MajorMinorRevision = ''

    # ----------------------------------------------------------------------- #
    # run the server you want
    # ----------------------------------------------------------------------- #
    #StartTcpServer(context, identity=identity, address=('', 50000))
    server = ModbusTcpServer(context=context, identity=identity, address=('', 50000))
    #server.server_address = "192.168.90.137"
    server.serve_forever()

if __name__ == "__main__":
    run_server()
```
**Client** 
```python
# ----------------------------------------------------------- #
# import client implementation
# ----------------------------------------------------------- #
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.constants import Endian

# ----------------------------------------------------------- #
# Configure client's logging
# ----------------------------------------------------------- #
import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s'
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

UNIT = 0x01

def run_client():
    # ----------------------------------------------------------- #
    # TCP Client
    # ----------------------------------------------------------- #
    client = ModbusClient(host='172.16.244.120', port=502)
    client.connect()

    # ----------------------------------------------------------- #
    # Requests and Responses Examples
    # ----------------------------------------------------------- #
    log.debug("Write to a holding register and read back")
    # rq = client.write_register(1, 2, unit=UNIT)
    rr = client.read_holding_registers(0, 7, unit=UNIT)
    print(rr.registers)
    assert (not rr.isError())  # test that there's no error
  
    # ----------------------------------------------------------- #
    # Close Client
    # ----------------------------------------------------------- #
    client.close()

if __name__ == '__main__':
    run_client()
```

I don't think that the logs will be of help, since the Server stays on and the values read by the Modbus Client are from it's own Modbus map.

I couldn't find any Posts with similiar issues...
I'm a beginner in this whole Thing, I'd be happy if someone could help me with understanding the Problem. 
Thanks in Advance!

dhoomakethu, 2020-05-28:
Your port's looks different , that could be a typo. But without logs from both server and client, its hard to offer a solution.

nat-kh, 2020-05-28:
The server that I'm working with isn't a modbus server, but I want to make it one; I don't know if what I wrote above doesn't work because of that (since I need to create a modbus map for the controller in order to communicate with it, I assume).
 Would the simulator example help me furthter with creating a modbus map in the server and enabling it of a modbus communication? 
I managaed to write and read to the client's own Modbus map.

dhoomakethu, 2020-07-02:
@nat-kh I am not sure getting your problem, are you telling you want to mock a real modbus slave(server) ? A client in modbus world is someone who reads or writes to a slave(server) and it doesn't have a modbus map of its own.

nat-kh, 2020-07-08:
@dhoomakethu I know. My Goal is to make my Controller a Modbus server. That's why I wanted to build a map for the Server and wanted to read it's Contents with my client. In this case, my Client happens to be a controller, that's capable of being both a Server and a client. Even though I only want it to be a Client to my server, I don't know why I'm getting the Contents of it's table (as if  I was treating it as a server). 
My biggest Question would be is, is it enough to build a map for my server, for it to start communicating over modbus, or should it be simulated beforehand  with the the port 502 open?
Can I entirely implement Modbus in my Controller through python?

nat-kh, 2020-07-09:
@dhoomakethu another question: By Setting my server's port to 502,
 `server= ModbusTcpServer(context=context, identity=identity, address=('', 502))` 
I would be reading the values that I set in my context when I run client's Code (i.e client.read_holding_registers) 
but this seems to only work when my host is set as local host:
'client = ModbusClient(host='localhost', port=502)'
This doesn't seem to work with other IP addresses. How come?

dhoomakethu, 2020-07-09:
You can try `server= ModbusTcpServer(context=context, identity=identity, address=('0.0.0.0', 502))` to enable the server to listen on any enabled network interface , you can then connect to the server with `ModbusClient(host='localhost', port=502)` or `ModbusClient(host='<IP ADDRESS>', port=502)`

dhoomakethu, 2020-07-09:
Regarding your attempt to have both server and client on the same controller, checkout https://github.com/riptideio/pymodbus/blob/master/examples/contrib/remote_server_context.py and https://github.com/riptideio/pymodbus/blob/master/examples/contrib/serial_forwarder.py.

nat-kh, 2020-07-10:
@dhoomakethu Thank you for your help! 
One final question, can I access my local host Server, that I built in python from a remote Client's software and read the server's values in that software? 

Thank you for everything and sorry, I ended up asking a lot of things, since I was curious about a lot of stuff.

nat-kh, 2020-07-10:
@dhoomakethu  actually even in python, is it possible to have the localhost as the server, whereas an extern controller is the client?

nat-kh, 2020-07-10:
@dhoomakethu Thank you for your help! 
Thank you for everything and sorry, I ended up asking a lot of things, since I was curious about a lot of stuff.

closed 2020-07-10 by nat-kh