pymodbus-dev/pymodbus #507: Performance

Culla95, 2020-06-12:
### Versions

* Python: 3.7
* OS: Windows/Linux(on docker)
* Pymodbus:
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp- sync
* Client: tcp- sync

### Description

I am testing pymodbus on Docker. So I built 2 images both based on Ubuntu(latest) one for the server in which I run "synchronous_server.py" and the client where I run "performance.py". It actually works but the results are not even close from the ones benchmarked in the "performance.py" file.

It is said to be 31430 requests/second on TCP whereas my results are around 2000 requests/second if I run it on docker and 400 request/second if I run it on the host machine.

Docker: [https://imgur.com/Nq3pq51](url)
Host machine: [https://imgur.com/ExmdBoK](url)

So how am I supposed to achive those 31430 requests/second? Or what could I be doing wrong?

dhoomakethu, 2020-06-13:
@Culla95 Could you provide information of about your host (CPU/RAM) ? The seems pretty low for TCP.

Culla95, 2020-06-13:
@dhoomakethu i5-6600k 16gb, it shouldnt be such a problem

Culla95, 2020-07-14:
@dhoomakethu how did you managed that benchmark of 31430 request/second?

dhoomakethu, 2020-07-15:
@Culla95 I had my server and the performance monitoring script running on the same host to calculate those values.

Culla95, 2020-07-15:
@dhoomakethu and what kind of host was it? because I have tried that on my i5-6600k 16gb and also in AWS EC2 and the best I am getting is close to 2000 request/second. If you could replicate the performance test I would appreciate it a lot. Thank you

dhoomakethu, 2020-07-15:
@Culla95 I happen to run the tests again now and I am not able to reach even 1000 req/second on i5 8GB machine. Will have to check what changed between the last time the tests were performed against now.

Culla95, 2020-07-15:
@dhoomakethu ok thank you, when you do please let me know. Thanks once again.

dhoomakethu, 2020-07-15:
``` $ python performance.py
multiprocessing_test
1134 requests/second
time taken to complete 1000 cycle by 10 workers is 0.8814148902893066 seconds

thread_test
1174 requests/second
time taken to complete 1000 cycle by 10 workers is 0.8515892028808594 seconds

thread_pool_exe_test
1181 requests/second
time taken to complete 1000 cycle by 10 workers is 0.8464820384979248 seconds```

These are the values I get with asynchronous server

tracernz, 2020-07-16:
Even if you use uvloop it's not that much better.

Going back to older, with python 3.6.11 and pymodbus 1.3.2 I get 1413 req/s async and 1578 sync compared to 1642 async / 1520 sync with python 3.8.4 and pymodbus 2.3.0.