pymodbus-dev/pymodbus #508: Python Modbus String decoding issue

BenoistM, 2020-06-17:
### Versions

* Python: python 3.6.9
* OS: linux ubuntu server 20.04
* Pymodbus: 2.3.0

### Pymodbus Specific
* Server: tcp
* Client: tcp

### Description

I have an issue with the pymodbus decoder with strings. For example, when I try to read 'abcdefg' pymodbus gives me 'badcfehg'. The byteorder and the wordorder don't change the result.

Here is my code: 

### Code and Logs
```
result=client.read_holding_registers(25000,4)
decoder = BinaryPayloadDecoder.fromRegisters(result.registers,byteorder=Endian.Little,wordorder=Endian.Big)
decoder.decode_string(8)
```

The order does not change the result (Endian.Little or Endian.Big for both wordorder and byteorder). I try with the builder and it's the same problem. However, I don't have this problem with 32 bits floats for example.

I also tried with an older version of pymodbus (1.3.2) and it works (Endian.Little and Endian.big returns differents results).

tracernz, 2020-06-19:
The string decoder doesn't do any swapping, it just passes the underlying buffer through. https://github.com/riptideio/pymodbus/blob/5048a1a8da5558aa40d1d0f867982218c2096afe/pymodbus/payload.py#L495

BenoistM, 2020-06-19:
> 
> 
> The string decoder doesn't do any swapping, it just passes the underlying buffer through.
> 
> https://github.com/riptideio/pymodbus/blob/5048a1a8da5558aa40d1d0f867982218c2096afe/pymodbus/payload.py#L489-L496

Yes but the decoder should swap depending on the byteorder and wordorder values, is not it?

tracernz, 2020-06-20:
> Yes but the decoder should swap depending on the byteorder and wordorder values, is not it?

It's not. I agree that it should, because as you have discovered, devices do swap strings as well.

dhoomakethu, 2020-07-02:
@BenoistM I tried the [modbus_payload.py](https://github.com/riptideio/pymodbus/blob/master/examples/common/modbus_payload.py) with all possible combination of Wordorder and byteorder against a synchornous server and I could not reproduce your issue. Here are the logs from the script

```
------------------------------------------------------------
Word Order: BIG
Byte Order: BIG

------------------------------------------------------------
Writing Registers
------------------------------------------------------------
[24930, 25444, 25958, 26472, 23278, 4777, 34834, 13567, 65517, 52242, 13398, 30794, 11466, 11329, 45752, 21185, 45752, 21247, 65535, 65313, 21057, 4370, 13398, 30942, 44478, 61202, 13398, 30942, 44478, 61248, 24284, 52428, 52428, 52672, 24284, 52428, 52428, 52480]

------------------------------------------------------------
Registers
------------------------------------------------------------
[24930, 25444, 25958, 26472, 23278, 4777, 34834, 13567, 65517, 52242, 13398, 30794, 11466, 11329, 45752, 21185, 45752, 21247, 65535, 65313, 21057, 4370, 13398, 30942, 44478, 61202, 13398, 30942, 44478, 61248, 24284, 52428, 52428, 52672, 24284, 52428, 52428, 52480]

------------------------------------------------------------
Decoded Data
------------------------------------------------------------
string	 b'abcdefgh'
bits	 [False, True, False, True, True, False, True, False]
8int	 -0x12
8uint	 0x12
16int	 -0x5678
16uint	 0x1234
32int	 -0x1234
32uint	 0x12345678
16float	 12.34375
16float2	 -12.34375
32float	 22.34000015258789
32float2	 -22.34000015258789
64int	 -0xdeadbeef
64uint	 0x12345678deadbeef
ignore	 None
64float	 123.45
64float2	 -123.45
------------------------------------------------------------
Word Order: BIG
Byte Order: LITTLE

------------------------------------------------------------
Writing Registers
------------------------------------------------------------
[24930, 25444, 25958, 26472, 23278, 4744, 43316, 4863, 65484, 60724, 4728, 22060, 18988, 51890, 16722, 47282, 49490, 47359, 65535, 65362, 8465, 16692, 4728, 22189, 57071, 48692, 4728, 22189, 57071, 48734, 16588, 56524, 52429, 52318, 49356, 56524, 52429, 52224]

------------------------------------------------------------
Registers
------------------------------------------------------------
[24930, 25444, 25958, 26472, 23278, 4744, 43316, 4863, 65484, 60724, 4728, 22060, 18988, 51890, 16722, 47282, 49490, 47359, 65535, 65362, 8465, 16692, 4728, 22189, 57071, 48692, 4728, 22189, 57071, 48734, 16588, 56524, 52429, 52318, 49356, 56524, 52429, 52224]

------------------------------------------------------------
Decoded Data
------------------------------------------------------------
string	 b'abcdefgh'
bits	 [False, True, False, True, True, False, True, False]
8int	 -0x12
8uint	 0x12
16int	 -0x5678
16uint	 0x1234
32int	 -0x1234
32uint	 0x12345678
16float	 12.34375
16float2	 -12.34375
32float	 22.34000015258789
32float2	 -22.34000015258789
64int	 -0xdeadbeef
64uint	 0x12345678deadbeef
ignore	 None
64float	 123.45
64float2	 -123.45
------------------------------------------------------------
Word Order: LITTLE
Byte Order: BIG

------------------------------------------------------------
Writing Registers
------------------------------------------------------------
[24930, 25444, 25958, 26472, 23278, 4777, 34834, 13549, 52479, 65366, 30738, 13386, 11466, 11448, 21057, 45752, 21185, 45633, 4385, 21247, 65535, 65470, 61406, 44374, 30738, 13502, 61406, 44374, 30738, 13516, 52684, 52444, 52288, 24268, 52684, 52444, 52416, 24064]

------------------------------------------------------------
Registers
------------------------------------------------------------
[24930, 25444, 25958, 26472, 23278, 4777, 34834, 13549, 52479, 65366, 30738, 13386, 11466, 11448, 21057, 45752, 21185, 45633, 4385, 21247, 65535, 65470, 61406, 44374, 30738, 13502, 61406, 44374, 30738, 13516, 52684, 52444, 52288, 24268, 52684, 52444, 52416, 24064]

------------------------------------------------------------
Decoded Data
------------------------------------------------------------
string	 b'abcdefgh'
bits	 [False, True, False, True, True, False, True, False]
8int	 -0x12
8uint	 0x12
16int	 -0x5678
16uint	 0x1234
32int	 -0x1234
32uint	 0x12345678
16float	 12.34375
16float2	 -12.34375
32float	 22.34000015258789
32float2	 -22.34000015258789
64int	 -0xdeadbeef
64uint	 0x12345678deadbeef
ignore	 None
64float	 123.45
64float2	 -123.45
------------------------------------------------------------
Word Order: LITTLE
Byte Order: LITTLE

------------------------------------------------------------
Writing Registers
------------------------------------------------------------
[24930, 25444, 25958, 26472, 23278, 4744, 43316, 4812, 60927, 65400, 22068, 4652, 18988, 51794, 47282, 16722, 47282, 49425, 16722, 8703, 65535, 65519, 48813, 56952, 22068, 4847, 48813, 56952, 22068, 4813, 52428, 52428, 56414, 16589, 52428, 52428, 56414, 49152]

------------------------------------------------------------
Registers
------------------------------------------------------------
[24930, 25444, 25958, 26472, 23278, 4744, 43316, 4812, 60927, 65400, 22068, 4652, 18988, 51794, 47282, 16722, 47282, 49425, 16722, 8703, 65535, 65519, 48813, 56952, 22068, 4847, 48813, 56952, 22068, 4813, 52428, 52428, 56414, 16589, 52428, 52428, 56414, 49152]

------------------------------------------------------------
Decoded Data
------------------------------------------------------------
string	 b'abcdefgh'
bits	 [False, True, False, True, True, False, True, False]
8int	 -0x12
8uint	 0x12
16int	 -0x5678
16uint	 0x1234
32int	 -0x1234
32uint	 0x12345678
16float	 12.34375
16float2	 -12.34375
32float	 22.34000015258789
32float2	 -22.34000015258789
64int	 -0xdeadbeef
64uint	 0x12345678deadbeef
ignore	 None
64float	 123.45
64float2	 -123.45
```

tracernz, 2020-07-03:
@BenoistM Are you reading the string from a 3rd party device, or a pymodbus server instance?

BenoistM, 2020-07-03:
Thank you for your tests. As your log shows, why the encoded registers are the same whatever the wordorder and byteorder? If you try to encode 'abcdefg' with different wordorder/byteorder the writing registers should not be the same is not it? 

I try to read the values from a schneider PLC, I can't define the byte/wordorder of the registers. I temporarily solve this issue with the python to_bytes function where the byteorder is working well.

tracernz, 2020-07-03:
Which family? Quantum, M340/580? I can do some tests at work on those.

BenoistM, 2020-07-03:
> Which family? Quantum, M340/580? I can do some tests at work on those.

It's an M340

dhoomakethu, 2020-07-04:
@BenoistM That's something I am currently looking in to. The issue is still open. Can you share the logs of your reads and write with the PLC please ?

BenoistM, 2020-07-04:
I can't write in this PLC, I only do reading:

Here is my code:

```from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.payload import BinaryPayloadBuilder
from pymodbus.constants import Endian
client=ModbusClient('192.168.x.x')
result=client.read_holding_registers(25000,10)
print("###### Registers")
print(result.registers)
decoder = BinaryPayloadDecoder.fromRegisters(result.registers,byteorder=Endian.Little,wordorder=Endian.Big)
print("###### Order: Little - Big")
decoder.decode_string(20)
decoder = BinaryPayloadDecoder.fromRegisters(result.registers,byteorder=Endian.Big,wordorder=Endian.Big)
print("###### Order: Big - Big")
decoder.decode_string(20)
decoder = BinaryPayloadDecoder.fromRegisters(result.registers,byteorder=Endian.Big,wordorder=Endian.Little)
print("###### Order: Big - Little")
decoder.decode_string(20)
decoder = BinaryPayloadDecoder.fromRegisters(result.registers,byteorder=Endian.Little,wordorder=Endian.Little)
print("###### Order: Little - Little")
decoder.decode_string(20)
```

and the logs:
```
###### Registers
[25956, 29287, 27753, 25964, 29301, 0, 0, 0, 0, 0]
###### Order: Little - Big
b'edrglielru\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
###### Order: Big - Big
b'edrglielru\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
###### Order: Big - Little
b'edrglielru\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
###### Order: Little - Little
b'edrglielru\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

```

The original word is: "degrilleur" (+empty bytes). I did tests on encoding only (I did not write in the PLC). The results of the encoding were the same as @dhoomakethu (the registers were the same, whatever the order).

tracernz, 2020-07-05:
M340:
[image]

```
>>> BinaryPayloadDecoder.fromRegisters(c.read_holding_registers(0, 5).registers, byteorder=Endian.Little).decode_string(10)
b'ehllworodl'
>>> BinaryPayloadDecoder.fromRegisters(c.read_holding_registers(0, 5).registers, byteorder=Endian.Big).decode_string(10)
b'ehllworodl'
```

Quantum:
[image]

```
>>> BinaryPayloadDecoder.fromRegisters(c.read_holding_registers(0, 5).registers, byteorder=Endian.Little).decode_string(10)
b'ehllworodl'
>>> BinaryPayloadDecoder.fromRegisters(c.read_holding_registers(0, 5).registers, byteorder=Endian.Big).decode_string(10)
b'ehllworodl'
```

The bytes definitely need to be swapped for Schneider's string type.

dhoomakethu, 2020-07-05:
Thanks for the logs @BenoistM and for the analysis @tracernz . I will add this to 3.x release

flyinggorilla, 2020-07-16:
ran into same bug

as a workaround I had to manually reverse the byteorder
```
    result = client.read_holding_registers(310,5, unit = UNIT)
    decoder = BinaryPayloadDecoder(b''.join(pack(Endian.Little + 'H', x) for x in result.registers)) # reverse byteorder
    print("Device Name: {}".format(decoder.decode_string(10).decode('ascii')))

```

MountainKing91, 2020-11-21:
Hi,
a bit late to the party but yes, devices sometimes swap bytes in strings too. I ran into a situation (production environment) where a bunch of strings were written from a PC with MoviconNext to a Delta PLC and they came up swapped in the HMI. I used pymodbus to debug what was going on and I used this function to sistematically swap bytes:
`
def swap_bytes(registers):
    swapped = []
    for r in registers:
        k = struct.pack('<H', r)
        k = int.from_bytes(k, 'big', signed=False)
        swapped.append(k)
    return swapped
`
It worked fine.

janiversen, 2022-05-25:
Closing due to missing information.

closed 2022-05-25 by janiversen