pymodbus-dev/pymodbus #513: asyncio.coroutine has been deprecated since Python 3.8 and will be removed in future

tirkarthi, 2020-07-19:
### Versions

* Python: 3.8
* OS: Ubuntu
* Pymodbus: master branch
* Modbus Hardware (if used): 

### Description

asyncio.coroutine has been deprecated since Python 3.8 and will be removed in future. Use `async def` instead.

### Code and Logs

```
pymodbus/client/asynchronous/asyncio/__init__.py
249:    @asyncio.coroutine
286:    @asyncio.coroutine
329:    @asyncio.coroutine
379:    @asyncio.coroutine
462:    @asyncio.coroutine
509:    @asyncio.coroutine
554:    @asyncio.coroutine
611:    @asyncio.coroutine
710:    @asyncio.coroutine
760:@asyncio.coroutine
777:@asyncio.coroutine

test/test_server_asyncio.py
69:    @asyncio.coroutine
80:    @asyncio.coroutine
88:    @asyncio.coroutine
96:    @asyncio.coroutine
109:    @asyncio.coroutine
143:    @asyncio.coroutine
183:    @asyncio.coroutine
216:    @asyncio.coroutine
248:    @asyncio.coroutine
286:    @asyncio.coroutine
322:    @asyncio.coroutine
361:    @asyncio.coroutine
407:    @asyncio.coroutine
425:    @asyncio.coroutine
433:    @asyncio.coroutine
449:    @asyncio.coroutine
464:    @asyncio.coroutine
485:    @asyncio.coroutine
527:    @asyncio.coroutine
565:    @asyncio.coroutine

test/asyncio_test_helper.py
24:            @asyncio.coroutine
```

JamesJeffryes, 2020-09-14:
duplicates #467 but with more complete information.

swamper123, 2021-03-04:
This issue is already solved in branch 3.0.

closed 2022-03-31 by swamper123

dhoomakethu, 2022-03-31:
Please be aware, we are supporting 3.7 also so removing co-routines completely would break things for pymodbus on python3.7

swamper123, 2022-03-31:
Isn't this decorator not just a compatible thing to async/await syntax?
Since my PR is a while ago, I will take a look again if I have fixed that the way it was right.

swamper123, 2022-04-01:
Well, it looks like I have just replaced one decorator, but replaced it with modern syntax.
I guess this is for codecompatibility with older versions, but isn't used anymore and not important for >= 3.7.
Even you replaced one last year. ;)

dhoomakethu, 2022-04-01:
My memory is fading :P , Thanks for taking care.