pymodbus-dev/pymodbus #516: Custom Datablock Issue

kile9, 2020-07-23:
I'm trying to develop custom datablock functionality using pyModbus and am following the example in the documentation: https://pymodbus.readthedocs.io/en/latest/source/example/custom_datablock.html

Unlike the example, I'm using the serial server (StartSerialServer) with RTU framer.
`StartSerialServer(context, framer=ModbusRtuFramer, identity=identity)`

When I run the code, I get errors every time a message is received. As far as I can tell, it's a failure of the response packet because the 'getValue' function is not specified.

`DEBUG:pymodbus.server.asynchronous:Datastore unable to fulfill request: Modbus Error: [Not Implemented] Datastore Value Retrieve`

I'm not looking to do anything custom for the retrieval of the data, so I don't understand why it can't inherit the getValues() functionality from the ModbusSpareDataBlock class. I tried adding an explicit overload for getValues() but that didn't resolve anything.

Either this is a bug or perhaps I don't quite understand how function overloads occur and the structure of the pyModbus stack. Any suggestions would be much appreciated.

Version info: Python 3.8.0 (tags/v3.8.0:fa919fd, Oct 14 2019, 19:37:50) [MSC v.1916 64 bit (AMD64)]

Here's the code:

```
from __future__ import print_function
from pymodbus.server.asynchronous import StartSerialServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSparseDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.transaction import ModbusRtuFramer

# --------------------------------------------------------------------------- # 
# configure the service logging
# --------------------------------------------------------------------------- # 

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

# serial stuff
import serial.tools.list_ports
serialports = [port.device for port in serial.tools.list_ports.comports()]

# --------------------------------------------------------------------------- # 
# create your custom data block here
# --------------------------------------------------------------------------- # 

class CustomDataBlock(ModbusSparseDataBlock):
    """ A datablock that stores the new value in memory
    and performs a custom action after it has been stored.
    """

    def setValues(self, address, value):
        """ Sets the requested values of the datastore

        :param address: The starting address
        :param values: The new values to be set
        """
        super(ModbusSparseDataBlock, self).setValues(address, value)

        # whatever you want to do with the written value is done here,
        # however make sure not to do too much work here or it will
        # block the server, espectially if the server is being written
        # to very quickly
        print("wrote {} to {}".format(value, address))
        
    def getValues(self, address, values):
        super(ModbusSparseDataBlock, self).getValues(address, values)

def run_custom_db_server():
    # ----------------------------------------------------------------------- # 
    # initialize your data store
    # ----------------------------------------------------------------------- # 
    block  = CustomDataBlock({0x01: 0})

    store  = ModbusSlaveContext(hr=block)
    context = ModbusServerContext(slaves=store, single=True)

    # ----------------------------------------------------------------------- #
    # initialize the server information
    # ----------------------------------------------------------------------- #

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/bashwork/pymodbus/'
    identity.ProductName = 'pymodbus Server'
    identity.ModelName = 'pymodbus Server'
    identity.MajorMinorRevision = '2.3.0'

    # ----------------------------------------------------------------------- #
    # run the server you want
    # ----------------------------------------------------------------------- #

    # p = Process(target=device_writer, args=(queue,))
    # p.start()
    StartSerialServer(context, framer=ModbusRtuFramer, identity=identity, #timeout=0,
                       port=serialports[0], baudrate=14400, parity='E')

if __name__ == "__main__":
    run_custom_db_server()
```

dhoomakethu, 2020-07-24:
It is because you are calling the `get_value` from the parent of `ModbusSparseDataBlock` which has these methods unimplemented.  If you change `super(ModbusSparseDataBlock, self).getValues(address, values)` to `super(CustomDataBlock, self).getValues(address, values)` . Your example should work as expected.

closed 2020-07-24 by dhoomakethu

tracernz, 2020-07-24:
Also need to implement validate I think? Unless you only want the registers passed into the ModbusSparseDataBlock initialiser to work.

kile9, 2020-07-28:
Thanks, everyone. Would you like me to correct the example?

dhoomakethu, 2020-07-28:
@kile9 it will be taken care in the next release, if you have the code ready.Please raise a PR.