pymodbus-dev/pymodbus #518: RTU Async Server not working properly

szarkadani, 2020-07-31:
## Versions

* Python: 3.7.3
* OS: Raspbian GNU/Linux 10
* Pymodbus: 2.3
* Modbus Hardware (if used): Waveshare can rs485 hat

### Pymodbus Specific
* Server: rtu - async
* Client: rtu- sync

### Description
I'm using 2 Pis for server and client over RTU. 
I want to make a server that updates some registers from redis and the client just reads it, but before trying to achieve the loop and a redis get I want to make sure the reading works properly. Well it doesn't.

The reading sometimes works and sometimes not but only when baudrate = 210600. It's ridiculously high, and I think  it's unnecessary. In the log you can see that it worked only on the 3rd time.
### Code and Logs
Server:
```python

from pymodbus.server.asynchronous import StartSerialServer

from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.transaction import (ModbusRtuFramer)

import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s'
          ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def run_async_server():
    store = ModbusSlaveContext(
        di=ModbusSequentialDataBlock(0, [17]*100),
        co=ModbusSequentialDataBlock(0, [17]*100),
        hr=ModbusSequentialDataBlock(0, [17]*100),
        ir=ModbusSequentialDataBlock(0, [17]*100)
    )
    context = ModbusServerContext(slaves=store, single=True)
    StartSerialServer(context, identity=None, port='/dev/serial0', framer=ModbusRtuFramer,
			baudrate=210600, timeout=1)

if __name__ == "__main__":
    run_async_server()
```
Client
```python
#!/usr/bin/env python

from pymodbus.client.sync import ModbusSerialClient as ModbusClient
from pymodbus.transaction import ModbusRtuFramer

import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

UNIT = 1

def run_sync_client():

    client = ModbusClient(method='rtu', port='/dev/serial0', timeout=2,
                           baudrate=210600)
    client.strict = False
    client.connect()
[... 78 lines cut ...]
  File "client.py", line 39, in run_sync_client
    print(rr.registers)
AttributeError: 'ModbusIOException' object has no attribute 'registers'

pi@raspberrypi:~/Desktop/modbus $ python3 client.py

2020-07-31 16:34:01,277 MainThread      DEBUG    client         :35       Write to a holding register and read back
2020-07-31 16:34:01,278 MainThread      DEBUG    transaction    :115      Current transaction state - IDLE
2020-07-31 16:34:01,278 MainThread      DEBUG    transaction    :120      Running transaction 1
2020-07-31 16:34:01,279 MainThread      DEBUG    transaction    :219      SEND: 0x1 0x3 0x0 0x1 0x0 0x1 0xd5 0xca
2020-07-31 16:34:01,279 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2020-07-31 16:34:01,280 MainThread      DEBUG    transaction    :228      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-07-31 16:34:01,286 MainThread      DEBUG    transaction    :304      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2020-07-31 16:34:01,286 MainThread      DEBUG    transaction    :233      RECV: 0x1 0x3 0x2 0x0 0x11 0x78 0x48
2020-07-31 16:34:01,287 MainThread      DEBUG    rtu_framer     :180      Getting Frame - 0x3 0x2 0x0 0x11
2020-07-31 16:34:01,287 MainThread      DEBUG    factory        :266      Factory Response[ReadHoldingRegistersResponse: 3]
2020-07-31 16:34:01,288 MainThread      DEBUG    rtu_framer     :115      Frame advanced, resetting header!!
2020-07-31 16:34:01,288 MainThread      DEBUG    transaction    :383      Adding transaction 1
2020-07-31 16:34:01,288 MainThread      DEBUG    transaction    :394      Getting transaction 1
2020-07-31 16:34:01,289 MainThread      DEBUG    transaction    :193      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
---------------
ReadRegisterResponse (1)
[17]
---------------
```

dhoomakethu, 2020-08-01:
Pymodbus should work with any baudrate host is supports. The problem with serial ports is that you will have to make the lines are terminated properly. Your best would be to go with TCP server/Client. Also, you can check if the received response is error with `rr.isError()` and discard those responses.

szarkadani, 2020-08-01:
Serial is a must, I won't go with TCP. I'm going to check response error, but the question is why did it work for the 3rd time and if I change the baud rate why would the server receive the messages in more parts? 
I mean you can see the first 2 response from the server is way different than the 3rd. If I change the baud rate, the error from Data Received: 0x1 0x3 0x0 0x1 0x0 0x1 0xd5 0xca goes to Data Received: 0x1... What happens if I set the baud rate that insanly high and reads for the 3rd time?

szarkadani, 2020-08-04:
Update: Version: 2.4.0rc1 working flawlessly with baud rate 19200. 👍

closed 2020-08-04 by szarkadani