pymodbus-dev/pymodbus #519: Pymodbus DE/RE RS485 pin control

BullT, 2020-08-01:
Hi,

How can I control the DE/RE pin in my RS485 board if I using the Pymodbus library?

Can I configure the Pymodbus for use the DE/RE pin automaticly?

Thanks, XCO

dhoomakethu, 2020-08-01:
Pymodbus uses pyserial to communicate over serial bus and accepts all the parameters pyserial takes . Refer https://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.Serial .

Simonefardella, 2020-09-22:
@dhoomakethu i have a MAX485 module wired like this one: 
[image]

Actually i'm not able to figure out how to control the RE/DE pin from the library using the [Synchronous Client Example](https://pymodbus.readthedocs.io/en/latest/source/example/synchronous_client.html)

How i can control the RE/DE pin using a common (in my case the 23) GPIO? 
Thanks!

closed 2021-03-08 by github-actions[bot]