pymodbus-dev/pymodbus #521: customized  Modbus protocol implementation.

abdulmannanrauf, 2020-08-01:
I need to implement a customized Modbus protocol implementation with pymodbus.
The message structure to read the hold register data is:

address (1byte) + func code (1byte) + Serial No:(10 bytes)               +      start reg: address (2 bytes) +length ( 2bytes)+ checksum

    00                      03                          31 32 31 34 33 32 34 33 35 33              00 00                                17 00                6E 25

There is an addiiton of 10 byte serial no after function code.
Can we implement such modified modbus protocol using pymodbus.
 If not whats the recommendation

dhoomakethu, 2020-08-01:
You will have to create a custom framer with your requirements deriving from `ModbusFramer` or in your case `ModbusRtuFramer` class and implement the required methods. Refer https://github.com/riptideio/pymodbus/tree/master/pymodbus/framer for the examples.

abdulmannanrauf, 2020-08-02:
Many Thanks @dhoomakethu. I spent quite some time on it however still struggling. Can you also help by sharing an example simulation (generator) using rtuframer.py  . i.e https://github.com/riptideio/pymodbus/blob/master/pymodbus/framer/rtu_framer.py

I  would like to test the example with virtual serial port driver (implementing both master and slave).

Thank a lot

closed 2020-12-30 by dhoomakethu