pymodbus-dev/pymodbus #522: client.read_holding_registers（）Why can't a single device read more than 120 points

luli440, 2020-08-07:
### Versions

* Python:
* OS:
* Pymodbus:
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
 # code and logs here.

```

xvart, 2020-08-07:
I guess read the spec, there is a maximum packet length defined for response to function code 23(0x17) of 8bits so 256 bytes which works out to around 120 registers.

Weirdly had some one from Schneider ask this same question.

luli440, 2020-08-07:
> 我猜想已经阅读了规范，为响应8位功能代码23（0x17）定义了最大数据包长度，因此需要256个字节，可以处理大约120个寄存器。
> 
> 奇怪的是有一个来自施耐德的人问了同样的问题。

Where can I change the maximum packet length?

tracernz, 2020-08-07:
You can read 125 registers or 2000 coils, per the [protocol specification](https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf).

closed 2020-08-07 by dhoomakethu