pymodbus-dev/pymodbus #523: IS there any system configuration requirements for running pymodbus efficiently?

AnkitaPadma, 2020-08-07:
### Versions

* Python:
* OS:
* Pymodbus:
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
 # code and logs here.

```

AnkitaPadma, 2020-08-07:
@dhoomakethu  Do you have any idea about this. Please suggest

dhoomakethu, 2020-08-07:
What client are you talking about here ? Can you provide more information on your observation? What was the metrics used to identify the slowing down of communication?

AnkitaPadma, 2020-08-16:
@dhoomakethu - So I m using pymodbus with django rest framework. So what I noticed was that first time when I had installed pymodbus on my laptop the performance was very good..as to the read and write operations. The speed was decent.

Now when I reinstalled the setup in new laptop, the communication speed has slowed down to a large extent, that the reads itself for a single register is taking over a min.
Additionally I would like to point that communication works perfectly fine when working with Modbus tool or minimal modbus.
So I m not able to figure out what might be the issue.

 I m very new to registers and modbus communication. So if you could let me what is that I can provide info on that would help you understand the issue.
Or if any of the things I mentioned makes sense, then please suggest a work around.

Thanks

dhoomakethu, 2020-08-17:
@AnkitaPadma Can you run this script https://pymodbus.readthedocs.io/en/latest/source/example/performance.html ? Please update the script with correct serial settings and register offsets. I am not sure how you are using pymodbus in your code. When you say the it works perfectly fine with modbus tool or minimal modbus , are you running the same code (read your application) with pymodbus replaced with minimal modbus or is it the standalone code that you are talking. If yes, please share the minimal modbus code you have used as well the pymodbus code.

AnkitaPadma, 2020-08-17:
@dhoomakethu Yes using the same code, by just changing the library from pymodbus to minimal
For reference

 Pymodbus code:   
    client = ModbusSerialClient(
        method='rtu',
        port=meter.port,
        baudrate=9600,
        timeout=60,
        parity='N',
        stopbits=2,
        bytesize=8
    )
    if client.connect():
        readVal = client.read_input_registers(reg.regAddress, 2, unit=1)
        while readVal.isError():
            readVal = client.read_input_registers(reg.regAddress, 2, unit=1)
            continue

        print(readVal )
        client.close()
    else:
        print('Cannot connect to the Modbus Server/Slave')
    
Minimal modbus code:
instrument = minimalmodbus.Instrument(meter.port, 1)
    instrument.serial.port = meter.port                    
    instrument.serial.baudrate = 9600        
    instrument.serial.bytesize = 8
    instrument.serial.parity   = serial.PARITY_NONE
    instrument.serial.stopbits = 2
    instrument.serial.timeout  = 0.05          
    instrument.mode = minimalmodbus.MODE_RTU   

    try:
        readVal = instrument.read_float(reg.regAddress,4)
   
        print(readVal)
    except IOError:
        print("Failed to read from instrument")

    instrument.serial.close()

dhoomakethu, 2020-08-17:
Timeout of 60s for pymodbus and 0.5 sec for minimal modbus. Is there a reason for this? What happens if you use timeout of `0.5` sec with pymodbus as well ? Also what is the slave device under test here ?

AnkitaPadma, 2020-08-17:
For testing if the connection is getting timeout, i had set 60s in pymodbus, as it wasnt responding only.
Energy meter is the slave device I m testing here

dhoomakethu, 2020-08-17:
installation process remains the same, `pip install pymodbus` `strict=False` would make the connections kind of stable on windows machine (just a hack). What is the make of energy meter/spec etc?

closed 2021-03-08 by github-actions[bot]