pymodbus-dev/pymodbus #524: ReadDeviceInformationRequest (Frame check failed, ignoring!!)

Fnikan, 2020-08-10:
### Versions

* Python:3.7
* OS:Windows
* Pymodbus: RTU
* Modbus Hardware (if used): RS485 USB

### Pymodbus Specific

* Client: rtu

### Description

Hi, 
i want to get the device identifications from a DGZ-meter [https://www.dzg.de/fileadmin/dzg/content/downloads/produkte-zaehler/dvh4013/DZG_Modbus_Protocol.pdf](url) but i get frame check failure.
actually what i recieve as the buffer is correct. but i have no idea why i get this error.
I tried also the strict=False, timeout= 5 but nothing changed. I think it is on calculateRtuFrameSize(cls, buffer).
thank you for your response!

### Code and Logs

```python
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
from pymodbus.mei_message import ReadDeviceInformationRequest
import logging

FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

comm = ModbusClient(method='rtu', port='COM8', baudrate=9600, stopbits=2, bytesize=8, parity='E', strict=False, timeout= 1)
comm.connect()
req =ReadDeviceInformationRequest(unit=53 , object_id=0)
w = comm.execute(req)
print(w.information)

```
2020-08-10 16:36:47,099 MainThread      DEBUG    transaction    :115      Current transaction state - IDLE
2020-08-10 16:36:47,099 MainThread      DEBUG    transaction    :120      Running transaction 1
2020-08-10 16:36:47,099 MainThread      DEBUG    transaction    :219      SEND: 0x35 0x2b 0xe 0x1 0x0 0xc1 0xb3
2020-08-10 16:36:47,099 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2020-08-10 16:36:47,100 MainThread      DEBUG    transaction    :228      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
2020-08-10 16:36:47,263 MainThread      DEBUG    transaction    :304      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2020-08-10 16:36:47,263 MainThread      DEBUG    transaction    :233      RECV: 0x35 0x2b 0xe 0x1 0x1 0x0 0x0 0x3 0x0 0xa 0x44 0x5a 0x47 0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x1 0xa 0x44 0x56 0x48 0x34 0x30 0x31 0x33 0x47 0x31 0x35
2020-08-10 16:36:47,263 MainThread      DEBUG    rtu_framer     :232      Frame check failed, ignoring!!
2020-08-10 16:36:47,263 MainThread      DEBUG    rtu_framer     :128      Resetting frame - Current Frame in buffer - 0x35 0x2b 0xe 0x1 0x1 0x0 0x0 0x3 0x0 0xa 0x44 0x5a 0x47 0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x1 0xa 0x44 0x56 0x48 0x34 0x30 0x31 0x33 0x47 0x31 0x35
2020-08-10 16:36:47,263 MainThread      DEBUG    transaction    :394      Getting transaction 53
2020-08-10 16:36:47,263 MainThread      DEBUG    transaction    :193      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'

dhoomakethu, 2020-08-13:
What version of pymodbus is this ? You are not getting the complete response back from the device and there is a CRC checksum failure.

Fnikan, 2020-08-19:
> 
> 
> What version of pymodbus is this ? You are not getting the complete response back from the device and there is a CRC checksum failure.

Hi, i use pymodbus 2.3

closed 2021-03-03 by dhoomakethu