pymodbus-dev/pymodbus #525: Read RTU Holding Register through Serial Forwarder/TCP.

Abbad-Zafar, 2020-08-11:
I have Energy Meter contented through RTU and able to get holding registers data through simple RTU Code. 
Now i want to make Convert this RTU to TCP through Forwarder. I want to send data to TCP which forwards the command to RTU and fetches data for me

**Code for Simple RTU Read**

- import pymodbus
from pymodbus.pdu import ModbusRequest
from pymodbus.client.sync import ModbusSerialClient as ModbusClient 
#initialize a serial RTU client instance
from pymodbus.transaction import ModbusRtuFramer

#count= the number of registers to read
#unit= the slave unit this request is targeting
#address= the starting address to read from

client = ModbusClient(method = 'rtu', port='/dev/extcomm/0/0', stopbits = 1, bytesize = 8, parity = 'N' , baudrate= 38400)

#Connect to the serial modbus server
 connection = client.connect()
 print(connection)

#Starting add, num of reg to read, slave unit.
read  = client.read_holding_registers(address = 0x01,count =2, unit=1)
data = read.registers

print(data)

#Closes the underlying socket connection
client.close()

Abbad-Zafar, 2020-08-11:
@dhoomakethu

dhoomakethu, 2020-08-11:
@Abbad-Zafar Do you get any errors when using the serial_forwarder example? Any logs would be helpful.

dhoomakethu, 2020-08-11:
The serial_forwarder example basically ignores the slave-id's (refer `single=True`) . If you want to make the forwarder specific to your device address, you will have to do something like `context = ModbusServerContext(slaves={<device_id>: store, single=False)` . This would make the serial forwarder work only for the device id specified, If you want to handle multiple devices then keep updating the store dict (Make sure you have client defined for each device).

```def run_serial_forwarder():
    # ----------------------------------------------------------------------- #
    # initialize the datastore(serial client)
    # ----------------------------------------------------------------------- #
    client = ModbusClient(method='rtu', port='/dev/ptyp0')
    store = RemoteSlaveContext(client)
    context = ModbusServerContext(slaves=store, single=True)
    # or
    # The below line will make the serial_forwarder accept requests only for device with id 2 and will discard every other request.
    # context = ModbusServerContext(slaves={2: store}, single=True)

    # ----------------------------------------------------------------------- #
    # run the server you want
    # ----------------------------------------------------------------------- #
    StartServer(context, address=("localhost", 5020))```

Abbad-Zafar, 2020-08-11:
Thanks for the reply Sir
Here is Code of Forwarder that i am now using :
    client = ModbusClient(method='rtu', port='/dev/ttyUSB0')
    store = RemoteSlaveContext(client)
    context = ModbusServerContext(slaves = {0x01: store}, single=False)
    # ----------------------------------------------------------------------- #
    # run the server you want
    # ----------------------------------------------------------------------- #
    StartServer(context, address=("localhost", 5010))

Here is Client Code:
host = "localhost"
    
port = 5010 

client = ModbusClient(host, port, timeout =10)
    
client.connect()
    
print(client.connect())

read = client.read_holding_registers(address = 0x001, count =65, unit=1)
print(read)

Its connecting to server i think but When i print read it says ExceptionResponse(31,3, IllegalAdrress).

Abbad-Zafar, 2020-08-11:
My Device Id is 0x01 or 1. The register address is also fine because i have tested it with simple RTU read code.

dhoomakethu, 2020-08-11:
Illegal address is when you try to access the wrong register offsets. Are
you sure the reading works directly with serial clients? Can you share the
logs with serial client accessing directly and with serial forwarder please?

ಅವರು ಬರೆದಿದ್ದಾರೆ:

Abbad-Zafar, 2020-08-11:
Yes Sir it working fine with simple RTU Read.
Simple RTU Read Code:
client = ModbusClient(method='rtu', port='/dev/ttyUSB0', baudrate=9600, timeout=0.5)

client.connect()

read=client.read_holding_registers(address = 0x0010, count =2, unit=1)

data=read.registers

print(data) #print register data

client.close()

It returns  [0, 226]

Abbad-Zafar, 2020-08-11:
**Log of siple RTU Read:**
> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x0 0x10 0x0 0x2 0xc5 0xce
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
> DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x4 0x0 0x0 0x0 0xe2 0x7a 0x7a
> DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x4 0x0 0x0 0x0 0xe2
> DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
> DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
> DEBUG:pymodbus.transaction:Adding transaction 1
> DEBUG:pymodbus.transaction:Getting transaction 1
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> [0, 226]
**Log of Forwarder:**
DEBUG:pymodbus.server.sync:Started thread to serve client at ('127.0.0.1', 54467)
DEBUG:pymodbus.server.sync:Client Connected [127.0.0.1:54467]
DEBUG:pymodbus.server.sync:Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x1 0x0 0x41
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x1 0x0 0x41
DEBUG:pymodbus.factory:Factory Request[ReadHoldingRegistersRequest: 3]
DEBUG:pymodbus.datastore.remote:validate[3] 1:65
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x0 0x1 0x0 0x41 0xd4 0x3a
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)) 
DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
ERROR:pymodbus.pdu:Exception Response(131, 3, IllegalAddress)
DEBUG:pymodbus.server.sync:send: [Exception Response(131, 3, IllegalAddress)]- b'000100000003018302'
DEBUG:pymodbus.server.sync:Handling data: 
DEBUG:pymodbus.framer.socket_framer:Processing: 
DEBUG:pymodbus.server.sync:Client Disconnected [127.0.0.1:54467]

dhoomakethu, 2020-08-11:
Thanks, My guess is that the `offset` and `count` combination you are using with serial forwarder is invalid. Can you try the reads when using the serial forwarder `(address=0x10, count=2, unit=1)` and  `(address=0x01, count=0x41, unit=1)` directly with serial client.  Please share the modbus specs ,if you have it handy,

Abbad-Zafar, 2020-08-11:
- using the serial forwarder (address=0x10, count=2, unit=1) NOT WORKING same Error.
- using Serial Client (address=0x01, count=0x41, unit=1) Working and getting response  [0, 1, 0, 1, 0, 1].

using this energy meter, reading voltage at address 0010: 
https://owen-brothers.co.uk/pub/media/wysiwyg/OB418_User_Manual.pdf

dhoomakethu, 2020-08-11:
Hmm, interesting, can you share the logs for serial forwarder for `(address=0x10, count=2, unit=1)` ? and logs for serial client for `(address=0x01, count=0x41, unit=1) ` for comparision.

dhoomakethu, 2020-08-11:
Also just for testing, can you try serial_forwarder with address +/- 1 compared to what you will do with regular serial client i.e use address (0xf and 0x11).

Abbad-Zafar, 2020-08-12:
**logs for serial forwarder for (address=0x10, count=2, unit=1)**

> DEBUG:pymodbus.server.sync:Started thread to serve client at ('127.0.0.1', 51775)
> DEBUG:pymodbus.server.sync:Client Connected [127.0.0.1:51775]
> DEBUG:pymodbus.server.sync:Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.factory:Factory Request[ReadInputRegistersRequest: 4]
> DEBUG:pymodbus.datastore.remote:validate[4] 16:2
> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: 0x0 0x4 0x0 0x10 0x0 0x2 0x71 0xdf
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)) 
> DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
> DEBUG:pymodbus.transaction:Getting transaction 0
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> ERROR:pymodbus.pdu:Exception Response(132, 4, IllegalAddress)
> DEBUG:pymodbus.server.sync:send: [Exception Response(132, 4, IllegalAddress)]- b'000100000003018402'
> DEBUG:pymodbus.server.sync:Handling data: 
> DEBUG:pymodbus.framer.socket_framer:Processing: 
> DEBUG:pymodbus.server.sync:Client Disconnected [127.0.0.1:51775]
> DEBUG:pymodbus.server.sync:Started thread to serve client at ('127.0.0.1', 46459)
> DEBUG:pymodbus.server.sync:Client Connected [127.0.0.1:46459]
> DEBUG:pymodbus.server.sync:Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.factory:Factory Request[ReadInputRegistersRequest: 4]
> DEBUG:pymodbus.datastore.remote:validate[4] 16:2
> DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
> DEBUG:pymodbus.transaction:Running transaction 2
> DEBUG:pymodbus.transaction:SEND: 0x0 0x4 0x0 0x10 0x0 0x2 0x71 0xdf
> DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - None, Current Time stamp - 1597217842.584101
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Incomplete message received, Expected 9 bytes Recieved 0 bytes !!!!
> DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
> DEBUG:pymodbus.transaction:RECV: 
> DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
> DEBUG:pymodbus.transaction:Getting transaction 0
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> ERROR:pymodbus.pdu:Exception Response(132, 4, IllegalAddress)
> DEBUG:pymodbus.server.sync:send: [Exception Response(132, 4, IllegalAddress)]- b'000100000003018402'
> DEBUG:pymodbus.server.sync:Handling data: 
> DEBUG:pymodbus.framer.socket_framer:Processing: 
> DEBUG:pymodbus.server.sync:Client Disconnected [127.0.0.1:46459]

**Logs for Serial RTU  for (address=0x01, count=0x41, unit=1)**

> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x0 0x1 0x0 0x41 0xd4 0x3a
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Incomplete message received, Expected 135 bytes Recieved 17 bytes !!!!
> DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
> DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0xc 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x1 0x62 0xe0
> DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0xc 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x1
> DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
> DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
> DEBUG:pymodbus.transaction:Adding transaction 1
> DEBUG:pymodbus.transaction:Getting transaction 1
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> [0, 1, 0, 1, 0, 1]

Abbad-Zafar, 2020-08-12:
@dhoomakethu Have you yourself tested any Serial forwarder code to forward rtu to tcp ? If yes could you send me that. Thanks

dhoomakethu, 2020-08-12:
@Abbad-Zafar I think your serial forwarder is working fine.  Looks like you are using wrong function code while polling tcp server. I see you are using function code 4 (Read Input Register) while reading from the forwarder.

Please try these against the tcp server

```
client = ModbusTcpClient(<address>, <port>)
read = client.read_holding_registers(address = 0x10, count =2, unit=1)

## And 
client = ModbusSerialClient(method="rtu", baudrate=9600, port=<port>, timeout=2)
read = client.read_holding_registers(address=0x010, count=2, unit=1)
```
This seems to be a invalid request `address = 0x001, count =65, unit=1` . I did not see any registers starting at address `0`. If you see the serial forwarder logs, you are getting the response `illegal address` that means the slave is not able to identify the address range. Also in your serial client logs you can see `DEBUG:pymodbus.transaction:Incomplete message received, Expected 135 bytes Recieved 17 bytes !!!!` . Which means you are either getting in complete response or your meter is just returning the values from the buffer. One possible explantion could be the read timeout you are using. I see you have a timeout of `10s` for the serial client used in the forwareder and `0.5s`for your stand alone serial client. Please run your experiments with the same time out and against the same function codes.

Abbad-Zafar, 2020-08-12:
Sir i am sending you code again.. 
**Code of RTU Forwarder to TCP:**

> from pymodbus.server.sync import StartTcpServer as StartServer
> from pymodbus.client.sync import ModbusSerialClient as ModbusClient
> from pymodbus.datastore.remote import RemoteSlaveContext
> from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
> 
> from pymodbus.client.sync import ModbusTcpClient
> 
> import logging
> logging.basicConfig()
> log = logging.getLogger()
> log.setLevel(logging.DEBUG)
> 
> def run_serial_forwarder():
>     
>     client = ModbusClient(method='rtu', port='/dev/ttyUSB0', baudrate=9600, timeout=0.5)
>     store = RemoteSlaveContext(client)
>     context = ModbusServerContext(slaves = {1: store}, single=False)
> 
>     StartServer(context, address=("0.0.0.0", 5020))
>     
> if __name__ == "__main__":
>     run_serial_forwarder()
**Logs of this code :**

> DEBUG:pymodbus.server.sync:Started thread to serve client at ('127.0.0.1', 56935)
> DEBUG:pymodbus.server.sync:Client Connected [127.0.0.1:56935]
> DEBUG:pymodbus.server.sync:Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.factory:Factory Request[ReadHoldingRegistersRequest: 3]
> DEBUG:pymodbus.datastore.remote:validate[3] 16:2
> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: 0x0 0x3 0x0 0x10 0x0 0x2 0xc4 0x1f
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)) 
> DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
> DEBUG:pymodbus.transaction:Getting transaction 0
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> ERROR:pymodbus.pdu:Exception Response(131, 3, IllegalAddress)
> DEBUG:pymodbus.server.sync:send: [Exception Response(131, 3, IllegalAddress)]- b'000100000003018302'
> DEBUG:pymodbus.server.sync:Handling data: 
> DEBUG:pymodbus.framer.socket_framer:Processing: 
> DEBUG:pymodbus.server.sync:Client Disconnected [127.0.0.1:56935]

**Code for Reading from Tcp forwarder:**

> from pymodbus.client.sync import ModbusTcpClient as ModbusClient
> 
> import logging
> logging.basicConfig()
> log = logging.getLogger()
> log.setLevel(logging.DEBUG)
> 
> host = "0.0.0.0"
> port = 5020 
> 
> client = ModbusClient(host, port, timeout = 0.5)
>     
> client.connect()
>     
> print("Is Client Connected =",client.connect())
> 
> read = client.read_holding_registers(address = 0x0010, count = 2, unit=1)
> print(read)
> 
> client.close()

**Log of this code:**

> Is Client Connected = True
> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)) 
> DEBUG:pymodbus.framer.socket_framer:Processing: 
> DEBUG:pymodbus.transaction:Getting transaction 1
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)
> Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)
>

Abbad-Zafar, 2020-08-12:
@dhoomakethu can there be RTUFramer issue ?

Abbad-Zafar, 2020-08-12:
Sir again getting same errors.
**Code of RTU Forwarder to TCP:**

> from pymodbus.server.sync import StartTcpServer as StartServer
> from pymodbus.client.sync import ModbusSerialClient as ModbusClient
> from pymodbus.datastore.remote import RemoteSlaveContext
> from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
> 
> from pymodbus.client.sync import ModbusTcpClient
> 
> import logging
> logging.basicConfig()
> log = logging.getLogger()
> log.setLevel(logging.DEBUG)
> 
> def run_serial_forwarder():
>     
>     client = ModbusClient(method='rtu', port='/dev/ttyUSB0', baudrate=9600, timeout=10)
>     store = RemoteSlaveContext(client)
>     context = ModbusServerContext(slaves = {1: store}, single=False)
> 
>     StartServer(context, address=("0.0.0.0", 5020), timeout=10)
>     
> if __name__ == "__main__":
>     run_serial_forwarder()

**Logging Data of this Code:**

> DEBUG:pymodbus.server.sync:Started thread to serve client at ('127.0.0.1', 52951)
> DEBUG:pymodbus.server.sync:Client Connected [127.0.0.1:52951]
> DEBUG:pymodbus.server.sync:Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.factory:Factory Request[ReadHoldingRegistersRequest: 3]
> DEBUG:pymodbus.datastore.remote:validate[3] 16:2
> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: 0x0 0x3 0x0 0x10 0x0 0x2 0xc4 0x1f
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)) 
> DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
> DEBUG:pymodbus.transaction:Getting transaction 0
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> ERROR:pymodbus.pdu:Exception Response(131, 3, IllegalAddress)
> DEBUG:pymodbus.server.sync:send: [Exception Response(131, 3, IllegalAddress)]- b'000100000003018302'
> DEBUG:pymodbus.server.sync:Handling data: 
> DEBUG:pymodbus.framer.socket_framer:Processing: 
> DEBUG:pymodbus.server.sync:Client Disconnected [127.0.0.1:52951]
> DEBUG:pymodbus.server.sync:Started thread to serve client at ('127.0.0.1', 34173)
> DEBUG:pymodbus.server.sync:Client Connected [127.0.0.1:34173]
> DEBUG:pymodbus.server.sync:Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.factory:Factory Request[ReadHoldingRegistersRequest: 3]
> DEBUG:pymodbus.datastore.remote:validate[3] 16:2
> DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
> DEBUG:pymodbus.transaction:Running transaction 2
> DEBUG:pymodbus.transaction:SEND: 0x0 0x3 0x0 0x10 0x0 0x2 0xc4 0x1f
> DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - None, Current Time stamp - 1597235031.439268
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Incomplete message received, Expected 9 bytes Recieved 0 bytes !!!!
> DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
> DEBUG:pymodbus.transaction:RECV: 
> DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
> DEBUG:pymodbus.transaction:Getting transaction 0
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> ERROR:pymodbus.pdu:Exception Response(131, 3, IllegalAddress)
> DEBUG:pymodbus.server.sync:send: [Exception Response(131, 3, IllegalAddress)]- b'000100000003018302'
> DEBUG:pymodbus.server.sync:Handling data: 
> DEBUG:pymodbus.framer.socket_framer:Processing: 
> DEBUG:pymodbus.server.sync:Client Disconnected [127.0.0.1:34173]

**Code of RTU to TCP Reciever:**

> import logging
> logging.basicConfig()
> log = logging.getLogger()
> log.setLevel(logging.DEBUG)
> 
> host = "0.0.0.0"
> port = 5020 
> 
> client = ModbusClient(host, port, timeout = 10)
>     
> client.connect()
>     
> print("Is Client Connected =",client.connect())
> 
> read = client.read_holding_registers(address = 0x0010, count = 2, unit=1)
> 
> print(read)
> client.close()

**Log of this code:**

> Is Client Connected = True
> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 [...]

dhoomakethu, 2020-08-12:
@Abbad-Zafar I see you are missing `unit` in your `RemoteSlaveContext` can you change it to `store = RemoteSlaveContext(client, unit=1)` and try. This should work.

Abbad-Zafar, 2020-08-13:
Sir there is some progress but receiving new errors can you please look into it. Thanks

**Code of Serial Forwarder:**

> from pymodbus.server.sync import StartTcpServer as StartServer
> from pymodbus.client.sync import ModbusSerialClient as ModbusClient
> from pymodbus.datastore.remote import RemoteSlaveContext
> from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
> 
> from pymodbus.client.sync import ModbusTcpClient
> 
> import logging
> logging.basicConfig()
> log = logging.getLogger()
> log.setLevel(logging.DEBUG)
> 
> def run_serial_forwarder():
>     
>     client = ModbusClient(method='rtu', port='/dev/ttyUSB0', baudrate=9600, timeout=10)
>     store = RemoteSlaveContext(client, unit = 1)
>     context = ModbusServerContext(slaves = {1: store}, single=False)
> 
>     StartServer(context, address=("0.0.0.0", 5020), timeout=10)
>     
> if __name__ == "__main__":
>     run_serial_forwarder()

**Log of this code:**

> 
> DEBUG:pymodbus.server.sync:Started thread to serve client at ('127.0.0.1', 48321)
> DEBUG:pymodbus.server.sync:Client Connected [127.0.0.1:48321]
> DEBUG:pymodbus.server.sync:Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.factory:Factory Request[ReadHoldingRegistersRequest: 3]
> DEBUG:pymodbus.datastore.remote:validate[3] 16:2
> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x0 0x10 0x0 0x2 0xc5 0xce
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
> DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x4 0x0 0x0 0x0 0xcf 0xba 0x67
> DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x4 0x0 0x0 0x0 0xcf
> DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
> DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
> DEBUG:pymodbus.transaction:Adding transaction 1
> DEBUG:pymodbus.transaction:Getting transaction 1
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> DEBUG:pymodbus.datastore.remote:get values[3] 16:2
> DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
> DEBUG:pymodbus.transaction:Running transaction 2
> DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x0 0x10 0x0 0x2 0xc5 0xce
> DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1597304260.345888, Current Time stamp - 1597304260.353971
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)) 
> DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
> DEBUG:pymodbus.transaction:Getting transaction 1
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> ERROR:pymodbus.server.sync:Socket exception occurred Traceback (most recent call last):
>   File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/server/sync.py", line 202, in handle
>     single=single)
>   File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
>     self._process(callback)
>   File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/framer/socket_framer.py", line 181, in _process
>     callback(result)  # defer or push to a thread?
>   File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/server/sync.py", line 86, in execute
>     self.send(response)
>   File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/server/sync.py", line 227, in send
>     pdu = self.framer.buildPacket(message)
>   File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/framer/socket_framer.py", line 206, in buildPacket
>     data = message.encode()
>   File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/register_read_message.py", line 77, in encode
>     result = int2byte(len(self.registers) * 2)
> TypeError: object of type 'ModbusIOException' has no len()
> 
> DEBUG:pymodbus.server.sync:Client Disconnected [127.0.0.1:48321]

**Code of Receiver:**

> from pymodbus.client.sync import ModbusTcpClient as ModbusClient
> 
> import logging
> logging.basicConfig()
> log = logging.getLogger()
> log.setLevel(logging.DEBUG)
> 
> host = "0.0.0.0"
> port = 5020 
> 
> client = ModbusClient(host, port, timeout = 10)
>     
> client.connect()
>     
> print("Is Client Connected =",client.connect())
> 
> read = client.read_holding_registers(address = 0x0010, count = 2, unit=1)
> 
> print(read)
> 
> client.close()

**Log of this Code:**

> Is Client Connected = True
> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 [...]

Abbad-Zafar, 2020-08-13:
Thanks Sir i got it working with the above code .. Sometimes it gives the correct Handling_register value sometimes it gives the error above.

Abbad-Zafar, 2020-08-13:
print(read.registers)
AttributeError: 'ModbusIOException' object has no attribute 'registers'
Sometimes i get this error on receiver side .. sometime it works fine . Can you idetify the problem ? Thanks

Abbad-Zafar, 2020-08-13:
When the Code is working

**Forwarder Log:**

> DEBUG:pymodbus.server.sync:Started thread to serve client at ('127.0.0.1', 58051)
> DEBUG:pymodbus.server.sync:Client Connected [127.0.0.1:58051]
> DEBUG:pymodbus.server.sync:Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.factory:Factory Request[ReadHoldingRegistersRequest: 3]
> DEBUG:pymodbus.datastore.remote:validate[3] 16:2
> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x0 0x10 0x0 0x2 0xc5 0xce
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
> DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x4 0x0 0x0 0x0 0xd1 0x3a 0x6f
> DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x4 0x0 0x0 0x0 0xd1
> DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
> DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
> DEBUG:pymodbus.transaction:Adding transaction 1
> DEBUG:pymodbus.transaction:Getting transaction 1
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> DEBUG:pymodbus.datastore.remote:get values[3] 16:2
> DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
> DEBUG:pymodbus.transaction:Running transaction 2
> DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x0 0x10 0x0 0x2 0xc5 0xce
> DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1597308468.790109, Current Time stamp - 1597308468.793193
> DEBUG:pymodbus.framer.rtu_framer:Waiting for 3.5 char before next send - 4.01 ms
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
> DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x4 0x0 0x0 0x0 0xd1 0x3a 0x6f
> DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x4 0x0 0x0 0x0 0xd1
> DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
> DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
> DEBUG:pymodbus.transaction:Adding transaction 1
> DEBUG:pymodbus.transaction:Getting transaction 1
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> DEBUG:pymodbus.server.sync:send: [ReadRegisterResponse (2)]- b'000100000007010304000000d1'
> DEBUG:pymodbus.server.sync:Handling data: 
> DEBUG:pymodbus.framer.socket_framer:Processing: 
> DEBUG:pymodbus.server.sync:Client Disconnected [127.0.0.1:58051]

**Receiver Log:**

> Is Client Connected = True
> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
> DEBUG:pymodbus.transaction:RECV: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x0 0x0 0xd1
> DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x0 0x0 0xd1
> DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
> DEBUG:pymodbus.transaction:Adding transaction 1
> DEBUG:pymodbus.transaction:Getting transaction 1
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> ReadRegisterResponse (2)
> [0, 209]
> ReadRegisterResponse (2)

dhoomakethu, 2020-08-13:
You can use `read.isError()` to check if you are getting valid response or error response and process accordingly.

dhoomakethu, 2020-08-13:
@Abbad-Zafar Will be closing this issue.

closed 2020-08-13 by dhoomakethu

Abbad-Zafar, 2020-08-13:
OK Sir Thanks for your help.

Abbad-Zafar, 2020-08-13:
@dhoomakethu i am getting error response. getting socket error as i have posted previously. Could you help me in it ?

commit e2dc876a (2020-09-08): Update documentation for serial forwarder example. Fixes #525
commit 29f694c8 (2020-09-11): #515 fix repl broadcast (#531)

* 1. update requirements
2. Fix examples
3. Fix #494 - handle_local_echo
4. Fix #500 -- asyncio serial client with already running loop
5. Fix #486 - Pass serial args for asyncio serial client
6. Fix #490 - Typo in decode_data for socker_framer
7. Fix #385 - Support timeouts to break out of responspe await when server goes offline
8. Misc updates

* #516 custom data block fix

* Fix broadcast error  with REPL client #515

* Fix #509 Wrong unit ID referenced in framers

* Update documentation for serial forwarder example. Fixes #525

* Fix unit tests, support pyt [... cut]