pymodbus-dev/pymodbus #527: Getting Socket errors most of time when getting response

Abbad-Zafar, 2020-08-17:
### Versions

* Python: v3
* OS: Raspbian
* Pymodbus: Latest
* Modbus Hardware (if used): OB418 https://owen-brothers.co.uk/pub/media/wysiwyg/OB418_User_Manual.pdf

### Pymodbus Specific
* Server: RTU to TCP Forwarder
* Client: Read TCP Register by sending request to Forwarder

### Description

I am using RTU to TCP Forwarder. Sometimes it fetches the right register value but most tof the times it gives me error. Can somebody look into it.

### Code and Logs

**Forwarder Code:**
```
from pymodbus.server.sync import StartTcpServer as StartServer
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
from pymodbus.datastore.remote import RemoteSlaveContext
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

from pymodbus.client.sync import ModbusTcpClient

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def run_serial_forwarder():
    
    client = ModbusClient(method='rtu', port='/dev/ttyUSB0', baudrate=9600, timeout=1)
    store = RemoteSlaveContext(client, unit = 1)
    context = ModbusServerContext(slaves = {1: store}, single=False)

    StartServer(context, address=("0.0.0.0", 5020), timeout=1)
    
if __name__ == "__main__":
    run_serial_forwarder()

```
**Forwarder Log:**
``
DEBUG:pymodbus.server.sync:Started thread to serve client at ('127.0.0.1', 53797)
DEBUG:pymodbus.server.sync:Client Connected [127.0.0.1:53797]
DEBUG:pymodbus.server.sync:Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
DEBUG:pymodbus.factory:Factory Request[ReadHoldingRegistersRequest: 3]
DEBUG:pymodbus.datastore.remote:validate[3] 16:2
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 9
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x0 0x10 0x0 0x2 0xc5 0xce
DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1597649623.128918, Current Time stamp - 1597650872.805284
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x4 0x0 0x0 0x0 0xce 0x7b 0xa7
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x4 0x0 0x0 0x0 0xce
DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
DEBUG:pymodbus.datastore.remote:get values[3] 16:2
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 10
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x0 0x10 0x0 0x2 0xc5 0xce
DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1597650872.861957, Current Time stamp - 1597650872.869699
DEBUG:pymodbus.framer.rtu_framer:Waiting for 3.5 char before next send - 4.01 ms
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)) 
DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
**ERROR:pymodbus.server.sync:Socket exception occurred Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/server/sync.py", line 202, in handle
    single=single)
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/framer/socket_framer.py", line 181, in _process
    callback(result)  # defer or push to a thread?
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/server/sync.py", line 86, in execute
    self.send(response)
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/server/sync.py", line 227, in send
    pdu = self.framer.buildPacket(message)
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/framer/socket_framer.py", line 206, in buildPacket
    data = message.encode()
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/register_read_message.py", line 77, in encode
    result = int2byte(len(self.registers) * 2)
TypeError: object of type 'ModbusIOException' has no len()**

DEBUG:pymodbus.server.sync:Client Disconnected [127.0.0.1:53797]
``
**Receiver Code:**
``
rom pymodbus.client.sync import ModbusTcpClient as ModbusClient

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

host = "0.0.0.0"
port = 5020 

client = ModbusClient(host, port, timeout = 1)
    
client.connect()
    
print("Is Client Connected =",client.connect())

read = client.read_holding_registers(address = 0x0010, count = 2, unit=1)

print(read)
print(read.registers)
``

**Receiver Logs:**
``
Is Client Connected = True
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x2
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)) 
DEBUG:pymodbus.framer.socket_framer:Processing: 
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
abbad
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)
Traceback (most recent call last):
  File "tcp.py", line 26, in <module>
    print(read.registers)
AttributeError: 'ModbusIOException' object has no attribute 'registers'

``

Abbad-Zafar, 2020-08-17:
@dhoomakethu

Abbad-Zafar, 2020-08-17:
Have changed the timeout but no change still getting error most of time. Sometimes i get the correct register value otherwise errors

dhoomakethu, 2020-08-17:
It also depends on how fast you are querying the results. The general practice is to poll every <n> seconds where n>read timeout . Pymodbus generally is the mediator and sometime the slave unit is not quick enough to handle multiple request with in a short spawn of time.

Abbad-Zafar, 2020-08-17:
Sir mostly when i turn on Energy Meter and send the single request but it still gives me error.

dhoomakethu, 2020-08-17:
Your logs says there is no response received, could be that the slave is still not ready. Can you write a simple code which polls the meter every <n> seconds and log the number of failures against the success cases. That should provide you more inputs on what could be happening. Also , is this the issue faced only with the forwarder or is the same issue observed with standl alone RTU client as well?

Abbad-Zafar, 2020-08-17:
No Sir standalone RTU works fine ... Okay Sir i will write code to poll meter every seconds

Abbad-Zafar, 2020-08-17:
Sir tried this code. But it stops as i get error. Do you know how to get it looping to log failure and success cases?

> 
> from pymodbus.client.sync import ModbusTcpClient as ModbusClient
> 
> import time
> 
> import logging
> logging.basicConfig()
> log = logging.getLogger()
> log.setLevel(logging.DEBUG)
> 
> while 1:
>     
>     time.sleep(1)
> 
>     host = "0.0.0.0"
>     port = 5020 
> 
>     client = ModbusClient(host, port, timeout = 2)
>         
>     client.connect()
>         
>     print("Is Client Connected =",client.connect())
> 
>     read = client.read_holding_registers(address = 0x0010, count = 2, unit=1)
> 
>     if read.isError():
>         print("error")
>     print(read)
>
>     print(read.registers)
> 
> 
>     #---------------------------------------------------------------------------# 
>     # close the client
>     #---------------------------------------------------------------------------# 
>     client.close()
>

Abbad-Zafar, 2020-08-17:
Sir sometimes it reads successfully the registers in 1 seconds. and sometimes it doesn't. Cant identify the problem

Abbad-Zafar, 2020-08-17:
Update: Tried a while loop which constantly read register until it receives correct value

dhoomakethu, 2020-08-17:
Does that work @Abbad-Zafar ? How many errors before getting the actual result?

Abbad-Zafar, 2020-08-17:
There is no specific time. Sometime it receives correct result on first try some time it takes 4,5 tries.

Abbad-Zafar, 2020-08-20:
@dhoomakethu something in your mind other than while loop ?. I have tried everything still could get a clean response.

dhoomakethu, 2020-10-28:
@Abbad-Zafar Are you still having this issue or can this be closed?

Abbad-Zafar, 2020-10-28:
Yes its resolved. Thank You very much.

closed 2020-10-28 by Abbad-Zafar