pymodbus-dev/pymodbus #528: example scripts for async usage with tornado?

dcneeme, 2020-08-17:
Could anybody confirm that the example scripts (linked below) to test async pymodbus with tornado are still valid for pymodbus 2.3.0 - 2.4.0 (have problems)? If not, has anyone a working script (both rtu and tcp are of interest)?

https://pymodbus.readthedocs.io/en/latest/source/example/async_tornado_client.html
https://pymodbus.readthedocs.io/en/latest/source/example/async_tornado_client_serial.html

Thanks

dhoomakethu, 2020-08-17:
what problems are you facing with those scripts?

dcneeme, 2020-08-18:
4 async trials below. using python3.7.5 and tornado 6.0.4. sync communication with the same rtu slave and tcp server is ok, with both pymodbus versions.

1. # python3 async_tornado_client.py # pymodbus2.3.0 tcp / keeps running, outputs nothing
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:asyncio:Using selector: EpollSelector
INFO:pymodbus.client.asynchronous.thread:Starting Event Loop: 'PyModbus_ioloop'
DEBUG:root:Client connected
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.7/logging/__init__.py", line 1025, in emit
    msg = self.format(record)
  File "/usr/lib/python3.7/logging/__init__.py", line 869, in format
    return fmt.format(record)
  File "/usr/lib/python3.7/logging/__init__.py", line 608, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.7/logging/__init__.py", line 369, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/usr/lib/python3.7/threading.py", line 890, in _bootstrap
    self._bootstrap_inner()
  File "/usr/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pymodbus/client/asynchronous/thread.py", line 37, in _start
    self._start_loop(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/dist-packages/tornado/platform/asyncio.py", line 149, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 534, in run_forever
    self._run_once()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1771, in _run_once
    handle._run()
  File "/usr/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "async_tornado_client.py", line 112, in callback
    return err(exp)
  File "async_tornado_client.py", line 105, in err
    log.error("Err", args, kwargs)
Message: 'Err'
Arguments: ((TypeError("__init__() got an unexpected keyword argument 'io_loop'"),), {})

2. # python3 async_tornado_client_serial.py # pymodbus2.3.0 rtu - keeps running, outputs nothing, no activity on serial line
2020-08-17 18:27:43,559 MainThread      DEBUG    selector_events:53       Using selector: EpollSelector
2020-08-17 18:27:43,561 MainThread      DEBUG    selector_events:53       Using selector: EpollSelector
2020-08-17 18:27:43,563 MainThread      INFO     thread         :44       Starting Event Loop: 'PyModbus_ioloop'
2020-08-17 18:27:43,567 MainThread      ERROR    __init__       :314      __init__() got an unexpected keyword argument 'io_loop'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pymodbus/client/asynchronous/tornado/__init__.py", line 312, in connect
    self.stream = SerialIOStream(conn, io_loop=self.io_loop)
  File "/usr/local/lib/python3.7/dist-packages/pymodbus/client/asynchronous/tornado/__init__.py", line 248, in __init__
    super(SerialIOStream, self).__init__(*args, **kwargs)
TypeError: __init__() got an unexpected keyword argument 'io_loop'
2020-08-17 18:27:43,569 MainThread      DEBUG    __init__       :317      Client connected

3. # python3 async_tornado_client.py # pymodbus2.4.0 tcp - script stops
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:asyncio:Using selector: EpollSelector
INFO:pymodbus.client.asynchronous.thread:Starting Event Loop: 'PyModbus_ioloop'
#

4. # python3 async_tornado_client_serial.py # pymodbus2.4.0 rtu - script stops, no activity on serial line
- script stops, no activity on serial line
2020-08-17 18:30:33,256 MainThread      DEBUG    selector_events:53       Using selector: EpollSelector
2020-08-17 18:30:33,258 MainThread      DEBUG    selector_events:53       Using selector: EpollSelector
2020-08-17 18:30:33,259 MainThread      INFO     thread         :45       Starting Event Loop: 'PyModbus_ioloop'
2020-08-17 18:30:33,263 MainThread      ERROR    __init__       :314      __init__() got an unexpected keyword argument 'io_loop'
Traceback (most recent call last):
  File "/root/pyapp/pymodbus/client/asynchronous/tornado/__init__.py", line 312, in connect
    self.stream = SerialIOStream(conn, io_loop=self.io_loop)
  File "/root/pyapp/pymodbus/client/asynchronous/tornado/__init__.py", line 248, in __init__
    super(SerialIOStream, self).__init__(*args, **kwargs)
TypeError: __init__() got an unexpected keyword argument 'io_loop'
2020-08-17 18:30:33,265 MainThread      DEBUG    __init__       :317      Client connected
#

dhoomakethu, 2020-08-19:
Thanks, Will take a look. I assume you have tested on pymodbus 2.3.0 and 2.4.0rc1 ?

dcneeme, 2020-08-19:
yes

dcneeme, 2020-08-28:
hi,
keeping fingers crossed that you get a chance to look at it some time soon.

dhoomakethu, 2020-08-28:
@dcneeme seems to be working with the simulation setup . Any particular real hardware you are trying this against?

dhoomakethu, 2020-08-28:
Ok, on 2.3.0 do `pip install tornado==4.5.2` .This was the last compatible tornado with pymodbus. Please give it a try and update your findings.

dcneeme, 2020-09-24:
Thanks for the tornado version hint, it helped further. Tried both pymodbus 2.3.0 and 2.4.0 with this tornado version, both TCP and RTU. Both have similar problems and are unfortunately unusable as of now.

async tcp is slightly broken  - works with 10 s pauses every now and then.
async rtu is totally broken and unusable.

Cougar has taken over my investigations. See his findings about async  RTU problems and possible solution in issue 533 (he has got it working with some clever patching). I hope he has time to look into the async TCP issues as well.

dhoomakethu, 2020-09-24:
@dcneeme I am pretty sure there are issues with tornado but asyncio should be working based on my limited testing. If you have any error logs for asyncio please share so that I can take a look.

closed 2021-03-03 by dhoomakethu