pymodbus-dev/pymodbus #530: Examples related to Modbus TLS connection.

LordGameleo, 2020-09-08:
### Versions

* Python: Python3
* OS: Ubuntu
* Pymodbus: Latest
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

Is there any example related to ModbusTLS Connection? Both Server and Client Side. 
How can I test both? Is there any software for it?

LordGameleo, 2020-09-08:
Never Mind. I found it in comments of PR https://github.com/riptideio/pymodbus/pull/439 .  If possible please add direct link to the Readme.md

dhoomakethu, 2020-09-08:
Just in case. Here are the examples.

https://github.com/riptideio/pymodbus/blob/dev/examples/common/synchronous_server.py#L122

LordGameleo, 2020-09-08:
@dhoomakethu thanks for response over such a short time. Found it. But why do we have client example in /contrib/ and server example in /common/ Can we create a simple example for server as well in /contrib/ directory?

dhoomakethu, 2020-10-28:
@LordGameleo those examples are provided by the contributors and hence are maintained there.

closed 2020-10-28 by dhoomakethu